evcc-io/evcc #2869: Negative Erzeugungsleistung falsch dargestellt

premultiply, 2022-03-12:
Problem:  Negative Erzeugungsleistung (WR-Eigenverbrauch nachts) wird in der WebUI als (positive) Erzeugung betrachtet.

Lösungsvorschlag: Negative Erzeugung sollte dem Verbrauch zugeschlagen werden und Erzeugung dann 0 sein.

Details siehe #2836

### Discussed in https://github.com/evcc-io/evcc/discussions/2836

<div type='discussions-op-text'>

<sup>Originally posted by **hillwig** March  9, 2022</sup>
Hallo, mein pv-meter zeigt immer den Verbrauch des Wechselrichters (Solaredge-Hybrid) als negativ und im web als Erzeugung an.
Gibt es dazu Lösungen? evcc meter:

grid1
-----
Power: 7W

pv2
---
Power: -67W

pv3
---
Power: 14W

battery3
--------
Power: 351W
SoC:   46%

charge4
-------
Power:          0W
Current L1..L3: 0A 0A 0A

Danke und Grüße
Marcus
</div>

andig, 2022-03-13:
Handling im UI (weil die Werte ja alle richtig sind) oder im Backend? /cc @naltatis

naltatis, 2022-03-13:
Ich find den Vorschlag von @premultiply richtig deb Stromverbrauch des WRs auch in den Hausverbrauch mit aufzunehmen. Der wird ja inzwischen im Backend bestimmt.

naltatis, 2022-03-13:
Eine Sache noch die ich nicht genau weiß. Ich meine wir hatten mal die Logik in der UI, dass der Max Wert für PV immer 0 war - also negative Werte ignoriert wurden. @DerAndereAndi hatte in seiner lokalen Konfiguration aber die Produktionswerte für PV als negative Werte. Daher kommt auch die jetzige abs() Logik.

Mit dieser Umstellung muss der PV Produktionswert dann immer ein positiver Wert sein. Möglicherweise ist das ein Breaking Change für bestehende Installationen.

Für die Umsetzung: Eigentlich brauchen wir ja nur negative PV Werte ignorieren. Das Zuschlagen auf den Hausverbrauch passiert ja automatisch.

andig, 2022-03-13:
> Mit dieser Umstellung muss der PV Produktionswert dann immer ein positiver Wert sein. Möglicherweise ist das ein Breaking Change für bestehende Installationen.

Das ist bisher schon so, keine Änderung. Du siehst es auch an dem Check im PR.

closed 2022-03-15 by andig