evcc-io/evcc #2880: Add support for DSMR digital meters from Belgium and Netherlands

hyperbart, 2022-03-14:
**Is your feature request related to a problem? Please describe.**
Currently there is no ability to add DSMR meters to EVCC. Since every home in Belgium/Netherlands will be equipped with a digital meter instead of a disc meter it makes sense to add support for this.

**Describe the solution you'd like**
Enable EVCC to interface with DSMR as a meter

**Describe alternatives you've considered**
Buying and installing a MODBUS meter but this does not always measure the same values as the DSMR does.

**Additional context**
https://github.com/roaldnefs/go-dsmr
[image]

andig, 2022-03-14:
@hyperbart I‘d be happy to add this. However, the dsmr repo doesn‘t show any example how to connect to the meter and retrieve data. It only seems to provide the parsing. Could you fill the gap?

andig, 2022-03-14:
Got it. Seems it should be read over serial (are there other options?) and example is in https://github.com/basvdlei/gotsmart/blob/master/gotsmart.go.

hyperbart, 2022-03-14:
Ja, genau it is a serial connection.
Of course one can do serial over TCP (which is quite common in the case of the DSMR) as the metercabinet is often located further away from the server/NAS that is running at a home.

andig, 2022-04-02:
Closing here. Without access implementation won‘t be possible.

stefaanbolle, 2022-04-21:
DSMR Reader ([https://github.com/dsmrreader/dsmr-reader](https://github.com/evcc-io/evcc/discussions/url) is very easy to setup (ie raspberry pi) and provides a wealth of smart meter data. Besides it is capable of exporting it's readings to MQTT or to store it in an InfluxDB. It also provides an API to obtain read values and the API documentation is public.

hyperbart, 2022-04-22:
I have procured a P1 Reader over Ethernet. It's arriving next week.

hyperbart, 2022-04-30:
By the way: The Belgian meters are based on the DSMR protocol which originated from the Netherlands. Belgian meters are 100% compatible with DSMR but also include some more data which the Dutch meters don't:

https://www.fluvius.be/sites/fluvius/files/2020-01/dmk-demo-v2.1-rtc.pdf

--> Slide 45

[image]

Telegram structure:

[image]

Telegram detail:

[image]

hyperbart, 2022-04-30:
@stefaanbolle : are you Belgian or Dutch? Do you have a 3phase meter available to publish?

hyperbart, 2022-04-30:
@andig : 0-1:17.0.0 and 1-0:31.4.0 are both values that our grid operators can use to limit the kW total and Amps per phase. Right now it is not being used in Belgium but it is one of those things we Belgians have but Dutch people don't.

andig, 2022-04-30:
That will only become relevant when we support load management.

hyperbart, 2022-04-30:
I have reached out to the Dutch forum I mentioned earlier. Hope somebody reports back here and allows you to access a 3P DSMR.

JeroenEeuwes, 2022-05-01:
@hyperbart @andig I have a 3 phase DMSR4 meter you can use for testing.

andig, 2022-05-01:
Implementation is done, waiting for license approval

fluppie, 2022-05-01:
Belgium 3P+N smart meter DSMR5
[image]
[image]

No solar panels, export data is from the construction crane, while braking it acted for a very short period of time as a generator :).

closed 2022-05-01 by andig

stefaanbolle, 2022-05-05:
For the ones interested, implementing DSMR-Reader via API:

```
meters:
- name: dsmr
  type: custom
  power:
    source: calc
    add:
    - source: http
      uri: http://<DSMR-READER URL>/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: <API Key>
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_delivered}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://<DSMR-READER URL>/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: <API Key>
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_returned}| join(" ")
      scale: -1
      timeout: 10s
  energy:
    source: calc
    add:
    - source: http
      uri: http://<DSMR-READER URL>/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: <API Key>
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_merged}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://<DSMR-READER URL>/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: <API Key>
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_returned_merged}| join(" ")
      scale: -1
      timeout: 10s
  currents:
  - source: http
    uri: http://<DSMR-READER URL>/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: <API Key>
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l1}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://<DSMR-READER URL>/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: <API Key>
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l2}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://<DSMR-READER URL>/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: <API Key>
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l3}| join(" ")
    scale: 1
    timeout: 10s
```

hyperbart, 2022-10-11:
I will report back if I have set up the needed remote access.

vinc-vo, 2022-10-28:
So what exactly is meant with TCP Connection? I bought a USB to P1 adapter and intend to connect it straight to one of the USB ports of a Raspberry Pi running EVCC. 
Will ser2net on localhost work? Or do I need a hardware p1 ethernet reader?

hyperbart, 2022-10-28:
Hardware p1 ethernet.
Ser2net might work. P1 ethernet has been tested.

Biestig, 2022-10-29:
Hey i also have the p1 meter from zuidwijk.com.  Is there already a way to implement this reader with evcc?

hyperbart, 2022-10-29:
You should start at reading the manual.
Install EVCC as you would like and then configure the DSMR reader as described in the template. So answer: yes, there is a way. You will have to just configure it.

Biestig, 2022-10-29:
ok so if i understand correctly, i have to install esp link to the slimme meter + (follow manual on zuidwijk.com) and then connect it to dsmr reader on home assistant.  Next i have to config evcc with the template and then the meter should work.
Sorry for the dumb questions but like i said, this is the first for me with some coding.

hyperbart, 2022-10-29:
You can't install anything on the slimme meter.
You simply buy the zuidwijk.com meter, plug it in and configure EVCC.
The template is for zuidwijk.com meter. It has nothing to do with ESP link. I just bought it, plugged it in and done.

Digital Meter --> P1 Reader Ethernet --> Network cable

Biestig, 2022-10-29:
i see that you have the zuidwijk.com ethernet version, i have bought the slimme lezer + from his site, it's the wireless version, it came with esp home on it and i can switch to esp link. i'm trying it now and see what it does.  When i did the template it said that it couldnt connect to the adress...
i'll try some things first and report back

hyperbart, 2022-10-29:
Aha, that explains the confusion. I didnt understand what ESP Link had to do with it.
Post some screenshots maybe of the Slimme Lezer (web interface and command output, can you connect text based to it? That's how I had to connect to my meter via Putty and then it just started outputting all the data on my screen).

Based on that we can maybe start a new thread and see if @andig sees it fit to add support for the Slimme Lezer/other ways to interpret the DSMR output too?

Biestig, 2022-10-29:
ok, now i'm a bit confused :-).  So ive installed esp link to it, i can see the data in home assistant with the integration dsmr slimme meter. Have not configured the integration dsmr reader.
If i go to ip adress of the slimme meter + i get the web interface, of what do you need a screenshot of?

Biestig, 2022-10-29:
ok, ive entered the ip in evcc, now it no longer gives error from grid meter, just from the charger but i haven't configured that yet cause it's an alfen charger and i need a sponser token first.  Just wanted to make sure the meter was ok

andig, 2022-10-29:
I would appreciate if we didn't have off-topic discussions in long-closed PRs. I'm a bit confused though what is really being discussed here. DSMR protocol over TCP is implemented. What is this "other" DSMR that is being discussed here and could we please have a new issue for that?

Biestig, 2022-10-29:
sorry for the off-topic, will make look out for it in the future.

vinc-vo, 2022-11-07:
For anyone interested, an update on my part: I've got it working using ser2net 🥳 

My /etc/ser2net.conf file looks like this:
`2001:raw:600:/dev/tty-DSMR_1:115200 NONE 1STOPBIT 8DATABITS XONXOFF LOCAL -RTSCTS`

and the grid meter section in the /etc/evcc.yaml looks like this:
`meters:
- type: template
  template: dsmr
  usage: grid
  host: localhost
  port: 2001
  energy: 1-0:1.8.1/1-0:1.8.2
  name: grid1`
  
It is a Belgian Sagemcom T211 (3 fase meter, double tariff) and I'm using a cheap Ali Express P1 adapter. Couldn't find any specifically for my meter, but there is lots for sale for Sagemcom XS210 and that is working just fine. My raspberry pi is close enough so I'm using a 3-meter usb extension cord and the P1 adapter adds another 3 meters of cable.

It is working very nicely, however I do get some warning triangle with some "errors" in EVCC dashboard.
`grid meter energy: not available: 1-0:1.8.1/1-0:1.8.2`
I guess since the meter sends info each second, evcc tries to read before the information has all gone through. Not sure though. There are enough good readings to just ignore this for now.

vinc-vo, 2022-11-07:
Because that is not supported by evcc yet

andig, 2022-11-07:
Afaiu there are DSMR meters with Ethernet and serial port? The latter ones are indeed currently not supported. Are they common?

vinc-vo, 2022-11-07:
All DSMR meters have a serial port, none of them (as far as I know) have an ethernet port. But because the meter is often far away from a server, there is 3rd party hardware that puts the serial on the ethernet. In my case the meter was close enough to plug it in directly into the server.

marciej123, 2023-02-07:
Is there anyone who got this working with a USB to P1 cable to get the values from here? I have DSMR reader set-up and working in my Home Assistant, but i cant seem to get EVCC working with the information from there..

vinc-vo, 2023-02-07:
I did using ser2net, see my answer above 😄 
You can't use HA and ser2net/EVCC on the same serial port at the same time. 

You'll have to configure one to read the information from the api of the other or find a tool to split the clone the serial output.