evcc-io/evcc #2895: Add Silence S01/Seat Mo vehicle api

patbab, 2022-03-15:
Liebe EVCC-Entwickler,

Der Silence S01 ist ein Elektroroller der L3e/A1-Klasse (bis 11 kW Dauerleistung), vmax=100 km/h.
https://www.silence.eco/
Er wird baugleich von Seat unter dem Namen "Seat Mo" vertrieben und wird von der in Barcelona ansässigen Firma Silence hergestellt. Von Silence gibt es inzwischen mehrere Versionen des S01, eine eher auf kommerzielle Verwendung abgestellte Version S02, sowie ein geplantes Kleinst-BEV (S04).
Der Roller findet immer mehr Verbreitung in Europa, in Deutschland gibt es u.a. eine rege Community auf https://www.elektroroller-forum.de/viewforum.php?f=128. 
Nachdem andere Roller wie NIU bereits in evcc ansprechbar sind, wäre es sicher sinnvoll, auch den S01 zu implementieren.

Es gibt bereits eine Implementation für Home Assistant, die vermutlich gut als Vorlage dienen kann:
https://github.com/lorenzo-deluca/homeassistant-silence
Aus mangelnder Erfahrung bzw. Zeit es selbst zu lernen, kann ich die Integration in evcc leider nicht selber in Angriff nehmen. Ich kann aber gern testen...

Der Roller lädt mit einer konstanten Leistung von ca. 600W bei 230V (also knapp 3A) und hat einen Akku von brutto 5,6 kWh. Effektiv nutzbar sind ca. 5 kWh. Als "Wallbox" fungiert idealerweise eine FritzDECT.

Es ist im Moment in der Silence-Community unklar, ob Seat für den Mo die selbe Schnittstelle benutzt. Die App schaut etwas anders aus.

Es würde mich und die Silence-Community sehr freuen, wenn Ihr tätig werden würdet.
Patrick

andig, 2022-03-15:
Das sieht machbar aus. Ich bräuchte allerdings Zugriff auf ein Testexemplar. Gerne Mail an info@evcc.io.

lucasimons, 2022-12-23:
Can you add the seat mò?

andig, 2022-12-23:
It has long since been added.