evcc-io/evcc #2904: Does not start after update from 0.85 to 0.86

RTTTC, 2022-03-16:
Same thing happened when updating from 0.85 to 0.86 using `sudo apt upgrade`

I got the error 
```

Unpacking evcc (0.86.0) over (0.85.0) ...
dpkg: warning: unable to delete old directory '/usr/local': Directory not empty
```
After the update EVCC does not start. Will have to try and change directories I guess?

_Originally posted by @RTTTC in https://github.com/evcc-io/evcc/discussions/2825#discussioncomment-2372544_

Pitillo, 2022-03-16:
I needed to edit the service file under ````/etc/systemd/system/evcc.service``` and change ````ExecStart=/usr/local/bin/evcc --log error``` to ````ExecStart=/usr/bin/evcc --log error```. Maybe you have one service file left in ```/etc/systemd/system/``` since it has higher priority than the one located under ```/usr/lib/systemd/system/```. After you altered the file you need to execute ```systemctl daemon-reload``` and ```systemctl restart evcc```.

For further debugging: could you post the output of ```journalctl -u evcc -n 25```?

RTTTC, 2022-03-16:
Thank you, that was exactly the case!!! 
Now everything runs fine.
Should I delete something that is old and not needed? And how to do that?

Pitillo, 2022-03-16:
In my opinion you could delete the file ```/etc/systemd/system/evcc.service``` using ```rm /etc/systemd/system/evcc.service``` if a similar file is inplace under ```/usr/lib/systemd/system/evcc.service``` afterwards run ```systemctl deamon-reload``` again. You should then be safe for further updates if they change something to service file. Further I would recommend to mark the issue as solved.

closed 2022-03-16 by RTTTC