evcc-io/evcc #2919: Saving the statistics after server restart

JanDragon, 2022-03-17:
**Is your feature request related to a problem? Please describe.**
It would be nice if the statistics of the load data would still be available and displayed after a server restart. 

**Describe the solution you'd like**
The data is saved and after the restart back to the old state. 

**Describe alternatives you've considered**
It might also be possible that a function is built into the config where you can set whether you want to reset the statistics after restart or whether the old should be reloaded. 

**Additional context**

[image]

This picture was taken before the restart. 

[image]

This picture was taken after the reboot.

andig, 2022-03-19:
/cc @xantalor Du hattest da schon eine gute Idee?

Bluppy, 2022-03-25:
Würde ich auch sehr gut finden. Da ich die Stromwerte monatlich erfasse.

naltatis, 2022-03-25:
> Würde ich auch sehr gut finden. Da ich die Stromwerte monatlich erfasse.

Steht auf unserer Todoliste. Allerdings nicht ganz oben 😄

thierolm, 2022-04-01:
@naltatis + @andig : Schaut euch bei Gelegenheit mal #3051 an, ob das eine Lösung sein könnte.

RTTTC, 2022-10-26:
It would be nice if data was also saved at least every day, say at midnight? 
That would somewhat mitigate data loss on grid failure, and I think would not be that damaging to SD card?

PR #4956 (2022-10-25): Persistent Savings
Save and restore savings data in db so they are not lost on restart.
Data is written to disk every minute or on shutdown.

fixes #2919

a new take on #3051
//cc @thierolm