evcc-io/evcc #2983: resetOnDisconnect und targetSoC bei Neustart ignoriert

andig, 2022-03-24:
### Discussed in https://github.com/evcc-io/evcc/discussions/2982

<div type='discussions-op-text'>

<sup>Originally posted by **BlueSnake830** March 24, 2022</sup>
### Describe the bug

Ich habe auf Version 0.87 aktualisiert und dies festgestellt:
- Lademodus "off" wird bei resetOnDisconnect nicht mehr berücksichtigt. Er springt immer auf minPV.
- targetSoC wird nicht mehr berücksichtigt.

Im Changelog der 0.87 steht "Remove deprecated configuration options #2900". In der Dokumentation steht, dass man "resetOnDisconnect" und den default "targetSoc" beim loadpoint definieren kann. Ist die Dokumentation evtl. veraltet und man muss diese Parameter inzwischen wo anders definieren?
targetSoC wäre nicht so schlimm, da ich für jedes Auto einen separaten unter "vehicles" defniert habe. Dass resetOnDisconnect anscheinend  nicht mehr funktioniert, finde ich schlecht, da ich meine Wallbox (Heidelberg EC) gerne gesperrt habe, wenn kein Auto angesteckt ist.

Die Yaml wurde seit der 0.86 nicht verändert. In dieser Version funktionierte resetOnDisconnect und targetSoC noch.

```
loadpoints:
- title: Carport
  charger: wallbox5
  mode: off # charge mode (off, now, minpv, pv)
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true # Einstellungen bei Abstecken des Autos auf Standardwerte zurücksetzen
  soc:
    poll:
      mode: charging
    target: 80
  vehicles:
    - tesla_model_y
   ```

### Steps to reproduce

evcc / Raspberry Pi neu starten. Lademodus springt dann nicht auf den definierten "off".

### Configuration details

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 10.1.20.2  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: 10.1.20.3  
  password: ***********  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 10.1.20.6  
  password: ***********  
  name: pv3
- type: template
  template: sma-inverter 
  usage: battery  
  host: 10.1.20.4  
  password: ***********  
  name: battery4

chargers:
- type: template
  template: heidelberg 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox5

loadpoints:
- title: Carport
  charger: wallbox5
  mode: off # charge mode (off, now, minpv, pv)
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true # Einstellungen bei Abstecken des Autos auf Standardwerte zurücksetzen
  soc:
    target: 80
  vehicles:
    - tesla_model_y

vehicles:
  - name: tesla_model_y
    type: tesla
    title: TESLA Model Y # display name for UI
    capacity: 79 # kWh
    onIdentify:
      mode: minpv
      targetSoC: 75
      minSoC: 30
    tokens:
      access: ***********
      refresh: ***********
    vin: # optional

site:
  title: Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    - pv3
    batteries:
    - battery4
  prioritySoC: 100 # Hausbatterie bekommt bis zum SoC 100% Priorität beim laden; 0 = deaktiviert
  residualPower: 100

hems:
  type: sma
  AllowControl: false
  
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.1975 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.095098 # [currency]/kWh

influx:
  url: http://localhost:8086
  database: evcc
  user: ***********
  password: ***********

# push messages
messaging:
  events:
    start: # charge start event
      title: Ladevorgang gestartet
      msg: Wallbox ${title} startete den Ladevorgang von ${vehicleTitle} im ${mode} Modus.
    stop: # charge stop event
      title: Ladevorgang gestoppt
      msg: Wallbox ${title} beendete den Ladevorgang von ${vehicleTitle} mit ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Fahrzeug verbunden
      msg: ${vehicleTitle} verbunden mit Wallbox ${title} bei ${pvPower:%.1fk}kW PV.
    disconnect: # vehicle connected event
      title: Fahrzeug getrennt
      msg: ${vehicleTitle} getrennt von Wallbox ${title} nach ${connectedDuration}.
  services:
   - type: pushover
     app: ***********
     recipients:
     - ***********
  #  - type: telegram
  #    token: *********** # bot id : jede laufende Instanz von evcc benötigt eine eigene bot id
  #    chats: #- # Liste von Chat oder Group IDs. Jeder Eintrag benötigt ein - Zeichen am Anfang und muss in einer eigenen Zeile sein.
  #     - -*********** #Achtung Group IDs in Telegram haben ein -Zeichen 

sponsortoken: ***********
```

### Log details

```
pi@evcc:~ $ evcc --log debug
[main  ] INFO 2022/03/24 08:19:04 evcc 0.87
[main  ] INFO 2022/03/24 08:19:04 using config file /etc/evcc.yaml
[main  ] INFO 2022/03/24 08:19:04 listening at 0.0.0.0:7070
[semp  ] WARN 2022/03/24 08:19:10 SEMP_BASE_URL unspecified, using http://10.1.20.10:7070 instead
[site  ] INFO 2022/03/24 08:19:10 site config:
[site  ] INFO 2022/03/24 08:19:10   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/03/24 08:19:10     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/03/24 08:19:10     pv 0:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/03/24 08:19:10     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/03/24 08:19:10     battery 0: power ✓ energy ✓ currents ✓ soc ✓
[lp-1  ] INFO 2022/03/24 08:19:10 loadpoint 1:
[lp-1  ] INFO 2022/03/24 08:19:10   mode:        off
[lp-1  ] INFO 2022/03/24 08:19:10   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/03/24 08:19:10   meters:      charge ✓
[lp-1  ] INFO 2022/03/24 08:19:10     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/03/24 08:19:10   vehicles:    ✓
[lp-1  ] INFO 2022/03/24 08:19:10     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/03/24 08:19:10 vehicle updated: unknown -> TESLA Model Y
[lp-1  ] DEBUG 2022/03/24 08:19:10 set charge mode: minpv
[lp-1  ] DEBUG 2022/03/24 08:19:10 pv timer elapse
[lp-1  ] DEBUG 2022/03/24 08:19:10 pv timer inactive
[lp-1  ] DEBUG 2022/03/24 08:19:10 set min soc: 30
[lp-1  ] DEBUG 2022/03/24 08:19:10 set target soc: 75
[site  ] DEBUG 2022/03/24 08:19:10 ----

```
Das Fahrzeug ist zum Zeitpunkt des Logs nicht angesteckt!

### What type of operating system are you running?

Linux

### Version

evcc version 0.87</div>

andig, 2022-03-27:
/cc @rivengh hast Du eine Idee woran das liegen könnte?

rivengh, 2022-03-27:
Was mir auf den ersten Blick aufgefallen ist, dass nur ein Fahrzeug konfiguriert ist. Das wird ja dann immer als verbunden angenommen:

https://github.com/evcc-io/evcc/blob/4460ed59dd00d1eea9135e0d14a7df0dc66a2e92/core/loadpoint.go#L536-L538

Dadurch werden dann - auch direkt nach dem Neustart wie lim Log zu sehen - die Werte dieses Fahrzeugs gesetzt.

Ich schaue es mir gerne aber nachher nochmal in Ruhe an.

andig, 2022-03-27:
@rivengh super, danke! Denke das ist es:

```
    onIdentify:
      mode: minpv
      targetSoC: 75
      minSoC: 30
    tokens:
      access: ***********
      refresh: ***********
    vin: # optional
```

Damit wird immer wieder die Config des Fahrzeugs geschrieben sobald es (oder eben auch ein anderes) verbunden wird.

@bluesnake: das sieht aus wie "works as designed"- Du hast unterschiedliche Konfigurationen die sich gegenseitig übersteuern. Das Fahrzeug gewinnt...

BlueSnake830, 2022-03-27:
Aber die Annahme, dass immer dieses Fahrzeug verbunden ist, macht doch keinen Sinn, oder? 
Ich habe onIdentify so verstanden, dass die Parameter eingestellt werden, sobald ein Fahrzeug als angesteckt erkannt wird. 

Selbe Konfiguration:
0.86 (meiner Meinung nach korrektes Verhalten):
- Raspberry neu gestartet: Modus "off"
- Fahrzeug wird eingesteckt -> Modus schaltet auf minPV.
- Fahrzeug abgesteckt: Modus schaltet auf "off". 

0.87:
- Raspberry neu gestartet (kein Auto angesteckt): Modus "minPV" (meiner Meinung nach falsch, da im Loadpoint onDisconnect "off"

premultiply, 2022-03-27:
Mach am loadpoint mal vehicles weg dann sollte es funktionieren.

andig, 2022-03-27:
…oder am Auto die Einstellungen. Dann gibts wenigstens noch den Soc ;)

BlueSnake830, 2022-03-28:
Also, ich habe das "vehicles" nun beim Loadpoint rausgenommen und bei einem Neustart wird der Modus wieder auf "off" gestellt. Soweit OK.
Mich wundert nur, dass sich das Verhalten von der 0.86 zur 0.87 geändert hat. Egal, Thema kann geschlossen werden.

Vielen Dank für eure sehr schnelle und kompetente Hilfe!

closed 2022-03-28 by andig