evcc-io/evcc #3008: App Wattpilot API

thecem, 2022-03-26:
**Is your feature request related to a problem? Please describe.**
Wattpilot API is out and working:

https://github.com/joscha82/wattpilot

Would be nice if we could implement here. To Stop the charging on SOC.

**Describe the solution you'd like**
Implemetation of Fronius Wattpilot API

**Describe alternatives you've considered**
To be able STOP Wattpilot on SOC level. But the most beautiful would be the implementation in the UI of EVCC!!!

**Additional context**
Add any other context or screenshots about the feature request here.

```
the@Air:~/wattpilot$ python3 example.py <Wattpilot IP> <Wattpilot Passwort>
Wattpilot: Wattpilot_33333332
Serial: 33333332
Connected: True
Car Connected: no car
Charge Status True
Mode: Eco
Power: 10
Charge: 0.00kW ---- 238V/236V/238V -- 0.00A/0.00A/0.00A -- 0.00kW/0.00kW/0.00kW

API Key: <API-Key>
```

andig, 2022-03-26:
Was wird hier angefragt: die Unterstützung der Wattpilot Wallbox?

thecem, 2022-03-26:
Die Unterstützung der Wallbox.

thecem, 2022-03-26:
Die API ist mit der V2 Cloud api identisch.

andig, 2022-03-26:
Gibts eine offizielle Doku zur Api?

thecem, 2022-03-26:
Nein von Fronius noch nicht, wird aber irgendwann kommen. Die Parameter in der Wattpilot API entsprechen 1:1 der GO-E API V2 Docu.

Ich kann gerne Abzüge zur Verfügung stellen.

andig, 2022-03-26:
Kannst Du dann nicht heute schon 

    type: go-e

Benutzen?

thecem, 2022-03-26:
Nein leider:

```
[lp-1  ] ERROR 2022/03/26 15:50:41 charge meter: wrong token
[lp-1  ] ERROR 2022/03/26 15:50:41 charge rater: wrong token
[lp-1  ] ERROR 2022/03/26 15:50:41 charger: wrong token
```
hiermit geht's token ist richtig:

`https://31212121.api.v3.go-e.io/api/status?token=EA12121212121212121212121212121212`

andig, 2022-03-27:
Ich verstehe die Frage nicht. Ich verlege das in Discussions bis klarer wird, was hier überhaupt zu tun ist.