evcc-io/evcc #3010: UI doesn't show SoC with custom vehicle

morphZ, 2022-03-26:
### Describe the bug

Using a `vehicle` of `type: custom` the SoC is not shown in the UI.

[image]

The `evcc` cmd-tool yields the correct SoC:
```bash
$ docker-compose exec evcc evcc vehicle
[main  ] INFO 2022/03/26 13:33:33 evcc 0.87
[main  ] INFO 2022/03/26 13:33:33 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/03/26 13:33:33 connecting evcc-847302006 at tcp://mosquitto:1883
[mqtt  ] DEBUG 2022/03/26 13:33:33 tcp://mosquitto:1883 connected
SoC:      76%
Range:    335km
Capacity: 64kWh
```

### Steps to reproduce

1. Configure vehicle as described in configuration section below.
2. Start evcc
3. Open UI

### Configuration details

```yaml
...

vehicles:
  - type: custom
    name: niro
    title: Kia E-Niro
    capacity: 64 # kWh
    soc:
      source: mqtt
      topic: hass/sensor/e_niro_ev_battery/state
    range:
      source: mqtt
      topic: hass/sensor/e_niro_range_by_ev/state
...

```

### Log details

```log
evcc            | [main  ] INFO 2022/03/26 13:37:16 evcc 0.87
evcc            | [main  ] INFO 2022/03/26 13:37:16 using config file /etc/evcc.yaml
evcc            | [main  ] INFO 2022/03/26 13:37:16 listening at 0.0.0.0:7070
evcc            | [mqtt  ] INFO 2022/03/26 13:37:16 connecting evcc-2115103272 at tcp://mosquitto:1883
evcc            | [mqtt  ] DEBUG 2022/03/26 13:37:16 tcp://mosquitto:1883 connected
evcc            | [site  ] INFO 2022/03/26 13:37:17 site config:
evcc            | [site  ] INFO 2022/03/26 13:37:17   meters:      grid ✓ pv ✓ battery ✗
evcc            | [site  ] INFO 2022/03/26 13:37:17     grid:      power ✓ energy ✓ currents ✗
evcc            | [site  ] INFO 2022/03/26 13:37:17     pv 0:      power ✓ energy ✗ currents ✗
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17 loadpoint 1:
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17   mode:        pv
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17   charger:     power ✓ energy ✗ currents ✓ phases ✗
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17   meters:      charge ✓
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17     charge:    power ✓ energy ✗ currents ✓
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17   vehicles:    ✗
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:17 max charge current: 6A
evcc            | [site  ] DEBUG 2022/03/26 13:37:17 ----
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:17 charge power: 3160W
evcc            | [site  ] DEBUG 2022/03/26 13:37:17 pv power: 4140W
evcc            | [site  ] DEBUG 2022/03/26 13:37:17 grid power: 190W
evcc            | [site  ] DEBUG 2022/03/26 13:37:17 site power: 190W
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:17 charge currents: [4.9 5 4.9]A
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:17 charger status: C
evcc            | [lp-1  ] INFO 2022/03/26 13:37:17 start charging ->
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:17 pv charge current: 5.72A = 6A + -0.275A (190W @ 3p)
evcc            | [site  ] DEBUG 2022/03/26 13:37:27 ----
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 charge power: 3190W
evcc            | [site  ] DEBUG 2022/03/26 13:37:27 pv power: 4130W
evcc            | [site  ] DEBUG 2022/03/26 13:37:27 grid power: 220W
evcc            | [site  ] DEBUG 2022/03/26 13:37:27 site power: 220W
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 charge currents: [4.9 5 5]A
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 detected phases: 3p
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 charger status: C
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 pv charge current: 5.68A = 6A + -0.319A (220W @ 3p)
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 site power 220W >= 200W disable threshold
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 pv disable timer start: 10m0s
evcc            | [lp-1  ] DEBUG 2022/03/26 13:37:27 pv disable in 10m0s
```

### What type of operating system are you running?

Docker container

### Version

`evcc 0.87`

naltatis, 2022-03-26:
Das Fahrzeug ist nicht erkannt. Sonst würde über dem Balken der Titel "Kia E-Niro" und nicht "Vehicle" stehen.

morphZ, 2022-03-26:
Ok... Wie funktioniert denn die Erkennung? Muss ich noch das `status` Objekt mappen?

morphZ, 2022-03-26:
In meinem bescheidenen Setup gibt es nur ein Fahrzeug und eine Ladestation. Wenn also die Ladestation lädt kann man davon ausgehen, dass es der Kia ist...