evcc-io/evcc #3027: hems not working in home assistant docker mode (maybe all docker container?)

AnduriI, 2022-03-28:
### Describe the bug

I have a running evcc instance on my home assistant machine and one for test purpose and interactive usage (e.g. creating config.yaml). I tried to enable my sma hems.

### Steps to reproduce

starting evcc with and without hems enabled.

### Configuration details

hems:
  type: sma
#  AllowControl: false

### Log details

in the log in my home assistant I see the following:
```
starting evcc: 'evcc --config  /config/evcc.yaml'
[main  ] INFO 2022/03/28 13:05:30 evcc 0.87
[main  ] INFO 2022/03/28 13:05:30 using config file /config/evcc.yaml
[main  ] INFO 2022/03/28 13:05:30 listening at 0.0.0.0:7070
[main  ] FATAL 2022/03/28 13:05:34 failed configuring hems: creating device id: machineid: machineid: open /etc/machine-id: no such file or directory
```

### What type of operating system are you running?

Linux

### Version

both are running evcc version 0.87

premultiply, 2022-03-28:
We do not provide docker support, sorry.
https://docs.evcc.io/docs/installation/docker/#docker-installation

closed 2022-03-28 by premultiply

andig, 2022-03-28:
See https://github.com/denisbrodbeck/machineid/issues/5

AnduriI, 2022-03-28:
Thanks both of your for the clarification, now I understand the problem. As I'm using `https://github.com/evcc-io/hassio-addon/` could it be possible to mount the machine-id along with the config file?