evcc-io/evcc #3030: Handle SMA "NaN-Values" via Modbus plugin

premultiply, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
SMA uses special "NaN-Values" on their own SMA Modbus-Profile (Non-SunSpec) which are currently unhandled by evcc so they appear es very large numbers if a value is currently not available if the inverter is in standby at night for example.
This special handling is required for SMA Smart Energy Hybrid inverters as there is currently no other way to retrieve all requested data values by other supported protocols like Speedwire or SunSpec Profiles.

**Describe the solution you'd like**
Implement a way to mask or map these special values so that they will return as 0.
The NaN-Values are documented by SMA. See attached document.

**Describe alternatives you've considered**
Use external scripts or other solutions to poll and process the values but this is not compatible with evcc's concept of integrated templates.

**Additional context**
[image]
[721_sma_modbus-tb-de-13.pdf](https://github.com/evcc-io/evcc/files/8365585/721_sma_modbus-tb-de-13.pdf)

andig, 2022-03-29:
Wenn ich es richtig verstehe:
- trifft nur SMA, direkte Registerwerte, kein SunSPEC
- wir wollen die NaNs auf `0` mappen
- wir müssten das tun bevor intern `float64` Werte gebildet werden

Ich hab eine Idee :)

closed 2022-03-29 by andig

premultiply, 2022-04-09:
Ich überlege gerade ob man das nicht von XXXXsma auf XXXXnan umbenennen sollte.
Das taucht auch immer mal wieder bei anderen Herstellern/Geräten auf. Dann wäre es etwas generischer.

andig, 2022-04-09:
Yep 👍🏻  Dann auch überall wo jetzt schon verwendet

premultiply, 2022-04-09:
#3120