evcc-io/evcc #3040: Telegram: * '' has invalid keys: services

aBua-MUC, 2022-03-30:
### Describe the bug

Hallo zusammen,
nach Eintrag der Services für Telegram bekomme ich beim Start von evcc die Fehlermeldung *** '' has invalid keys: services**

### Steps to reproduce

Der Bot läuft. Via Node-Red gibt es keine Probleme. Zum Test habe ich den Token nochmal neu generiert. Funktioniert aber auch nicht. 

Kann jemand helfen? Nachfolgend der Ausug aus der evcc.yaml (token und Chat-Id sind beispielhaft anonymisiert)

### Configuration details

services:
- type: telegram
  token: 1234567890:AAAAAAAAAAAAAAAAAAAAAAA
  - -123456789 

### Log details

[main  ] INFO 2022/03/30 10:59:14 evcc 0.87
[main  ] INFO 2022/03/30 10:59:14 using config file "path"\evcc.yaml
[main  ] FATAL 2022/03/30 10:59:14 failed parsing config file "path"\evcc.yaml: 1 error(s) decoding:

* '' has invalid keys: services

### What type of operating system are you running?

Docker container

### Version

0.88

aBua-MUC, 2022-03-30:
War ein Fehler in der yaml, jetzt geht es :-)

closed 2022-03-30 by aBua-MUC

buliwyf42, 2022-06-26:
> War ein Fehler in der yaml, jetzt geht es :-)

Wo war denn bei dir genau der Fehler? Ich habe hier genau das gleiche Problem und schon jede Zeile dreimal neu geschrieben.