evcc-io/evcc #3061: "host" parameter standardisieren

andig, 2022-04-03:
Fast jedes Template enthält eine unterschiedlich falsche Definition von "host", z.B.:

      en: tapo device ip address (local)
      de: Die IP-Adresse der Tapo Steckdose im lokalen LAN/WLAN

1. Es genügt von device zu sprechen. Dass es ein "tapo" ist ist klar wenn man sich im Kontext dessen bewegt.
2. IP ist falsch da der hostname genauso funktioniert.

Vorschlag: generische Übersetzung für alle Geräte.

thierolm, 2022-04-03:
Vorschlag:

```
  - name: host
    ...
    help:
      en: device specific local host address (ip xxx.xxx.xxx.xxx or url http://xxx.xxx.xxx.xxx)
      de: Die Geräte-Adresse der Steckdose im lokalen LAN/WLAN (IP xxx.xxx.xxx.xxx oder Url http://xxx.xxx.xxx.xxx)
  - name: user
    ...
    help:
      en: device specific user-account (e.g. e-mail address, user-id, etc.)
      de: Benutzerkonto (bspw. E-Mail Adresse, User-Id, etc.)
  - name: password
    ...
    help:
      en: password of device specific user-account
      de: Password des Benutzerkontos
```

premultiply, 2022-04-03:
Das Protokollprefix (z. B. http://) gehört da IMHO nicht hin. Das ist zu technisch und wird ohnehin über die jeweilige Geräte-API festgelegt.
IP-Adresse oder DNS-Hostname sollten reichen.

andig, 2022-04-03:
Schema ist sogar explizit falsch. Das wird jeweils im Template ergänzt. Es ist genau keine Url sondern der Host- sei es als IP oder Dns Name.

thierolm, 2022-04-03:
> help:
>       en: device specific local host address (ip or dns-hostname)
>       de: Die Geräte-Adresse der Steckdose im lokalen LAN/WLAN (IP-Adresse oder DNS-Hostname)

So besser?

premultiply, 2022-04-03:
👍 Ich denke „dns-“ kann man sogar weglassen.

andig, 2022-04-03:
```
help:
  en: device address (IP or host name)
  de: Geräte-Adresse (IP-Adresse oder Hostname)
```

Kurz und schmerzlos und ohne unnötige Einschränkungen (warum nur lokales LAN?)

andig, 2022-04-03:
Das Ganze dann hier https://github.com/evcc-io/evcc/blob/master/templates/definition/defaults.yaml rein und aus allen Templates raus?

closed 2022-04-07 by andig