evcc-io/evcc #3079: Kostal Plenticore integration bricht mit Fehler ab

binderth, 2022-04-04:
### Describe the bug

Ich habe einen Kostal Plenticore Plus 10 inkl. BYD Batterie.
Ich verwende die Konfiguration unter: https://docs.evcc.io/docs/devices/meters#plenticore-hybrid

also z.B.:
```
  - name: my_pv
    type: template
    template: kostal-plenticore
    usage: pv
    id: 71
    # Modbus: TCPIP
    uri: 192.0.2.2:1502 # IP-Adresse oder Hostname: Port 
```

Nach dem Start von evcc (als Docker-Image) erhalte ich den folgenden Fehler:
```
[main  ] FATAL 2022/04/04 20:15:10 cannot create meter 'my_pv': cannot create meter 'template': cannot create meter 'custom': power: add[0]: invalid modbus configuration: need either uri or device 
```

### Steps to reproduce

1. EVCC Docker Image runterladen
2. Verzeichnis /etc anlegen und verbinden
3. evcc.yaml mit dem Vorschlag aus den Docs füllen (IP-Adresse anpassen)
4. Docker starten

### Configuration details

Standard-YAML (mit Entfernungen) und nur dem Kostal:

```uri: 192.168.1.18:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: info
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
  - name: my_pv
    type: template
    template: kostal-plenticore
    usage: pv
    id: 71
    # Modbus: TCPIP
    uri: 192.168.1.13:1502 # IP-Adresse oder Hostname: Port 
    
    
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
- name: my_charger
  type: template
  template: tinkerforge-warp
  host: 192.168.1.9 # Die IP Adresse oder der Hostname des MQTT Brokers
  port: 1883 # Der Port des MQTT Brokers # Optional
  topic: warp2/pl12 # Optional
  timeout: 30s # Akzeptiere keine Daten die älter als dieser Wert sind # Optional 
    
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
vehicles:
  - name: car1
    type: renault
    title: Zoe
    capacity: 60 # kWh
    user: myuser # user
    password: mypassword # password
    vin: WREN...
    onIdentify: # set defaults when vehicle is identified
      minSoC: 20 # charge to at least 20% independent of charge mode
      targetSoC: 90 # limit charge to 90%

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    pvs:
      - pv # list of pv inverters/ meters
    battery: battery # battery meter
  prioritySoC: # give home battery priority up to this soc (empty to disable)
  bufferSoC: # ignore home battery discharge above soc (empty to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: wallbe # charger
    meter: charge # charge meter
    vehicle: audi
    # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
    # - ID.3
    # - e-Up
    mode: pv
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
      target: 100 # always charge to 100%
      estimate: false # set true to interpolate between api updates
    phases: 3 # ev phases (default 3)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.294 # EUR/kWh

    # # or variable via tibber
    # type: tibber
    # cheap: 0.2 # EUR/kWh
    # token: "476c477d8a039529478ebd690d35ddd80e3308ffc49b59c65b142321aee963a4" # access token
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # cheap: 0.2 # EUR/kWh
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: SoC updated
      msg: Battery charged to ${vehicleSoC:%.0f}%
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

2022-04-04 18:15:10 | stdout | [main  ] FATAL 2022/04/04 20:15:10 cannot create meter 'my_pv': cannot create meter 'template': cannot create meter 'custom': power: add[0]: invalid modbus configuration: need either uri or device
-- | -- | --
2022-04-04 18:15:10 | stdout | [main  ] INFO 2022/04/04 20:15:10 listening at 192.168.78.18:7070
2022-04-04 18:15:10 | stdout | [main  ] INFO 2022/04/04 20:15:10 using config file /etc/evcc.yaml
2022-04-04 18:15:10 | stdout | [main  ] INFO 2022/04/04 20:15:10 evcc 0.89

### What type of operating system are you running?

Docker container

### Version

latest -> 0.89

premultiply, 2022-04-04:
Probier mal bitte `host:` statt `uri:`

/cc @naltatis Hast du eine Idee wo hier das uri herkommt?

binderth, 2022-04-04:
Das ging ja fix! ;) 
leider nicht erfolgreich:
```
[main  ] FATAL 2022/04/04 20:44:51 cannot create meter 'my_pv': cannot create meter 'template': cannot create meter 'custom': power: add[0]: invalid modbus configuration: need either uri or device
```

der Abschnitt meters:
```
meters:
  - name: my_pv
    type: template
    template: kostal-plenticore
    usage: pv
    id: 71
    # Modbus: TCPIP
    host: 192.168.1.13:1502 # IP-Adresse oder Hostname: Port 

```

andig, 2022-04-04:
Dein host ist kein host. Nimm bitte

    evcc configure

dann passt acuh das Ergebnis.