evcc-io/evcc #3087: Fresh install: modbus exception

emcepe, 2022-04-05:
### Describe the bug

I'm new to evcc and just started to install latest stable version.

testing fails when selecting "VARTA Energiespeicher" and port 502 with:

`Wähle eines der folgenden PV Komplettsysteme aus, oder 'Mein Gerät ist nicht in der Liste' falls keines dieser Geräte vorhanden ist`

` Wähle ein PV System: VARTA Energiespeicher`

`Führe folgende Einstellungen durch:`

` IP-Adresse oder Hostname 192.168.189.45`
` Port 502`

`Teste die grid Konfiguration von VARTA Energiespeicher ...`

`Teste die pv Konfiguration von VARTA Energiespeicher ...`
`   Fehler: read failed: modbus: exception '2' (illegal data address), function '4'`

### Steps to reproduce

- evcc configure
- Standard Modus
- VARTA Energiespeicher
- 192.168.189.45
- 502

### Configuration details

not yet present

### Log details

[main  ] INFO 2022/04/05 08:55:25 evcc 0.89
[main  ] ERROR 2022/04/05 08:55:25 missing evcc config - switching into demo mode
[main  ] INFO 2022/04/05 08:55:25 listening at 0.0.0.0:7070
[site  ] INFO 2022/04/05 08:55:25 site config:
[site  ] INFO 2022/04/05 08:55:25   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/04/05 08:55:25     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/05 08:55:25     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/05 08:55:25     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/04/05 08:55:25 loadpoint 1:
[lp-1  ] INFO 2022/04/05 08:55:25   mode:        off
[lp-1  ] INFO 2022/04/05 08:55:25   charger:     power ✗ energy ✗ currents ✗ phases ✗
[lp-1  ] INFO 2022/04/05 08:55:25   meters:      charge ✓
[lp-1  ] INFO 2022/04/05 08:55:25     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/04/05 08:55:25   vehicles:    ✓
[lp-1  ] INFO 2022/04/05 08:55:25     vehicle 0: range ✗ finish ✗ status ✗ climate ✗
[lp-1  ] INFO 2022/04/05 08:55:25 vehicle updated: unknown -> e-Golf

### What type of operating system are you running?

Linux

### Version

evcc version 0.89

andig, 2022-04-05:
Das Template funktioniert nicht mit Deinem Gerät. Was für ein Speicher ist das genau?

/cc @premultiply

emcepe, 2022-04-05:
sorry, ich dachte ich hätte das geschrieben.

Varta Element 12/S4 von Mai 2021

andig, 2022-04-06:
> Varta Element 12/S4

Es scheint, als würden die Register von den uns bekannten abweichen. Ich konnte leider keine Modbus Doku finden- könntest Du die von Varta besorgen?

andig, 2022-04-06:
@emcepe bei genauerem drüber nachdenken könnte es auch ein einfacher Implementierungsfehler bei uns sein. Könntest Du Remotezugang zum Modbus Port an info@evcc.io schicken?

andig, 2022-04-06:
> Könntest Du Remotezugang zum Modbus Port an [info@evcc.io](mailto:info@evcc.io) schicken?

Oder alternativ den PR ausprobieren?

andig, 2022-04-07:
ping @emcepe

andig, 2022-04-09:
ping @emcepe

> Könntest Du Remotezugang zum Modbus Port an [info@evcc.io](mailto:info@evcc.io) schicken?

Laut https://github.com/evcc-io/evcc/discussions/2475#discussioncomment-2532895 funktioniert Varta Element 6 S4, daher ist eigentlich nicht einzusehen warum 12 S4 bei Dir nicht funktionieren sollte.

andig, 2022-04-29:
Ich kanns nachvollziehen. Aus der Doku:

> <div class="page" title="Page 2" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none;">

  |   |   |   |   |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
1102 | PV-sensor power | R | UINT16 | 1 | W | current AC production power measured by VARTA ́s PV-sensor |   |   |   |   |   | 
</div>

Diesen Sensor hat nur die "Varta pulse".

@premultiply hast Du noch eine Idee, woher PV Erzeugung kommen könnte?

premultiply, 2022-04-29:
Man muss den Varta-Speicher doch nicht als PV-Meter nehmen, oder?
Den/die PV-WR kann man doch auch direkt anfragen?

closed 2022-04-30 by andig

andig, 2022-04-30:
@emcepe da das bei Dir nicht geht haben wir einen zweiten Typ VARTA hinzugefügt: https://docs.evcc.io/docs/devices/meters/#energiespeicher-kein-pv-sensor

emcepe, 2022-05-04:
ok vielen Dank. Damit läuft das Setup nun einwandfrei durch.

Eine kurze Frage noch: gibt es irgendwo noch einen Test-Token der nicht abgelaufen ist, damit ich das Zusammenspiel mit der openWB vorab testen kann bevor ich aktiver Sponsor werde?

Danke.

andig, 2022-05-04:
Derzeit nicht- wir wissen aber, dass es funktioniert ;)

emcepe, 2022-05-10:
Ok, das glaube ich gerne :)

Dann lass es mich anders formulieren:

Warum braucht man evcc, wenn man eine openWB hat, die ebenso das ganze Lademanagement intus hat? Ich hab's leider noch nicht geblickt, welche Vorteile evcc + openWB Token hätte im Vergleich zu openWB alleine?

Deshalb wollte ich das eigentlich selber testen um mir diese Frage zu beantworten 😃

Dann werde ich halt die 2 € bezahlen und schauen, ob sich das für mich lohnt bzw. warum man openWB + evcc nutzen sollte.

Vielleicht gibt's aber irgendwo ein Wiki Eintrag oder so, wo das erklärt wird?

Danke.

andig, 2022-05-10:
Sorry, aber das hat nichts mit dem Varta Speicher zu tun. Bitte neues Thema!