evcc-io/evcc #3089: KIA EV6 wird nicht mit bluelinky erkannt

binderth, 2022-04-05:
### Describe the bug

Ich habe meinen KIA EV6 mit `evcc configure` eingerichtet. 
Beim anstecken des Fahrzeugs an die ebenfalls konfigurierte Wallbox (Warp2 Pro) wurde das entsprechend von evcc erkannt - allerdings mit der Fehlermeldung quittiert:

```
vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: failed to download stamps: Get \"https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/kia-e7bcd186-a5fd-410d-92cb-6876a42288bd.v2.json\": x509: certificate signed by unknown authority 
``` 

### Steps to reproduce

1. Konfiguriere KIA EV6 (wurde innerhalb `evcc configure` als korrekt erkannt)
2. Stecke EV6 an Wallbox
3. Fehlermeldung erscheint

### Configuration details

Konfiguration des EV6:
```
vehicles:
- type: template
  template: kia 
  title: Kia EV6  
  user: xxx  
  password: yyy
  vin: zzz 
  capacity: 77  
  phases: 3  
  language: de  
  mode: pv  
  minSoC: 33  
  targetSoC: 80  
  minCurrent: 6  
  maxCurrent: 16  
  cloud: false  
  name: KiaEV6
```

### Log details

Im Logfile erscheint folgendes:

2022-04-05 10:37:14 | stdout | [lp-1  ] ERROR 2022/04/05 12:37:14 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: failed to download stamps: Get "https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/kia-e7bcd186-a5fd-410d-92cb-6876a42288bd.v2.json": x509: certificate signed by unknown authority
-- | -- | --
2022-04-05 10:37:14 | stdout | [lp-1  ] INFO 2022/04/05 12:37:14 car connected
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53 vehicle updated: unknown -> unavailable
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53     vehicle 0: range ✗ finish ✗ status ✗ climate ✗
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53   vehicles:    ✓
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53     charge:    power ✓ energy ✓ currents ✓
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53   meters:      charge ✓
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53   charger:     power ✓ energy ✓ currents ✓ phases ✗
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53   mode:        pv
2022-04-05 08:52:53 | stdout | [lp-1  ] INFO 2022/04/05 10:52:53 loadpoint 1:
2022-04-05 08:52:53 | stdout | [site  ] INFO 2022/04/05 10:52:53     battery 0: power ✓ energy ✗ currents ✗ soc ✓
2022-04-05 08:52:53 | stdout | [site  ] INFO 2022/04/05 10:52:53     pv 0:      power ✓ energy ✗ currents ✗
2022-04-05 08:52:53 | stdout | [site  ] INFO 2022/04/05 10:52:53     grid:      power ✓ energy ✗ currents ✗
2022-04-05 08:52:53 | stdout | [site  ] INFO 2022/04/05 10:52:53   meters:      grid ✓ pv ✓ battery ✓
2022-04-05 08:52:53 | stdout | [site  ] INFO 2022/04/05 10:52:53 site config:
2022-04-05 08:52:51 | stdout | [warp  ] INFO 2022/04/05 10:52:51 connecting evcc-94092800 at tcp://192.168.78.9:1883
2022-04-05 08:52:50 | stdout | [main  ] INFO 2022/04/05 10:52:50 listening at 0.0.0.0:7070
2022-04-05 08:52:50 | stdout | [main  ] INFO 2022/04/05 10:52:50 using config file /etc/evcc.yaml
2022-04-05 08:52:50 | stdout | [main  ] INFO 2022/04/05 10:52:50 evcc 0.89

### What type of operating system are you running?

Docker container

### Version

0.89

goebelmeier, 2022-04-05:
```
❯ openssl s_client -connect raw.githubusercontent.com:443
CONNECTED(00000005)
depth=2 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Global Root CA
verify return:1
depth=1 C = US, O = DigiCert Inc, CN = DigiCert TLS RSA SHA256 2020 CA1
verify return:1
depth=0 C = US, ST = California, L = San Francisco, O = "GitHub, Inc.", CN = *.githubassets.com
verify return:1
---
Certificate chain
 0 s:/C=US/ST=California/L=San Francisco/O=GitHub, Inc./CN=*.githubassets.com
   i:/C=US/O=DigiCert Inc/CN=DigiCert TLS RSA SHA256 2020 CA1
 1 s:/C=US/O=DigiCert Inc/CN=DigiCert TLS RSA SHA256 2020 CA1
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert Global Root CA
---
```

Das DigiCert Global Root CA Certificate sollte doch eigentlich in dem verwendeten Docker Image enthalten sein: https://pkgs.alpinelinux.org/contents?file=DigiCert_Global_Root_CA.crt&path=&name=ca-certificates&branch=v3.15&repo=main&arch=aarch64

Passiert das auch, wenn du die evcc Binary lokal mit der gleichen Config ausführst und nicht als Container?

binderth, 2022-04-05:
> Passiert das auch, wenn du die evcc Binary lokal mit der gleichen Config ausführst und nicht als Container?

Ich habs grad versucht - dort läuft das, habe aber nen anderen Fehler... :

```
evcc --log debug
[main  ] INFO 2022/04/05 13:35:35 evcc 0.89
[main  ] INFO 2022/04/05 13:35:35 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/04/05 13:35:35 listening at 0.0.0.0:7070
[warp  ] INFO 2022/04/05 13:35:36 connecting evcc-842664911 at tcp://192.168.78.9:1883
[warp  ] DEBUG 2022/04/05 13:35:36 tcp://192.168.78.9:1883 connected
[warp  ] DEBUG 2022/04/05 13:35:36 tcp://192.168.78.9:1883 subscribe warp2/pl12/evse/state
[warp  ] DEBUG 2022/04/05 13:35:36 warp2/pl12/meter/state wait for initial value
[warp  ] DEBUG 2022/04/05 13:35:36 warp2/pl12/meter/detailed_values wait for initial value
[site  ] INFO 2022/04/05 13:35:38 site config:
[site  ] INFO 2022/04/05 13:35:38   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/04/05 13:35:38     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/05 13:35:38     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/05 13:35:38     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/04/05 13:35:38 loadpoint 1:
[lp-1  ] INFO 2022/04/05 13:35:38   mode:        pv
[lp-1  ] INFO 2022/04/05 13:35:38   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/04/05 13:35:38   meters:      charge ✓
[lp-1  ] INFO 2022/04/05 13:35:38     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/05 13:35:38   vehicles:    ✓
[lp-1  ] INFO 2022/04/05 13:35:38     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/04/05 13:35:38 vehicle updated: unknown -> Kia EV6
[lp-1  ] DEBUG 2022/04/05 13:35:38 set charge mode: pv
[lp-1  ] DEBUG 2022/04/05 13:35:38 set min current: 6
[lp-1  ] DEBUG 2022/04/05 13:35:38 set max current: 16
[lp-1  ] DEBUG 2022/04/05 13:35:38 set min soc: 33
[lp-1  ] DEBUG 2022/04/05 13:35:38 set target soc: 80
[lp-1  ] DEBUG 2022/04/05 13:35:40 max charge current: 6A
[site  ] DEBUG 2022/04/05 13:35:40 ----
[lp-1  ] DEBUG 2022/04/05 13:35:40 charge power: 0W
[server] FATAL 2022/04/05 13:35:40 remove /tmp/evcc: operation not permitted
```

goebelmeier, 2022-04-05:
OK, dann finde ich zum Einen spannend, dass wirklich im Container das CA Certificate der Root CA für https://raw.githubusercontent.com fehlt. Das kann ich mir nicht erklären. Und hast du ne Idee woher das `/tmp/evcc` stammt? hast du das in deiner Config irgendwo definiert?

binderth, 2022-04-05:
nein, die `/tmp/evcc` ist offenbar ein Socket, der regelmässig mit Start von evcc (aktuell um 14:38 einen restart veranlasst) angelegt wird:
```
ls -la /tmp/
total 40
drwxrwxrwt  9 root root 4096 Apr  5 14:39 .
drwxr-xr-x 18 root root 4096 Jan 28 02:22 ..
srwxr-xr-x  1 root root    0 Apr  5 14:39 evcc
-rw-r--r--  1 pi   pi   1024 Apr  5 14:38 .evcc.swp
...
```

goebelmeier, 2022-04-05:
Aber warum als `root:root` und nicht als `pi:pi`? Any suggestions @naltatis oder @andig ? Und habt ihr eine Idee für das Root CA Problem?

binderth, 2022-04-05:
> Ah, hast du ggf noch den Docker Container laufen? Hat der das `/tmp/evcc` eventuell als `root` angelegt und deshalb hat der manuelle Prozess der als `pi` läuft darauf keine Rechte?

Der Docker läuft auf der Synology - und ich hab einfach nen freien Raspi genommen und dort evcc installiert. d.h. die beiden haben nix miteinander zu tun. 
...und wenn ich den evcc stoppe, verschwindet der Socket auch wieder. Nur die `.evcc.swp` bleibt stehen.

andig, 2022-04-05:
Bitte erstmal ohne Docker nachvollziehen. Wir möchten hier keinen Docker Support geben.

closed 2022-04-05 by andig

binderth, 2022-04-05:
> These: Du hast `/etc` gemounted was eigentlich immer falsch ist...

das ist richtig, `/etc` verweist auf ein Verzeichnis im host. Mir ist leider nicht klar, wie ich ohne Bash sonst die evcc.yaml in Docker kriege?

Ansonsten haben wir ja herausgefunden, dass offenbar das Zertifikat nicht im Docker verfügbar ist, wenn ich das richtig verstanden habe - siehe: https://github.com/evcc-io/evcc/issues/3089#issuecomment-1088589093:

> Das DigiCert Global Root CA Certificate sollte doch eigentlich in dem verwendeten Docker Image enthalten sein: https://pkgs.alpinelinux.org/contents?file=DigiCert_Global_Root_CA.crt&path=&name=ca-certificates&branch=v3.15&repo=main&arch=aarch64

andig, 2022-04-05:
Das Zertifikat ist weg weil Du den Ordner platt gemacht hast. Du darfst `/etc` nicht mounten!!!

goebelmeier, 2022-04-05:
Was Andi sagen möchte: du musst /etc/evcc.yaml mounten, NICHT das gesamte /etc

Siehe https://docs.evcc.io/docs/installation/docker#docker-installation

sudo docker run \
-d \
--name evcc \
-v $(pwd)/evcc.yaml:/etc/evcc.yaml \
-p 7070:7070 \
andig/evcc:latest

binderth, 2022-04-06:
Ah. alles klar, den Link hatte ich auch noch nicht auf dem Schirm. Damit geht's jetzt natürlich! Super!