evcc-io/evcc #3107: Login Problem Volvo

pa-pa, 2022-04-07:
### Describe the bug

Hallo.

Ich versuche gerade einen Volvo mit dem entsprechenden Modul einzubinden. Leider klappt die Anmeldung nicht.

Meine Vermutung ist, dass es am Usernamen liegt. Dieser beginnt mit einem +. Die Credential sind in Ordnung, ich verwende sie in der App und OpenWB. Eine Abfrage mit CURL geht ebenfalls. Nur evcc will leider nicht.

### Steps to reproduce

Fahrzeugabfrage starten

### Configuration details

vehicles:
  - name: XC60
    type: template
    template: volvo
    user: +1234567890
    password: PASSWORD
    capacity: 11
    vin: XXXXXXXXXXXXXXXXXXXX

### Log details

`pi@garage:/opt/evcc $ ./evcc -c config.yaml --log trace vehicle
[main  ] INFO 2022/04/07 06:20:49 evcc 0.89
[main  ] INFO 2022/04/07 06:20:49 using config file config.yaml
[volvo ] TRACE 2022/04/07 06:20:50 GET https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customeraccounts
[volvo ] TRACE 2022/04/07 06:20:51 {"errorLabel":"Unauthorized","errorDescription":"Invalid credentials"}
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'volvo': cannot get vehicles: unexpected status: 401: Unauthorized: Invalid credentials`

### What type of operating system are you running?

Linux

### Version

evcc version 0.89

closed 2022-04-07 by andig

pa-pa, 2022-04-07:
Weder einfache noch doppelte Anführungszeichen ändern etwas am Verhalten

andig, 2022-04-07:
Gerne mal Zugangsdaten an info@evcc.io