evcc-io/evcc #3113: Timer für Wechsel von 3 auf 1 Phase lläuft obwohl garnicht geladen wird

Robbe64, 2022-04-08:
### Describe the bug

Bin nicht sicher ob es ein Bug oder ein Feature-Request ist. Auf jeden Fall kein ernsthaftes Problem:

Wenn EVCC noch im 3-Phasen modus ist, das Auto aktuell nicht geladen wird, da zu wenig PV-Leistung verfügbar ist und die PV-Leistung nun nachdem nicht geladen wurde für 1 Phasiges laden reicht, dann läuft zunächst der Timer wegen des Wechselns der Phasen obwohl aktuell ja garnicht geladen wird.

### Steps to reproduce

1. Kein Auto anschließen
2. EVCC neu starten
3. PV-Leistung reicht für 1-Phasiges laden
4. Auto anschließen
=> Timer für Wechsel von 3-Phasen auf 1-Phasen laden läuft ehe die Ladung beginnt

### Configuration details

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/8449412/evcc.yaml.txt)

### Log details

```
[lp-1  ] DEBUG 2022/04/08 08:48:37 pv charge current: 1.53A = 0A + 1.53A (-1056W @ 3p),
[lp-1  ] DEBUG 2022/04/08 08:48:37 phase scale1p in 6m40s,
[lp-1  ] DEBUG 2022/04/08 08:48:37 available power 1056W < 4140W min 3p threshold,
[lp-1  ] ERROR 2022/04/08 08:48:37 vehicle status: 408 Request Timeout,
[lp-1  ] DEBUG 2022/04/08 08:48:37 charger status: B,
[lp-1  ] DEBUG 2022/04/08 08:48:37 charge currents: [0.008 0.007 0.008]A,
[site  ] DEBUG 2022/04/08 08:48:37 site power: -1056W,
[site  ] DEBUG 2022/04/08 08:48:37 battery soc 0: 60%,
[site  ] DEBUG 2022/04/08 08:48:37 grid power: 40W,
[site  ] DEBUG 2022/04/08 08:48:37 battery power: -1196W,
[site  ] DEBUG 2022/04/08 08:48:37 pv power: 1469W,
[lp-1  ] DEBUG 2022/04/08 08:48:37 charge power: 0W
```

### What type of operating system are you running?

Docker container

### Version

v0.89 [167772df]

andig, 2022-04-08:
/cc @premultiply

andig, 2022-04-08:
> Wenn EVCC noch im 3-Phasen modus ist, das Auto aktuell nicht geladen wird, da zu wenig PV-Leistung verfügbar ist und die PV-Leistung nun nachdem nicht geladen wurde für 1 Phasiges laden reicht, dann läuft zunächst der Timer wegen des Wechselns der Phasen obwohl aktuell ja garnicht geladen wird.

@Robbe64 Die Frage hier ist: sollte der Phasentimer laufen solange bei angestecktem Auto nicht geladen wird? Richtig?

Robbe64, 2022-04-08:
Ja, richtig.
Aus meiner Sicht sollte der Phasentimer nicht laufen. Das verzögert, je nach Einstellung, unnötig den Ladevorgang. In der Oberfläche ist der Phasentimer in diesem Fall auch nicht ersichtlich wenn ich mich jetzt nicht irre.

andig, 2022-04-09:
Du kannst den Timer ja ändern. Je kürzer desto zappeliger die Regelung aufgrund fehlender Hysterese. Ich sehe nicht was hier geändert werden sollte. Wenn Du es präziser machen kannst in einem Satz gerne nochmal versuchen.

andig, 2022-04-10:
@Robbe64 ich würde dann zu machen?

premultiply, 2022-04-10:
Gemeint sein könnte das der Phasentimer immer sofort expired wenn ohnehin nicht geladen wird.
Bzw. aus einer Ladeunterbrechnung sofort mit dem richtigen Phasenmodus gestartet wird (nach ablauf des PV-Timers).

andig, 2022-04-10:
Die Gründe für nicht-laden können ja divers sein, z.B. nicht genug PV. Also muss auch ein Timer laufen.

Robbe64, 2022-04-10:
Es ist genau so gemeint wie premultiply es beschrieben hat.

closed 2022-04-10 by Robbe64