evcc-io/evcc #3117: fataler Fehler: /tmp/evcc kann nicht entfernt werden

binderth, 2022-04-08:
### Describe the bug

Beim Starten von evcc erscheint eine fataler Fehler:
```
remove /tmp/evcc: operation not permitted
```

### Steps to reproduce

1. installiere evcc
2. konfiguration wie unten
3. starten von evcc

### Configuration details

ich habe inzwischen evcc.io auf dem Raspberry Pi installiert, auf dem auch openHAB3 läuft. Ich frage dort per http-API die entsprechenden Werte ab, was auch soweit funktioniert hatte, bis ich irgendwas übersehen hatte?

<details>
<summary>evcc.yaml:</summary>

```
#uri: 0.0.0.0:7070 # uri for ui
uri: localhost:7070 # uri for ui
interval: 10s # control cycle interval

log: info
#log: debug

meters:
- type: custom
  name: ErzeugungsLeistung
  power:
    source: http
    uri: http://localhost:8080/rest/items/PVErzeugungsLeistung
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    #scale: 0.001 # floating point factor applied to result, e.g. for kW to W conversion
- type: custom
  name: Netzeinspeisung
  power:
    source: http
    uri: http://localhost:8080/rest/items/Netzeinspeisung
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    #scale: 0.001 # floating point factor applied to result, e.g. for kW to W conversion
    scale: -1
- type: custom
  name: LadeleistungHaus
  power:
    source: http
    uri: http://localhost:8080/rest/items/LadeleistungHaus
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    scale: -1
  soc:
    source: http
    uri: http://localhost:8080/rest/items/SoCHaus
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]

chargers:
- type: template
  template: tinkerforge-warp-pro
  host: 192.168.78.9
  port: 1883
  topic: warp2/pl12
  timeout: 30s
  name: WarpWallbox

vehicles:
- type: template
  template: kia
  title: Kia EV6
  user: xxx
  password: yyy
  vin: zzz
  capacity: 77
  phases: 3
  language: de
  mode: pv
  minSoC: 33
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
  cloud: false  
  name: KiaEV6

loadpoints:
- title: Einfahrt
  charger: WarpWallbox
  vehicles:
  - KiaEV6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.319 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.067 # [currency]/kWh

mqtt:
  broker: 192.168.78.9:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # user:
  # password:

site:
  title: Paul-Lincke
  meters:
    grid: Netzeinspeisung
    pvs:
    - ErzeugungsLeistung
    batteries:
    - LadeleistungHaus
  bufferSOC: 51		# Hausbatterie wird oberhalb SoC 50% als Puffer genutzt
  prioritySOC: 50	# Hausbatterie bekommt bis zum SoC 50% Priorität beim laden
  residualPower: 100
```
</details>

### Log details

im Logfile wird dann das entfernen von `/tmp/evcc` moniert:

<details>
<summary>evcc --log debug</summary>

```
evcc --log debug
[main  ] INFO 2022/04/08 19:58:50 evcc 0.89
[main  ] INFO 2022/04/08 19:58:50 using config file /etc/evcc.yaml
[main  ] INFO 2022/04/08 19:58:50 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/04/08 19:58:50 connecting evcc-771363082 at tcp://192.168.78.9:1883
[mqtt  ] DEBUG 2022/04/08 19:58:50 tcp://192.168.78.9:1883 connected
[warp  ] DEBUG 2022/04/08 19:58:50 warp2/pl12/meter/state wait for initial value
[warp  ] DEBUG 2022/04/08 19:58:50 warp2/pl12/meter/detailed_values wait for initial value
[site  ] INFO 2022/04/08 19:58:52 site config:
[site  ] INFO 2022/04/08 19:58:52   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/04/08 19:58:52     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/08 19:58:52     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/08 19:58:52     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/04/08 19:58:52 loadpoint 1:
[lp-1  ] INFO 2022/04/08 19:58:52   mode:        pv
[lp-1  ] INFO 2022/04/08 19:58:52   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/04/08 19:58:52   meters:      charge ✓
[lp-1  ] INFO 2022/04/08 19:58:52     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/08 19:58:52   vehicles:    ✓
[lp-1  ] INFO 2022/04/08 19:58:52     vehicle 0: range ✗ finish ✗ status ✗ climate ✗
[lp-1  ] INFO 2022/04/08 19:58:52 vehicle updated: unknown -> unavailable
[site  ] DEBUG 2022/04/08 19:58:52 ----
[lp-1  ] DEBUG 2022/04/08 19:58:52 charge power: 0W
[server] FATAL 2022/04/08 19:58:52 remove /tmp/evcc: operation not permitted
[main  ] FATAL 2022/04/08 19:58:52 listen tcp 0.0.0.0:7070: bind: address already in use
```
</details>

### What type of operating system are you running?

Linux

### Version

0.89

binderth, 2022-04-08:
Meta-Informationen:

1. alle URLs zu openHAB funktionieren (per curl auf der console getestet)
2. allerdings habe ich eine Rückmeldung zur Kia -API erhalten:
```
{
  "retCode": "F",
  "resCode": "5091",
  "resMsg": "Exceeds number of requests - Exceeds Number of Requests.",
  "msgId": "xxx"
}
```

Kann es sein, dass der "Du hast es zu oft versucht und bist am Rate-Limit der API angekommen"-Fehler dieses Verhalten produziert?

thierolm, 2022-04-08:
Normalerweise legt evcc in `/tmp` keine Dateien ab.

Kannst du mal bitte auf der Kommandozeile folgende Befehle absetzen und den Output posten:

```bash
ls -lad /tmp
````
```bash
ls -la /tmp/evcc
````
```bash
ls -la `which evcc`
````
```bash
ls -la /etc/systemd/system/evcc*
````
```bash
cat /etc/systemd/system/evcc.service
````
```bash
ps -ef | grep evcc
```

binderth, 2022-04-08:
```
ls -lad /tmp
drwxrwxrwt 10 root root 4096  8. Apr 22:13 /tmp

ls -la /tmp/evcc
srwxr-xr-x 1 root root 0  8. Apr 21:09 /tmp/evcc

ls -la `which evcc`
-rwxr-xr-x 1 root root 37814272  3. Apr 12:27 /usr/bin/evcc

ls -la /etc/systemd/system/evcc*
ls: Zugriff auf '/etc/systemd/system/evcc*' nicht möglich: Datei oder Verzeichnis nicht gefunden

cat /etc/systemd/system/evcc.service
cat: /etc/systemd/system/evcc.service: Datei oder Verzeichnis nicht gefunden

ps -ef | grep evcc
root      3114     1  0 21:09 ?        00:00:26 /usr/bin/evcc
root      3491   903  0 21:12 pts/0    00:00:00 sudo nano /etc/evcc.yaml
root      3492  3491  0 21:12 pts/0    00:00:00 nano /etc/evcc.yaml
admin     7025  6501  0 22:15 pts/2    00:00:00 grep --color=auto evcc
```

Wobei es bei meiner Installation dort liegt:
```

cat /etc/systemd/system/multi-user.target.wants/evcc.service
# evcc.service
#

[Unit]
Description=evcc
Requires=network-online.target
After=syslog.target network.target network-online.target
Wants=network-online.target
StartLimitIntervalSec=10
StartLimitBurst=10

[Service]
ExecStart=/usr/bin/evcc
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target
admin@smarthouse:~ $
```
Und
```
ls -la /etc/systemd/system/multi-user.target.wants/evcc*      lrwxrwxrwx 1 root root 32  8. Apr 09:05 /etc/systemd/system/multi-user.target.wants/evcc.service -> /lib/systemd/system/evcc.service
```

binderth, 2022-04-09:
Ich habe EVCC deinstalliert und erneut installiert und jetzt klappt es...?
Lag offenbar nicht am API-Rate-Limit.

thierolm, 2022-04-09:
Top, ich würde auch /tmp/evcc löschen!