evcc-io/evcc #3127: Vehicle Audi Q4 eTron doesn't work

Beninohl, 2022-04-09:
### Describe the bug

Hallo
Ich habe EVCC 089 unter Linux auf einem Raspberry für einen Audi Q4 mit installiert, laden funktioniert.
Der Test mit dem Befehl "evcc vehicle" funktioniert einwandfrei. Der Befehl "evcc" liefert einen fehlerhaften Status.
Auf der Weboberfläche ist der eTron  nicht ersichtlich.

Fehler im Webbrowser:
Meldungen:
vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': Get \"https://identity.vwgroup.io/oidc/v1/authorize?client_id=f4d0934f-32bf-4ce4-b3c4-699a7049ad26%40apps_vw-dilab_com&code_challenge=59br5sT7hoC-6_hRIq6GBwyS8y1e-pneU6uaJ6Ej9qM&code_challenge_method=S256&prompt=login&redirect_uri=myaudi%3A%2F%2F%2F&response_type=code+id_token+token&scope=openid+profile+mbb&ui_locales=de-DE&nonce=Y7YHFQEu0VfTgZ7DhaiAXkdO1JlF6EPEuJ0okRlxNRO&state=f350a0f4-d667-4961-ba51-331218bd4fd6\": dial tcp: lookup identity.vwgroup.io on [2a02:1210:22ce:ec00:a2b5:49ff:fe25:4d40]:53: dial udp [2a02:1210:22ce:ec00:a2b5:49ff:fe25:4d40]:53: connect: network is unreachable
4

### Steps to reproduce

Command: evcc vehicle

Command: evcc

### Configuration details

pi@raspberrypi:~ $ more evcc.yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: solarlog
  usage: grid
  host: 192.168.1.61
  port: 502
  name: grid1
- type: template
  template: solarlog
  usage: pv
  host: 192.168.1.61
  port: 502
  name: pv2

chargers:
- type: template
  template: go-e-v3
  host: 192.168.1.141
  name: wallbox4

vehicles:
- type: template
  template: etron
  title: Beni eTron
  user: xxx.xxx@xxx.xx
  password: 'xxxxxxxxx'
  vin: xxxxx...
  phases: 3
  mode: off
  minSoC: 20
  targetSoC: 80
 minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: audi

loadpoints:
- title: Garage
  charger: wallbox4
  vehicles:
  - audi
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Nohl
  meters:
    grid: grid1
    pvs:
    - pv2
pi@raspberrypi:~ $

### Log details

pi@raspberrypi:~ $ evcc --log debug
[main  ] INFO 2022/04/09 21:28:24 evcc 0.89
[main  ] INFO 2022/04/09 21:28:24 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/04/09 21:28:24 listening at 0.0.0.0:7070
[go-e  ] WARN 2022/04/09 21:28:25 automatic 1p3p phase switching requires sponsor token
[site  ] INFO 2022/04/09 21:28:26 site config:
[site  ] INFO 2022/04/09 21:28:26   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/04/09 21:28:26     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/09 21:28:26     pv 0:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/04/09 21:28:26 loadpoint 1:
[lp-1  ] INFO 2022/04/09 21:28:26   mode:        off
[lp-1  ] INFO 2022/04/09 21:28:26   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/04/09 21:28:26   meters:      charge ✓
[lp-1  ] INFO 2022/04/09 21:28:26     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/09 21:28:26   vehicles:    ✓
[lp-1  ] INFO 2022/04/09 21:28:26     vehicle 0: range ✓ finish ✓ status ✓ climate ✓
[lp-1  ] INFO 2022/04/09 21:28:26 vehicle updated: unknown -> Beni eTron
[lp-1  ] DEBUG 2022/04/09 21:28:26 set charge mode: off
[lp-1  ] DEBUG 2022/04/09 21:28:26 set min current: 6
[lp-1  ] DEBUG 2022/04/09 21:28:26 set max current: 16
[lp-1  ] DEBUG 2022/04/09 21:28:26 set min soc: 20
[lp-1  ] DEBUG 2022/04/09 21:28:26 set target soc: 80
[site  ] DEBUG 2022/04/09 21:28:27 ----
[server] FATAL 2022/04/09 21:28:27 remove /tmp/evcc: operation not permitted
[main  ] FATAL 2022/04/09 21:28:27 listen tcp 0.0.0.0:7070: bind: address already in use
pi@raspberrypi:~ $

### What type of operating system are you running?

Linux

### Version

pi@raspberrypi:~ $ evcc -v
evcc version 0.89

andig, 2022-04-09:
DNS funktioniert nicht, aus welchem Grund auch immer.

closed 2022-04-09 by andig

Beninohl, 2022-04-09:
Hallo andig 
DNS funktioniert auf dem Raspberry, hatte die gleiche ide.

andig, 2022-04-10:
Der Fehler ist UDP auf Port 53- das ist DNS!

Beninohl, 2022-04-10:
Guten Morgen andig 
Im Raspberry funktioniert DNS, kann Identity.vwgroup.io Auflösung. Kann oder muss ich etwas im evcc ändern?
Danke für eine Info.

Beninohl, 2022-04-10:
Wenn ich den Link in der Fehlermeldung kopieren und im Safari bekommen ich die Fehlermeldung HTTP Status 400 - Bad request

premultiply, 2022-04-10:
IPv4 vs IPv6 ?

andig, 2022-04-10:
Das ist DNS:

    dial tcp: lookup identity.vwgroup.io on [2a02:1210:22ce:ec00:a2b5:49ff:fe25:4d40]:53: dial udp [2a02:1210:22ce:ec00:a2b5:49ff:fe25:4d40]:53: connect: network is unreachable

Das sind 2 gleichzeitig laufende Instanzen:

    listen tcp 0.0.0.0:7070: bind: address already in use