evcc-io/evcc #3130: Sunny Portal can not controll wallbox

AnduriI, 2022-04-10:
### Describe the bug

I entered my wallbox to evcc and enabled hems support with active controll. But SunnyPortal only recognizes it as "nichtsteuerbare Verbraucher". Is there anything I have to add in addition to the hems section or set in the online portal?

### Steps to reproduce

start evcc with `AllowControll=true` and look into sunnyportal

### Configuration details

```
hems:
  type: sma
  AllowControl: true
```

### Log details

don't see anything related to sunnyportal there...
```
[site  ] DEBUG 2022/04/10 12:45:05 ----
[lp-1  ] DEBUG 2022/04/10 12:45:05 charge power: 0W
[lp-2  ] DEBUG 2022/04/10 12:45:05 charge power: 0W
[lp-3  ] DEBUG 2022/04/10 12:45:05 charge power: 0W
[site  ] DEBUG 2022/04/10 12:45:05 pv power: 1902W
[site  ] DEBUG 2022/04/10 12:45:05 battery power: 1832W
[site  ] DEBUG 2022/04/10 12:45:06 grid power: -15W
[site  ] DEBUG 2022/04/10 12:45:06 grid currents: [2.95 2.83 2.98]A
[site  ] DEBUG 2022/04/10 12:45:06 battery soc 0: 26%
[site  ] DEBUG 2022/04/10 12:45:06 site power: 1917W
[lp-2  ] DEBUG 2022/04/10 12:45:06 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/04/10 12:45:06 charger status: A
[site  ] DEBUG 2022/04/10 12:45:15 ----
[lp-1  ] DEBUG 2022/04/10 12:45:15 charge power: 0W
[lp-2  ] DEBUG 2022/04/10 12:45:15 charge power: 0W
[lp-3  ] DEBUG 2022/04/10 12:45:15 charge power: 0W
[site  ] DEBUG 2022/04/10 12:45:15 pv power: 1919W
[site  ] DEBUG 2022/04/10 12:45:15 battery power: 100W
[site  ] DEBUG 2022/04/10 12:45:16 grid power: -10W
[site  ] DEBUG 2022/04/10 12:45:16 grid currents: [2.94 2.83 2.96]A
[site  ] DEBUG 2022/04/10 12:45:16 battery soc 0: 26%
[site  ] DEBUG 2022/04/10 12:45:16 site power: 190W
[lp-3  ] DEBUG 2022/04/10 12:45:16 charge currents: [0 0 0]A
[lp-3  ] DEBUG 2022/04/10 12:45:16 charger status: A
[site  ] DEBUG 2022/04/10 12:45:25 ----
[lp-1  ] DEBUG 2022/04/10 12:45:25 charge power: 0W
[lp-2  ] DEBUG 2022/04/10 12:45:25 charge power: 0W
[lp-3  ] DEBUG 2022/04/10 12:45:25 charge power: 0W
[site  ] DEBUG 2022/04/10 12:45:25 pv power: 1944W
[site  ] DEBUG 2022/04/10 12:45:25 battery power: 61W
[site  ] DEBUG 2022/04/10 12:45:26 grid power: -9W
[site  ] DEBUG 2022/04/10 12:45:26 grid currents: [9.07 2.73 9.71]A
[site  ] DEBUG 2022/04/10 12:45:26 battery soc 0: 26%
[site  ] DEBUG 2022/04/10 12:45:26 site power: 152W
[lp-1  ] DEBUG 2022/04/10 12:45:26 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/10 12:45:26 charger status: A
[site  ] DEBUG 2022/04/10 12:45:35 ----
[lp-1  ] DEBUG 2022/04/10 12:45:35 charge power: 0W
[lp-2  ] DEBUG 2022/04/10 12:45:35 charge power: 0W
[lp-3  ] DEBUG 2022/04/10 12:45:35 charge power: 0W
[site  ] DEBUG 2022/04/10 12:45:35 pv power: 1966W
[site  ] DEBUG 2022/04/10 12:45:35 battery power: 1786W
[site  ] DEBUG 2022/04/10 12:45:36 grid power: 2W
[site  ] DEBUG 2022/04/10 12:45:36 grid currents: [9 2.85 9.74]A
[site  ] DEBUG 2022/04/10 12:45:36 battery soc 0: 26%
[site  ] DEBUG 2022/04/10 12:45:36 site power: 1888W
[lp-2  ] DEBUG 2022/04/10 12:45:36 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2022/04/10 12:45:36 charger status: A
[site  ] DEBUG 2022/04/10 12:45:45 ----
[lp-1  ] DEBUG 2022/04/10 12:45:45 charge power: 0W
[lp-2  ] DEBUG 2022/04/10 12:45:45 charge power: 0W
[lp-3  ] DEBUG 2022/04/10 12:45:45 charge power: 0W
[site  ] DEBUG 2022/04/10 12:45:45 pv power: 1998W
[site  ] DEBUG 2022/04/10 12:45:45 battery power: 1265W
[site  ] DEBUG 2022/04/10 12:45:46 grid power: -2W
[site  ] DEBUG 2022/04/10 12:45:46 grid currents: [2.88 2.89 2.99]A
[site  ] DEBUG 2022/04/10 12:45:46 battery soc 0: 26%
[site  ] DEBUG 2022/04/10 12:45:46 site power: 1363W
[lp-3  ] DEBUG 2022/04/10 12:45:46 charge currents: [0 0 0]A
[lp-3  ] DEBUG 2022/04/10 12:45:46 charger status: A
```

### What type of operating system are you running?

Linux

### Version

`evcc version 0.89`

premultiply, 2022-04-10:
Da gibt es seitens SP nichts zu steuern.
Damit sind nur diese einfachen Steckdosen gemeint die man via SP an und ausknipsen kann.

allowControl solltest du übrigens besser auf false setzen es sein du willst wirklich die unsäglichen internen Algorithmen des SHM nutzen im Minutentakt. Das will aber normalerweise niemand weil evcc das deutlich besser, schneller und genauer macht.

closed 2022-04-10 by premultiply

AnduriI, 2022-04-10:
ich hatte die Hoffnung dass SHM dann den Hausspeicher runter drehen kann wenn die Prognose sagt es kommt noch viel Solarenergie und dafür die 4,2kW+Überschuss für die Wallbox freigibt.
Ich bin bisher ganz zufrieden mit der Batteriesteuerung, würde nur halt gern den Überschuss sinnvoller ins Auto laden wollen. Muss ich das dann besser mittels eingener Routinen machen oder wie sage ich evcc, dass zwar das Auto geladen werden soll, aber bitte abends der Hausspeicher über 95% liegen soll.

premultiply, 2022-04-10:
https://docs.evcc.io/docs/reference/configuration/site/#prioritysoc