evcc-io/evcc #3151: Identifier funktionieren nicht (Go-Echarger V3)

Waldmensch1, 2022-04-13:
### Describe the bug

Da wir jetzt 2 E-Autos haben, muss ich identifier erstmalig nutzen. Im Go-Echarger habe ich 2 RFID Karten angelernt und Namen vergeben "Zoe" und "Enyaq"

evcc charger liefert:

```
Power:          0W
Energy:         597.7kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Charged:        0.0kWh
Identifier:     Zoe

Power:          0W
Energy:         597.7kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Charged:        0.0kWh
Identifier:     Enyaq
```
 Sobald ich einen Identifier bei einem vehicle eintrage, wird dieses vehicle nicht mehr abgerufen. Das andere vehicle, ohne identifier, wird ganz normal geladen

```
- name: zoe
  identifier: Zoe #es spielt keine Rolle ob mit oder ohne Anführungszeichen
  type: template
  template: renault
  title: ZOE # Wird in der Benutzeroberfläche angezeigt # Optional
  user: xxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: xxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  #vin: WREN... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
  capacity: 40 # Akku-Kapazität in kWh # Optional 

```

### Steps to reproduce

Ein Template Enyaq und ein Template Renault anlegen. Sobald in einem vehicle ein identifier angegeben wird, wird dieses vehicle ignoriert

### Configuration details

.

### Log details

trace, man sieht, das der Enyaq Daten vom Server gezogen hat, die Zoe, wo der identifier gesetzt ist, jedoch nicht. Im tracelog sind auch etliche logeinträge von [enyaq ] mit tonnenweise HTML output aber kein einziger Logeintrag von [renault]

vehicle 0 ist hier der Enyaq und vehicle 1 die Zoe

```
[lp-1  ] WARN 2022/04/13 13:15:48 ignoring phases config (3p) for switchable charger
[lp-1  ] DEBUG 2022/04/13 13:15:48 phase timer inactive
[lp-1  ] WARN 2022/04/13 13:15:48 PV mode enable threshold (0W) is larger than disable threshold (-500W)
[cache ] TRACE 2022/04/13 13:15:48 auth: {map[]}
[site  ] INFO 2022/04/13 13:15:48 site config:
[site  ] INFO 2022/04/13 13:15:48   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/04/13 13:15:48     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/04/13 13:15:48     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/13 13:15:48     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/04/13 13:15:48     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/04/13 13:15:48     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/04/13 13:15:48 loadpoint 1:
[lp-1  ] INFO 2022/04/13 13:15:48   mode:        pv
[lp-1  ] INFO 2022/04/13 13:15:48   charger:     power ✓ energy ✓ currents ✓ phases ✓
[lp-1  ] INFO 2022/04/13 13:15:48   meters:      charge ✓
[lp-1  ] INFO 2022/04/13 13:15:48     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/13 13:15:48   vehicles:    ✓
[lp-1  ] INFO 2022/04/13 13:15:48     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/04/13 13:15:48     vehicle 1: range ✗ finish ✗ status ✗ climate ✗
[cache ] TRACE 2022/04/13 13:15:48 siteTitle: Stadtweg
[cache ] TRACE 2022/04/13 13:15:48 gridConfigured: true
[cache ] TRACE 2022/04/13 13:15:48 pvConfigured: true
[cache ] TRACE 2022/04/13 13:15:48 batteryConfigured: true
[cache ] TRACE 2022/04/13 13:15:48 prioritySoC: 100
[cache ] TRACE 2022/04/13 13:15:48 currency: EUR
[cache ] TRACE 2022/04/13 13:15:48 savingsSince: 1649848548
[cache ] TRACE 2022/04/13 13:15:48 lp-1/title: Hof
[cache ] TRACE 2022/04/13 13:15:48 lp-1/minCurrent: 6
[cache ] TRACE 2022/04/13 13:15:48 lp-1/maxCurrent: 16
[cache ] TRACE 2022/04/13 13:15:48 lp-1/phases: 0
[cache ] TRACE 2022/04/13 13:15:48 lp-1/activePhases: 3
[cache ] TRACE 2022/04/13 13:15:48 lp-1/hasVehicle: true
[cache ] TRACE 2022/04/13 13:15:48 lp-1/mode: pv
[cache ] TRACE 2022/04/13 13:15:48 lp-1/targetSoC: 90
[cache ] TRACE 2022/04/13 13:15:48 lp-1/minSoC: 0
[go-e  ] TRACE 2022/04/13 13:15:48 GET http://192.168.178.147/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2022/04/13 13:15:48 {"alw":true,"eto":597672,"car":3,"wh":0,"trx":3,"nrg":[230,226,228,0,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"User 1","energy":0,"cardId":false},{"name":"Enyaq ","energy":0,"cardId":true},{"name":"Zoe","energy":0,"cardId":true},{"name":"User 4","energy":0,"cardId":false},{"name":"User 5","energy":0,"cardId":false},{"name":"User 6","energy":0,"cardId":false},{"name":"User 7","energy":0,"cardId":false},{"name":"User 8","energy":0,"cardId":false},{"name":"User 9","energy":0,"cardId":false},{"name":"User 10","energy":0,"cardId":false}]}
[go-e  ] TRACE 2022/04/13 13:15:48 GET http://192.168.178.147/api/set?amp=6
[go-e  ] TRACE 2022/04/13 13:15:48 {"amp":true}
[lp-1  ] DEBUG 2022/04/13 13:15:48 max charge current: 6A
[site  ] DEBUG 2022/04/13 13:15:48 ----
```

hier nochmal ein trace, ohne gesetzten identifier bei der Zoe. Man sieht in der übersicht, das nun bei beiden vehicles Häkchen sind und ich sehe im Tracelog auch einträge mit [renault] und wie die Daten vom Server geholt werden

```
[lp-1  ] WARN 2022/04/13 13:35:19 ignoring phases config (3p) for switchable charger
[lp-1  ] DEBUG 2022/04/13 13:35:19 phase timer inactive
[lp-1  ] WARN 2022/04/13 13:35:19 PV mode enable threshold (0W) is larger than disable threshold (-500W)
[cache ] TRACE 2022/04/13 13:35:19 auth: {map[]}
[site  ] INFO 2022/04/13 13:35:19 site config:
[site  ] INFO 2022/04/13 13:35:19   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/04/13 13:35:19     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/04/13 13:35:19     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/04/13 13:35:19     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/04/13 13:35:19     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/04/13 13:35:19     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/04/13 13:35:19 loadpoint 1:
[lp-1  ] INFO 2022/04/13 13:35:19   mode:        pv
[lp-1  ] INFO 2022/04/13 13:35:19   charger:     power ✓ energy ✓ currents ✓ phases ✓
[lp-1  ] INFO 2022/04/13 13:35:19   meters:      charge ✓
[lp-1  ] INFO 2022/04/13 13:35:19     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/13 13:35:19   vehicles:    ✓
[lp-1  ] INFO 2022/04/13 13:35:19     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/04/13 13:35:19     vehicle 1: range ✓ finish ✓ status ✓ climate ✓
[cache ] TRACE 2022/04/13 13:35:19 siteTitle: Stadtweg
[cache ] TRACE 2022/04/13 13:35:19 gridConfigured: true
[cache ] TRACE 2022/04/13 13:35:19 pvConfigured: true
[cache ] TRACE 2022/04/13 13:35:19 batteryConfigured: true
[cache ] TRACE 2022/04/13 13:35:19 prioritySoC: 100
[cache ] TRACE 2022/04/13 13:35:19 currency: EUR
[cache ] TRACE 2022/04/13 13:35:19 savingsSince: 1649849719
[cache ] TRACE 2022/04/13 13:35:19 lp-1/title: Hof
[cache ] TRACE 2022/04/13 13:35:19 lp-1/minCurrent: 6
[cache ] TRACE 2022/04/13 13:35:19 lp-1/maxCurrent: 16
[cache ] TRACE 2022/04/13 13:35:19 lp-1/phases: 0
[cache ] TRACE 2022/04/13 13:35:19 lp-1/activePhases: 3
[cache ] TRACE 2022/04/13 13:35:19 lp-1/hasVehicle: true
[go-e  ] TRACE 2022/04/13 13:35:19 GET http://192.168.178.147/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[cache ] TRACE 2022/04/13 13:35:19 lp-1/mode: pv
[cache ] TRACE 2022/04/13 13:35:19 lp-1/targetSoC: 90
[cache ] TRACE 2022/04/13 13:35:19 lp-1/minSoC: 0
[go-e  ] TRACE 2022/04/13 13:35:19 {"alw":true,"eto":597672,"car":3,"wh":0,"trx":3,"nrg":[231,221,224,0,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"User 1","energy":0,"cardId":false},{"name":"Enyaq ","energy":0,"cardId":true},{"name":"Zoe","energy":0,"cardId":true},{"name":"User 4","energy":0,"cardId":false},{"name":"User 5","energy":0,"cardId":false},{"name":"User 6","energy":0,"cardId":false},{"name":"User 7","energy":0,"cardId":false},{"name":"User 8","energy":0,"cardId":false},{"name":"User 9","energy":0,"cardId":false},{"name":"User 10","energy":0,"cardId":false}]}
[go-e  ] TRACE 2022/04/13 13:35:19 GET http://192.168.178.147/api/set?amp=6
[go-e  ] TRACE 2022/04/13 13:35:19 {"amp":true}
[lp-1  ] DEBUG 2022/04/13 13:35:19 max charge current: 6A
[cache ] TRACE 2022/04/13 13:35:19 lp-1/chargeCurrent: 6
[site  ] DEBUG 2022/04/13 13:35:19 ----
```

### What type of operating system are you running?

Linux

### Version

Actual Master branch

andig, 2022-04-13:
Kannst Du bitte mal ein komplettes Log (ohne cache) schicken mit Kommentierung der Timestamp was wo passiert? Das Stückwerk ist immer schwer verständlich.

andig, 2022-04-13:
Und schau bitte auch nochmal ob Du keine Fehler im Log hast!

andig, 2022-04-13:
identifiers:
      - Zoe

ist eine Liste. Deine Config ist falsch und das wird im Logfile auch so angegeben!

closed 2022-04-13 by andig

Waldmensch1, 2022-04-13:
ich bin nach Doku vorgegangen https://docs.evcc.io/docs/guides/vehicles
```
vehicles:
  - name: zoe
    ...
    identifier: "<kopierte Wert>"
```