evcc-io/evcc #3157: Nissan Leaf: SoC connection still not working with updated api

Percy2Live, 2022-04-13:
### Describe the bug

Nissan Leaf vehicles cannot be connected to evcc to receive the SoC

### Steps to reproduce

evcc.yaml configuration of the vehicle:

`vehicles:
- type: template
  template: nissan 
  title: Leaf  
  user: mailOfUser  
  password: passwordOfUser  
  vin: SJNFAAZE1XXXXXXXX  
  capacity: 40  
  name: ev1`

cli states this:

`/sys/module # evcc vehicle
[main  ] INFO 2022/04/13 22:25:19 evcc 0.89
[main  ] INFO 2022/04/13 22:25:19 using config file /etc/evcc.yaml
ev1
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'nissan': login failed: Unauthorized: Authentication Failed
Capacity: 0kWh
`

### Configuration details

Running evcc on a Synology docker, release 0.89

### Log details

`/sys/module # evcc --log debug
[main  ] INFO 2022/04/13 22:29:46 evcc 0.89
[main  ] INFO 2022/04/13 22:29:46 using config file /etc/evcc.yaml
[main  ] INFO 2022/04/13 22:29:46 listening at 192.168.178.30:7070
[go-e  ] WARN 2022/04/13 22:29:47 automatic 1p3p phase switching requires sponsor token
[go-e  ] WARN 2022/04/13 22:29:48 automatic 1p3p phase switching requires sponsor token
[site  ] INFO 2022/04/13 22:29:48 site config:
[site  ] INFO 2022/04/13 22:29:48   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/04/13 22:29:48     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/04/13 22:29:48     pv 0:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/13 22:29:48 loadpoint 1:
[lp-1  ] INFO 2022/04/13 22:29:48   mode:        pv
[lp-1  ] INFO 2022/04/13 22:29:48   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/04/13 22:29:48   meters:      charge ✓
[lp-1  ] INFO 2022/04/13 22:29:48     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/04/13 22:29:48   vehicles:    ✗
[lp-2  ] INFO 2022/04/13 22:29:48 loadpoint 2:
[lp-2  ] INFO 2022/04/13 22:29:48   mode:        pv
[lp-2  ] INFO 2022/04/13 22:29:48   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-2  ] INFO 2022/04/13 22:29:48   meters:      charge ✓
[lp-2  ] INFO 2022/04/13 22:29:48     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2022/04/13 22:29:48   vehicles:    ✗
[site  ] DEBUG 2022/04/13 22:29:48 ----
[main  ] FATAL 2022/04/13 22:29:48 listen tcp 192.168.101.30:7070: bind: address already in use`

### What type of operating system are you running?

Docker container

### Version

evcc version 0.89

andig, 2022-04-14:
Did you try `carwings` too? Please send login credentials to info@evcc.io.

Percy2Live, 2022-04-14:
Mail is sent out, thanks. carwings instead of nissan leads to a similar issue:

```
ev2
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'carwings': login failed: received status code -2010 (INVALID PARAMS)
Capacity: 0kWh

```

closed 2022-04-14 by andig