evcc-io/evcc #3167: How can I integrate the EV Charger from SMA

Padm59, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
I noticed that the EV Charger from SMA is not listed in the compatible device list.

**Describe the solution you'd like**
Add a configuration option for the EV Charger from SMA and support it officially.

**Describe alternatives you've considered**
Is it somehow possible to add the EC Charger manual? Has somebody ever done that before?

**Additional context**
I hope somebody can help me and others who also have the EV Charger from SMA. Thanks in advance.

andig, 2022-04-15:
This will only be possible if anybody is able to provide a) documentation and b) access to test device.

premultiply, 2022-04-15:
The SMA EV Charger is (currently?) a vendor locked-in device. It is not possible to control it in any way apart from emulating a SHM2 as SEMP controller and Speedwire Energy Meter in theory.

But this would be limited at least the same way as the vendor solution in many aspects.
It MAY be possible to control it much faster and more precisely as the SHM itself but other things may not even be controlable at all as they are locked in to other unknown local and cloud API interfaces.
And of course there are hardware limitations like the large mechanical mode switch which needs to be adjusted outside at the device by design.

andig, 2022-04-15:
https://sandbox.smaapis.de/smarthome/index.html bietet keine Möglichkeit, die Ladeleistung zu kontrollieren. https://developer.sma.de/api-plans.html zeigt auf, dass SMA für die bescheidnen Fähigkeit Geld haben möchte.

Closing as wontfix.

closed 2022-04-15 by andig