evcc-io/evcc #3175: Something might be wrong with the arm64 image of version 0.90

MehrCurry, 2022-04-16:
### Describe the bug

By running the latest relase version 0.90 on my raspberry pi i got

`standard_init_linux.go:228: exec user process caused: exec format error`

which normally means that this image has been build for another architecture.

### Steps to reproduce

I have kubernetes deployment which looks like this:

```
apiVersion: apps/v1
kind: Deployment
metadata:
  labels:
    app: evcc
    app.kubernetes.io/instance: evcc
  name: evcc
  namespace: evcc
spec:
  progressDeadlineSeconds: 600
  replicas: 1
  revisionHistoryLimit: 10
  selector:
    matchLabels:
      app: evcc
  strategy:
    rollingUpdate:
      maxSurge: 25%
      maxUnavailable: 25%
    type: RollingUpdate
  template:
    metadata:
      annotations:
        checksum/config: a999e91af978d2e312c4947e81aa828bf0e0e833d2b4778053b4f80e4b2a8055
      labels:
        app: evcc
    spec:
      containers:
        - image: 'andig/evcc:0.90'
          imagePullPolicy: Always
          livenessProbe:
            httpGet:
              path: /api/health
              port: 7070
            initialDelaySeconds: 15
            periodSeconds: 10
          name: evcc
          ports:
            - containerPort: 7070
          readinessProbe:
            httpGet:
              path: /api/health
              port: 7070
            initialDelaySeconds: 10
            periodSeconds: 5
          resources:
            limits:
              cpu: 200m
              memory: 64Mi
            requests:
              cpu: 100m
              memory: 32Mi
          terminationMessagePath: /dev/termination-log
          terminationMessagePolicy: File
          volumeMounts:
            - mountPath: /etc/evcc.yaml
              name: config-volume
              subPath: evcc.yaml
      restartPolicy: Always
      terminationGracePeriodSeconds: 30
      volumes:
        - name: config-volume
          secret:
            secretName: evcc-config

```
This leads the the above error. With image tag of 0.89 all works fine.

### Configuration details

Doesn't matter

### Log details

No real log output since the binary is wrong.

### What type of operating system are you running?

Linux

### Version

0.90

andig, 2022-04-16:
Tut mir leid, kein Docker/K8tes support geschweige denn eine Möglichkeit das nachzuvollziehen.

closed 2022-04-16 by andig

MehrCurry, 2022-04-17:
For those faceing the same problem. It was my fault. I am using Helm for deployment and my  values file looks like this:

Wrong Version:
```
image:
  repository: andig/evcc
  tag: 0.90
```

This worked fine for all previous versions. But since the tag is treated as a number the whole image descriptor for the current version in converted to `docker.io/andig/evcc:0.9` which seems to be a version from 2 years ago where no ARM64 was existiing.

So the right values.yaml is now
```
image:
  repository: andig/evcc
  tag: "0.90"
```
where the tag is treated as as string.

andig, 2022-04-17:
Oh mann… ja, das ist das tückische Problem mit yaml. Immer wieder überraschend, wenn es zuschlägt… schönes Osterei 🐣

andig, 2022-04-17:
Tatsächlich entspricht Deine Fehlerbeschreibung auch nicht dem Issue:

    image: 'andig/evcc:0.90'

Das hätte eigentlich funktionieren sollen, da hier ja alles gequoted ist.

MehrCurry, 2022-04-17:
Ja, das stimmt. Du hast mich beim Schummeln erwischt.

Ich habe den Issue aufgemacht, nachdem ich den "Downgrade" auf 0.89 schon gemacht hatte. Dann hab ich den 0.89 Deployment Descriptor kopiert und dann per Hand auf 0.90 geändert. Das heißt, ich habe den Fehler gleich doppelt gemacht. Wenn schon, denn schon.