evcc-io/evcc #3198: Include the Charger Energy Counter Values (if available) to evcc \ influx

kaluxx, 2022-04-20:
Many chargers have their own absolute energy counter included which is accessable through evcc `evcc charger`. 

-> Please query the charger Energy value while charging and forward those vales to the influx db.

Reasons: 

1. I want to better understand and display (Grafana) charging behaviour and charging losses. Today for the charged energy there is only the evcc counter for chargedEnergy available. 
2. Today ev-charging dominates the energy flow in my household- when active. Other contributors to house energy consumption are easily hidden. I want to provide a view without ev-charging (like in the past) which shows small energy contributions as well. This may be done with the help of the requested wallbox (charger) counter. -> Energy for charging / Energy in house

andig, 2022-04-20:
What would be the measurement names?

premultiply, 2022-04-20:
TotalEnergy?

kaluxx, 2022-04-20:
ChargerTotalEnergy?
to make clear it is not from evcc ?

kaluxx, 2022-05-18:
Verstehe ja die professionelle Enscheidungsfindung hier nicht so ganz. Ich kann aber nachvollziehen, dass der nachgefragte Wert nicht unabdingbar für evcc ist. Trotzdem - er ist eh bekannt und soll ja nur zusätzlich abgepeichert werden. 
Bin ja mal gespannt, was rauskommt. 
Sonst muss ich noch eine extra Abfrage zur wallbox (totalEnergy) machen obwohl eigentlich alles schon über evcc abgefragt wurde. 
Aus welchem Grund wird überhaupt der ganze andere Kram eigentlich in die influxdb geschrieben? Verwendet das jemand zum Loggen. Die meisten user hier werden damit doch eher ihr smart home anreichern.

> ["activePhases"],["chargeCurrent"],["chargeCurrents"],["chargePower"],["chargeRemainingEnergy"],["chargedEnergy"],["gridCurrents"],["gridEnergy"],["gridPower"],["homePower"],["loadpoint"],["maxCurrent"],["minCurrent"],["minSoC"],["phases"],["prioritySoC"],["pvPower"],["savingsAmount"],["savingsEffectivePrice"],["savingsGridCharged"],["savingsSelfConsumptionCharged"],["savingsSelfConsumptionPercent"],["savingsSince"],["savingsTotalCharged"],["targetSoC"],["targetTime"],["targetTimeProjectedStart"],["tariffFeedIn"],["tariffGrid"],["vehicleCapacity"],["vehicleOdometer"],["vehicleRange"],["vehicleSoC"]

andig, 2022-05-18:
Neben aller Entscheidung miss es halt auch jemand machen ;). 
OT: Falls Dich Influx stört- warum nicht einfach nicht verwenden?

kaluxx, 2022-05-20:
Nee, Nee, die Möglichbkeit mir dem Speichern der Werte in die Datenbank finde ich super. ;-) Habe mich halt nur gefragt wofür das ursprünglich  gedacht ist. 
Wenn es nur für evcc interne Zwecke benutzt wird verstehe ich eine super restriktive Handhabung.  
Ich verwende es, wie vielleicht manch anderer, als Datenquelle zu Hause. Da wäre es halt schön, wenn evcc einfach alles, was es sowieso im Rahmen der Regelung erfasst, auch in die DB schreibt. Jetzt wird halt manches geschrieben und anderes nicht - wollte die Gründe verstehen.  Idealerweise wäre konfigurierbar was exportiert wird- aber das ist wahrscheinlich ein Riesenaufwand und nicht Ziel von evcc...

andig, 2022-05-20:
Es *wird* alles geschrieben was erfasst wird.  PV Energy wird aber nicht erfasst bzw aktuell nur sehr rudimentär verarbeitet…

andig, 2022-05-28:
In https://github.com/evcc-io/evcc/pull/3384 ist eine ähnliche Anfrage für PV. Aus Konsistenzgründen könnten wir hier hinzufügen: `gridEnergy` und `chargeEnergy` where the prefix is identical to the meter name. While doing so we should also rename `chargedEnergy` to `sessionEnergy`.

@naltatis @premultiply I would appreciate your votes.

premultiply, 2022-05-28:
gridEnergy wäre dann nur der Netzbezug (ohne Einspeisung) ?

Müssen/sollten dann alle Meter dann auch immer die Energie mitliefern?
Da wären dann ggf. noch einige Templates anzupassen.

andig, 2022-05-29:
> gridEnergy wäre dann nur der Netzbezug (ohne Einspeisung) ?

Arghhh. Guter Punkt. Ich denke für's erste ja- wobei sich da direkt wieder die Frage stellt, ob der Namen dann richtig gewählt ist. Besser `gridImport(Energy)` oder  `grid(Energy)Import` um es zukunftsfähig zu halten?

> Müssen/sollten dann alle Meter dann auch immer die Energie mitliefern? Da wären dann ggf. noch einige Templates anzupassen.

Ja, aber das könnte ja sukzessive bei Bedarf erfolgen.

closed 2022-08-07 by github-actions[bot]

andig, 2022-08-07:
Ich hab mir das nochmal angeschaut. Der Zählerstand des Ladezählers ist leider ziemlich verborgen und hat aktuell keinen Zugriff darauf, die Werte auszugeben. Ich fürchte fürs erste daher wontfix, sorry.