evcc-io/evcc #3207: NO_PUBKEY error when installing

jpalo, 2022-04-22:
### Describe the bug

```
openhabian@openhab:~/sofar $ sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https
[sudo] password for openhabian:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'raspbian-archive-keyring' instead of 'debian-archive-keyring'
apt-transport-https is already the newest version (1.8.2.3).
raspbian-archive-keyring is already the newest version (20120528.2).
The following package was automatically installed and is no longer required:
  triggerhappy
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  debian-keyring
0 upgraded, 1 newly installed, 0 to remove and 82 not upgraded.
Need to get 32.4 MB of archives.
After this operation, 33.7 MB of additional disk space will be used.
Get:1 http://raspbian.mirror.vu.lt/raspbian/raspbian buster/main armhf debian-keyring all 2019.02.25 [32.4 MB]
Fetched 32.4 MB in 4s (8,516 kB/s)
Selecting previously unselected package debian-keyring.
(Reading database ... 103428 files and directories currently installed.)
Preparing to unpack .../debian-keyring_2019.02.25_all.deb ...
Unpacking debian-keyring (2019.02.25) ...
Setting up debian-keyring (2019.02.25) ...
Updating FireMotD available updates count ...
openhabian@openhab:~/sofar $ curl -1sLf 'https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/evcc-stable.list
# Source: evcc
# Site: https://github.com/evcc-io/evcc
# Repository: evcc / stable
# Description: Sonne tanken ☀️🚘 - Releases

deb https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main

deb-src https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main
openhabian@openhab:~/sofar $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:3 http://phoscon.de/apt/deconz buster InRelease
Hit:4 https://deb.nodesource.com/node_8.x buster InRelease
Get:5 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease [5,104 B]
Hit:6 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease
Err:5 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EAD5D0E07B0EC0FD
Reading package lists... Done
W: GPG error: https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EAD5D0E07B0EC0FD
E: The repository 'https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
openhabian@openhab:~/sofar $ sudo apt install -y evcc
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package evcc
```

### Steps to reproduce

Follow installation instructions in documentation.

### Configuration details

N/A

### Log details

N/A

### What type of operating system are you running?

Linux

### Version

stable

goebelmeier, 2022-04-22:
You forgot this step:

curl -1sLf 'https://dl.cloudsmith.io/public/evcc/stable/gpg.key' | sudo tee /etc/apt/trusted.gpg.d/evcc-stable.asc

andig, 2022-04-22:
@jpalo nice to see evcc make it's way to Finland :)

closed 2022-04-24 by andig

Macrisu, 2022-05-17:
sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https
curl -1sLf 'https://dl.cloudsmith.io/public/evcc/stable/gpg.key' | sudo tee /etc/apt/trusted.gpg.d/evcc-stable.asc
curl -1sLf 'https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/evcc-stable.list

I've cheched etc/apt/trusted.gpg.d/evcc-stable.asc and /etc/apt/sources.list.d/evcc-stable.list

I get the same error.

sudo apt update
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                                        
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                               
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease                                
Get:5 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease [5,104 B]
Err:5 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EAD5D0E07B0EC0FD
Reading package lists... Done
W: GPG error: https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EAD5D0E07B0EC0FD
E: The repository 'https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

jaydkay, 2022-05-22:
Can confirm the above, I'm seeing the same problem.

I've installed the key as suggested in the installation instructions and the key seems to be there:

**#apt-key list**
```
/etc/apt/trusted.gpg.d/evcc-stable.asc
--------------------------------------
pub   rsa3072 2021-12-03 [SCEA]
      9968 5CA9 149A 5389 F6AC  7B06 EAD5 D0E0 7B0E C0FD
uid        [ unbekannt ] Cloudsmith Package (evcc/evcc) <support@cloudsmith.io>
```
It still doesn't work:

**#sudo apt update**
```
Holen:2 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease [5.113 B]
Fehl:2 https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease
  Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY EAD5D0E07B0EC0FD
Paketlisten werden gelesen... Fertig
W: GPG-Fehler: https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY EAD5D0E07B0EC0FD
E: Das Depot »https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease« ist nicht signiert.
N: Eine Aktualisierung von solch einem Depot kann nicht auf eine sichere Art durchgeführt werden, daher ist es standardmäßig deaktiviert.
N: Weitere Details zur Erzeugung von Paketdepots sowie zu deren Benutzerkonfiguration finden Sie in der Handbuchseite apt-secure(8).
```
Any ideas?

Best regards,
Jochen

Macrisu, 2022-05-22:
https://github.com/evcc-io/evcc/issues/3396#issuecomment-1129584589

andig, 2022-05-22:
> Any ideas?

Need PR