evcc-io/evcc #3214: Kommunikationsproblem Auto - go-e-Wallbox

stbrumm847, 2022-04-22:
### Describe the bug

Hallo,

ein Kollege hat eine Go-Wallbox mit der er sein Auto lädt. Ich starte das evcc und versuche das Laden via den Kommandos zu starten. Die einzige Rückmeldung ist 'verbunden' . Ein Laden (egal welcher Modus) erfolgt anscheinend nicht. Mit seiner App geht es schon, dann wird auch in der evcc-GUI angezeigt, das es geladen wird. Wo könnte der Fehler sein? 

LG
Steffen

### Steps to reproduce

go-e-box
E-Auto

Auto an Ladestation anschließen, evcc starten, die diversen Modi testen

### Configuration details

uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: info
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
  - name: smartpi
    type: custom
    power: 
      source: mqtt
      topic: SmartPi/P1  
  
  - name: Fronius
    type: custom
    power:
      source: http
      uri: http://wa-web-back-staging.watt-analytics.com/api/solar_api/v1/GetPowerFlowRealtimeData.fcgi
      jq: if .Body.Data.Inverters."1".P == null then 0 else .Body.Data.Inverters."1".P end
  

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
  - name: go-e
    type: template
    template: go-e-v3
    host: helios.ddnss.de:52/mqtt?payload=amp=32 # ModBus address
    #host: localhost:8080
 
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
vehicles:
  - name: car1
    type: renault
    title: Zoe
    capacity: 60 # kWh
    user: myuser # user
    password: mypassword # password
    vin: WREN...
[... 244 lines cut ...]
[lp-1  ] DEBUG 2022/04/22 15:16:56 charger disable: contactor delay 3m18s
[site  ] DEBUG 2022/04/22 15:17:04 ----
[lp-1  ] DEBUG 2022/04/22 15:17:06 charge power: 0W
[site  ] DEBUG 2022/04/22 15:17:06 pv power: 0W
[site  ] DEBUG 2022/04/22 15:17:06 grid power: 4W
[site  ] DEBUG 2022/04/22 15:17:06 site power: 4W
[lp-1  ] DEBUG 2022/04/22 15:17:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/22 15:17:08 charger status: A
[lp-1  ] DEBUG 2022/04/22 15:17:08 charger disable: contactor delay 3m5s
[site  ] DEBUG 2022/04/22 15:17:14 ----
[lp-1  ] DEBUG 2022/04/22 15:17:17 charge power: 0W
[site  ] DEBUG 2022/04/22 15:17:17 pv power: 0W
[site  ] DEBUG 2022/04/22 15:17:17 grid power: 4W
[site  ] DEBUG 2022/04/22 15:17:17 site power: 4W
[lp-1  ] DEBUG 2022/04/22 15:17:17 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/22 15:17:17 charger status: A
[lp-1  ] DEBUG 2022/04/22 15:17:17 charger disable: contactor delay 2m56s

### What type of operating system are you running?

Windows

### Version

0.8.9

RTTTC, 2022-04-22:
For security reasons - remove your tokens, id, passwords and usernames form your post! They are now public knowledge  ...

andig, 2022-04-24:
Formatier bitte das Issue- die Config ist so nicht lesbar. Die Fehlermeldung sehe ich nirgendwo?

stbrumm847, 2022-04-25:
url