evcc-io/evcc #3215: JSON Parsing error

hyperbart, 2022-04-22:
### Describe the bug

After disconnecting my Alfen charging equipment EVCC and now turning it back on again EVCC is giving me an error:

[{"savingsSelfConsumptionCharged":NaN,"loadpoints.0.pvRemaining":0,"gridConfigured":true,"gridEnergy":4164.2,"loadpoints.0.connectedDuration":34,"loadpoints.0.chargeCurrent":0,"loadpoints.0.chargedEnergy":0,"tariffFeedIn":0.12,"savingsAmount":NaN,"loadpoints.0.targetTime":null,"loadpoints.0.phases":0,"loadpoints.0.activePhases":3,"loadpoints.0.mode":"off","loadpoints.0.title":"Alfen Laadpaal","loadpoints.0.minCurrent":6,"loadpoints.0.maxCurrent":16,"homePower":481.94,"loadpoints.0.chargePower":0,"auth":{"vehicles":{}},"prioritySoC":0,"currency":"EUR","savingsSelfConsumptionPercent":NaN,"loadpoints.0.targetTimeProjectedStart":null,"loadpoints.0.loadpoint":1,"loadpoints.0.chargeDuration":0,"tariffGrid":0.42,"savingsTotalCharged":NaN,"savingsEffectivePrice":NaN,"siteTitle":"Dynamic Charging","batteryConfigured":false,"savingsSince":1650547669,"pvConfigured":true,"savingsGridCharged":NaN,"loadpoints.0.hasVehicle":false,"loadpoints.0.minSoC":0,"gridPower":-300.06,"loadpoints.0.charging":false,"loadpoints.0.enabled":false,"sponsor":"hyperbart","pvPower":782,"loadpoints.0.connected":false,"loadpoints.0.targetSoC":100,"loadpoints.0.chargeCurrents":[0,0,0],"loadpoints.0.pvAction":"inactive"}] 

### Steps to reproduce

Turn off EVSE for a while, turn it back on, no car has connected in the mean time

### Configuration details

interval: 5s

site:
- title: Dynamic Charging
  meters:
    grid: HuisMeter
    pvs:
    - pvHuis
    - pvTuinHuis

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.42 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12 # [currency]/kWh

meters:
  - name: HuisMeter
    type: modbus
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.4.184:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter
  - name: pvHuis
    type: sma
    uri: 192.168.4.230
    password: xxxxxxx
  - name: pvTuinHuis
    type: sma
    uri: 192.168.4.231
    password: xxxxxx

sponsortoken: xxxxxxxxx

log: debug
levels:
    core: debug

chargers:
- name: ChargingPointAlfen
  type: alfen
  uri: 192.168.4.178 # TCP ModBus address

loadpoints:
- title: Alfen Laadpaal # display name for UI
  charger: ChargingPointAlfen # Wallbox Referenz
  vehicles:
  mode: pv # charge mode (off, now, minpv, pv)
  phases: 3
  mincurrent: 6
[... 990 lines cut ...]
[site  ] DEBUG 2022/04/22 15:44:01 site power: -1813W
[lp-1  ] DEBUG 2022/04/22 15:44:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/22 15:44:01 charger status: A
[site  ] DEBUG 2022/04/22 15:44:05 ----
[lp-1  ] DEBUG 2022/04/22 15:44:05 charge power: 0W
[site  ] DEBUG 2022/04/22 15:44:05 pv power: 2257W
[site  ] DEBUG 2022/04/22 15:44:06 grid power: -1806W
[site  ] DEBUG 2022/04/22 15:44:06 site power: -1806W
[lp-1  ] DEBUG 2022/04/22 15:44:06 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/22 15:44:06 charger status: A
[site  ] DEBUG 2022/04/22 15:44:10 ----
[lp-1  ] DEBUG 2022/04/22 15:44:10 charge power: 0W
[site  ] DEBUG 2022/04/22 15:44:10 pv power: 2669W
[site  ] DEBUG 2022/04/22 15:44:11 grid power: -2562W
[site  ] DEBUG 2022/04/22 15:44:11 site power: -2562W
[lp-1  ] DEBUG 2022/04/22 15:44:11 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/04/22 15:44:11 charger status: A

### What type of operating system are you running?

Linux

### Version

Version 0.90

andig, 2022-04-24:
> EVCC is giving me an error:

What error?

Please attach log as file, show error and format config.

hyperbart, 2022-04-24:
Show error:

[image]

andig, 2022-04-24:
@naltatis hier gibts ein Problem:

> "savingsSelfConsumptionCharged":NaN

naltatis, 2022-04-24:
@andig Schau ich mir an.

hyperbart, 2022-04-27:
@naltatis @andig after a destroy of the docker container and restart I don't get the error anymore.

PR #3298 (2022-05-02): Fix Websocket JSON; represent NaN as null
`NaN` is not a valid value in JSON. Changed the marshalling to prepresent NaN values as `null`.

Fixes #3215
closed 2022-05-02 by andig