evcc-io/evcc #3219: cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': no such file or directory

bjoernevers, 2022-04-23:
### Describe the bug

when i test the charger configuration i get the error: cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': no such file or directory

### Steps to reproduce

sudo evcc -l debug charger

### Configuration details

  GNU nano 5.4                                                                                  /etc/evcc.yaml                                                                                            
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxx

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.30
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.30
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.178.30
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

vehicles:
- type: template
  template: hyundai
  title: Hyundai Kona
  user: xxxxxxx
  password: xxxxxxxx
  capacity: 39
  phases: 1
  language: de
  mode: pv
  minSoC: 25
  targetSoC: 80
  minCurrent: 6

### Log details

[main  ] INFO 2022/04/23 20:02:40 evcc 0.90
[main  ] INFO 2022/04/23 20:02:40 using config file /etc/evcc.yaml
[main  ] INFO 2022/04/23 20:02:40 listening at :7070
[main  ] FATAL 2022/04/23 20:02:40 cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': no such file or directory

### What type of operating system are you running?

Linux

### Version

0.9.0

andig, 2022-04-24:
Konfiguration fehlt

closed 2022-04-24 by andig

bjoernevers, 2022-04-25:
Konfiguration (evcc.yaml) ist angelegt. Er liest ja aus der Konfiguration "cannot create charger 'wallbox1': cannot create charger 'template': cannot create"

andig, 2022-04-25:
Das mag alles sein. Ist in dem unformatierten Text oben aber nicht erkennbar. Da gibts kein "wallbox1". Wenn Du Hilfe brauchst musst Du uns bitte Deine Config zeigen, nicht irgendetwas Anderes!