evcc-io/evcc #3221: Go-E: use „Ausstecken simulieren“ for wakeup

andig, 2022-04-24:
Hallo,

wir haben bei dem Setup leider noch manchmal das Problem, das der Corsa an wechselhaften Tagen das laden stoppt. (Ich vermute weil das Laden zu oft unterbrochen und neu gestartet wird). Manchmal muss das Ladekabel gezogen werden, manchmal reicht es auf sofort und dann wieder PV laden zu schalten.

Unsere Konfiguration ist eigentlich auf Standard, gibt es da vielleicht noch Verbesserdungspotenzial? Wie geht es anderen Corsa Besitzern?

```
delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge p>
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 10m # switch charger contactor not more often than this (default 5m)##
```

Ich habe jedenfalls bei der Go-E V3 Firmware V053.1 folgenden Punkt gefunden:

"neuer API key "sua" der die Funktion "Ausstecken simulieren" dauerhaft aktiviert während das Laden nicht erlaubt ist"

Wird dieser API key schon verwendet? Falls nicht, könnte er die Zuverlässigkeit erhöhen? Wäre es ein großer Aufwand diesen zu implementieren?

_Originally posted by @Texzor in https://github.com/evcc-io/evcc/discussions/3217_

Sillium007, 2022-04-25:
Sollte man die FW-Version gegenprüfen oder möchtest Du das quasi über die Doku machen?

andig, 2022-04-25:
Keep it simple- Doku!

andig, 2022-04-25:
> Das kann beim Peugeot auch passieren.Aber bis jetzt hatte ich es beim EVCC noch nicht.Stecker muss aber aus dem Auto gezogen werden.

Hier gehts aber um die Wallbox und nicht ums Auto ;)

Trunks1982, 2022-04-25:
> > Das kann beim Peugeot auch passieren.Aber bis jetzt hatte ich es beim EVCC noch nicht.Stecker muss aber aus dem Auto gezogen werden.
> 
> Hier gehts aber um die Wallbox und nicht ums Auto ;)

Hatte so verstanden.Sorry😉

rhocco, 2022-04-28:
Ist das der wirklich bessere Weg , das ausstecken zu simulieren und "unnötige Schaltvorgänge" am Auto zu triggern? Wäre es da nicht besser mit einem Puffer zu arbeiten? So dass an bewölkten Tagen Die Ladung erst nach 2 Minuten anspringt, wenn genug Überschuss da ist und mindestens 10 Minuten läuft. Auch die "residualPower" auf 500 oder 1000 
 stellen. Man kommt dann zwar etwas in den Netzbezug rein aber das ist vernachlässigbar. Ich habe es so gemacht und habe kaum bis garkeine Aus/An Schaltvorgänge. 

Ich hab in der config 1Phase fest, und 6-14 Ampere variabel, damit reduzieren siehc auch bei Sonne/Wolken wechsel die Schaltvorgänge auf ein Minimum. Ich habe ein ungutes gefühl, am Autro das Laden oft zu starten/stoppen... Dafür ist das Konzept des Ladens (noch) nicht gemacht. 

Nur so meine Gedanken.

andig, 2022-04-28:
@rhocco hier gehts nicht um die Diskussion sondern um die Implementierung wann und wo sie notwendig ist. Notwendigkeit wird hier nicht besprochen (da wäre https://github.com/evcc-io/evcc/discussions/3217 der Platz).

Es scheint allerdings, als würden wir nichts implementieren, weil das Feature in der Go-E schlicht nicht funktioniert.

@texzor oder gibts hier noch was zu tun?

premultiply, 2022-04-28:
Mir kommt es so vor von dem was ich hier gelesen hab als könnte man dies ohne über die App generell ein/ausschalten, so dass evcc das nicht nochmal explizit antriggern muss.

premultiply, 2022-04-29:
Gut, dann machen wir hier erstmal zu.
Falls es neue Erkenntnisse gibt und es doch ein explizite Kommando seitens evcc bräuchte dann einfach nochmal hier melden.

closed 2022-04-29 by premultiply