evcc-io/evcc #3228: Berechnung Solarleistung

knight1009, 2022-04-24:
Hallo, ich habe das heute mal alles installiert und getestet..Läuft auf anhieb fast perfekt.. Tolles Projekt. 
Eine Kleinigkeit ist mir aber aufgefallen.
Ich habe ein Shelly3EM und einen GoEcharger. Mein Solarwechselrichter läßt sich leider nicht einbinden. Anlage macht 3,8 kWp. In der Konfiguration habe ich PV-Laden so eingestellt, dass 2 kW aus dem Netz bezogen werden dürfen.
PV-Überschuss wird erkannt, Laden beginnt.
Leistung Charger: 4 kW
Verbrauch bzw Netzbezug: 2 kW
Erzeugung bzw Eigenverbrauch: 0 W <-passt nicht
Kann ich da irgendwas in der Config so ändern, dass die erzeugte Leistung berechnet und angezeigt wird? Denn so funktioniert auch die Auswertung "x Prozent Sonnenenergie" nicht.

_Originally posted by @knight1009 in https://github.com/evcc-io/evcc/discussions/2104#discussioncomment-2624184_

 open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 192.168.0.77
  name: grid1

chargers:
- type: template
  template: go-e
  host: 192.168.0.30
  name: wallbox3

vehicles:
- type: template
  template: tesla
  title: Enterprise2
  accessToken: 
  refreshToken: 
  capacity: 60
  phases: 3
  mode: pv
  minSoC: 20
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev2

loadpoints:
- title: Carport
  charger: wallbox3
  vehicles:
  - ev2
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  residualPower: -2000 # 2000W Netzbezug im PV Modus erlaubt
  title: Thorstens Ladestation
  meters:
    grid: grid1
tariffs:
  currency: EUR # (default EUR)
[... 145 lines cut ...]
Apr 24 13:08:19 openHABianDevice evcc[20639]: [lp-1  ] INFO 2022/04/24 13:08:19   vehicles:    ✓
Apr 24 13:08:19 openHABianDevice evcc[20639]: [lp-1  ] INFO 2022/04/24 13:08:19     vehicle 1: range ✓ finish ✓ status ✓ climate ✗
Apr 24 13:08:19 openHABianDevice evcc[20639]: [lp-1  ] INFO 2022/04/24 13:08:19 vehicle updated: unknown -> Enterprise2
Apr 24 13:09:39 openHABianDevice evcc[20639]: [lp-1  ] INFO 2022/04/24 13:09:39 start charging ->
Apr 24 13:22:39 openHABianDevice evcc[20639]: [lp-1  ] INFO 2022/04/24 13:22:39 stop charging <-
Apr 24 14:22:49 openHABianDevice evcc[20639]: [lp-1  ] ERROR 2022/04/24 14:22:49 vehicle soc: 408 Request Timeout
Apr 24 14:39:58 openHABianDevice systemd[1]: Stopping evcc...
Apr 24 14:39:58 openHABianDevice systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Apr 24 14:39:58 openHABianDevice systemd[1]: evcc.service: Failed with result 'exit-code'.
Apr 24 14:39:58 openHABianDevice systemd[1]: Stopped evcc.
Apr 24 14:39:58 openHABianDevice systemd[1]: Started evcc.
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main  ] INFO 2022/04/24 14:39:58 evcc 0.90
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main  ] INFO 2022/04/24 14:39:58 using config file /etc/evcc.yaml
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main  ] INFO 2022/04/24 14:39:58 listening at :7070
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site  ] INFO 2022/04/24 14:40:00 site config:
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site  ] INFO 2022/04/24 14:40:00   meters:      grid ✓ pv ✗ battery ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site  ] INFO 2022/04/24 14:40:00     grid:      power ✓ energy ✓ currents ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00 loadpoint 1:
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00   mode:        pv
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00   charger:     power ✓ energy ✗ currents ✓ phases ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00   meters:      charge ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00     charge:    power ✓ energy ✗ currents ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00   vehicles:    ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00     vehicle 1: range ✓ finish ✓ status ✓ climate ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1  ] INFO 2022/04/24 14:40:00 vehicle updated: unknown -> Enterprise2

thierolm, 2022-04-24:
Welcher Teil aus dem Log ist denn der relevante? Ich sehe ganz viele Abbrüche und Restarts?

Hast du meter, charger und vehicle mal einzeln getestet, ohne den den Service laufen zu lassen?
Siehe: https://docs.evcc.io/docs/installation/linux#testen

Aus dem Log ab 14:39:58 lässt sich nicht viel ableiten. Allerdings scheint dein Tesla nicht sauber erkannt zu werden.
Und setze in deiner `evcc.yaml` den log-level mal von `info` auf `debug`.
 
```bash
Apr 24 14:39:58 openHABianDevice systemd[1]: Started evcc.
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main ] INFO 2022/04/24 14:39:58 evcc 0.90
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main ] INFO 2022/04/24 14:39:58 using config file /etc/evcc.yaml
Apr 24 14:39:58 openHABianDevice evcc[25142]: [main ] INFO 2022/04/24 14:39:58 listening at :7070
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site ] INFO 2022/04/24 14:40:00 site config:
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site ] INFO 2022/04/24 14:40:00 meters: grid ✓ pv ✗ battery ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [site ] INFO 2022/04/24 14:40:00 grid: power ✓ energy ✓ currents ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 loadpoint 1:
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 mode: pv
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 charger: power ✓ energy ✗ currents ✓ phases ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 meters: charge ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 charge: power ✓ energy ✗ currents ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 vehicles: ✓
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 vehicle 1: range ✓ finish ✓ status ✓ climate ✗
Apr 24 14:40:00 openHABianDevice evcc[25142]: [lp-1 ] INFO 2022/04/24 14:40:00 vehicle updated: unknown -> Enterprise2
```

knight1009, 2022-04-24:
Ich werde morgen mal ein neues Log posten. Ich gebe zu, dass ich viel mit dem System gespielt und getestet habe und dann neu gestartet habe. 
Der Tesla scheint zu schlafen. Verbinde ich mich per App, klappt anschließend wieder die Verbindung per evcc. 
Leider geht bis dahin auch der aktuelle Ladestand verloren. 
Ich teste weiter 👍

knight1009, 2022-04-24:
<Hast du meter, charger und vehicle mal einzeln getestet, ohne den Service laufen zu lassen?

Erfolgreich ausgeführt. Wecke ich den Tesla per App, funktioniert auch dieser prima.
Zwischenzeitlich dachte ich, das Problem mit dem calc-plugin zu lösen. Ich stell mal meine Lösung ein, die nicht funktioniert(-en kann):

#- type: custom
#       power: Power
#       usage: pv
#       source: calc
#       add:
#       - source: wallbox3
#       - source: grid1
#       scale: -1
#       name: pv1

Ziel ist es also, vorhanden, eingebundene Geräte und dessen Werte zu addieren.

premultiply, 2022-04-24:
Zu was sollte das gut sein?
Die Geräte die unter `site.pvs` aufgeführt werden hinsichtlich power addiert und als Erzeugung behandelt.

Wallbox und grid zu addieren macht wenig eher wenig Sinn, oder?

knight1009, 2022-04-24:
Da hast du recht.  Ich wollte subtrahieren. 
Mein "Problem" ist ja oben beschrieben.  Fzg lädt mit 4 kw, 2kw kommen davon aus dem Netz, zwei weitere von der PV. 
Das erkennt evcc aber nicht, da ich den erzeugten Strom ja nicht separat messen kann. Laut evcc beziehe ich 2kw aus dem Netz und lade trotzdem mit 4kw

knight1009, 2022-04-25:
Hallo, aktuell Lade ich das Fahrzeug.. Habe das Setup soweit angepasst, dass ich nun 1-phasig lade.
Im ersten Screenshot sieht man mein Problem vielleicht schon ganz gut. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

premultiply, 2022-04-25:
Ladepunkt ohne Zähler und nur Gridmeter?

evcc meter?
evcc charger?
Konfig?

knight1009, 2022-04-25:
Ich lade über den GoECharger... Der hat einen Zähler. Die Leistung wird auf der Seite ja auch angezeigt...
Was fehlt wäre die Berechnung der Solarleistung, wenn kein separater Zähler für die PV-Anlage vorhanden ist.
Wie in meinem Beispiel: Wenn ich das Auto mit 4 kW lade, aus dem Netz 2 kW beziehe, dann müssen ja wohl 2 (weitere) kW von der PV-Anlage kommen. 
Hatte gehofft das ich einfach selbst die Werte subtrahieren kann, habe aber keinen Plan wie...

knight1009, 2022-04-25:
> https://docs.evcc.io/docs/guides/faq/#ich-habe-eine-pv-anlage-aber-keinen-netzanschluss-z%C3%A4hler-kann-ich-evcc-trotzdem-nutzen

Gute Idee, aber ich nutze das schon in dieser Weise: 
site:
  residualPower: -1000 # 1000W Netzbezug im PV Modus erlaubt 

Ansonsten reicht die Leistung meiner PV nicht zum Laden aus ;)

knight1009, 2022-04-25:
> Nein, denn der sonstige Hausverbrauch ist völlig unbekannt.

Ändert aber nichts daran, dass, bleiben wir mal bei meinem Beispiel, mindestens 2 kW von der PV-Anlage produziert werden. Vielleicht mehr. Aber definitiv nicht Null.

premultiply, 2022-04-25:
Vielleicht. Ungefähr. Wer weiß.
Sagen wir zirka unbekannt.

thierolm, 2022-04-25:
@knight1009 : Was hast du denn konkret für einen Wechselrichter? Warum lässt der sich nicht auslesen?

knight1009, 2022-04-25:
@thierolm Ich habe einen SMA Sunnyboy 4000TL. Ca 11 Jahre alt. Der hat nur Bluetooth und optional RS485.

premultiply, 2022-04-25:
RS485 reicht doch für Modbus.
Ansonsten Zähler dran, fertig.
Shelly 3EM oder ein einfacher Modbus-Zähler reicht da völlig.

thierolm, 2022-04-25:
Du könntest dir überlegen, ob du die PV Leistung dann nicht einfach mit einem Shelly EM misst ...

Noch ein Cent von mir zu deiner Site `residualpower` Config:

M.E. "missbrauchst" du den Zweck dieses Parameters und solltest ihn mit einem positiven Wert auf deinen Durchschnitts Hausverbrauch setzen.
Den "Offset" von -1000W zum Start des chargers solltest du über den Loadpoint Parameter `enable.threshold` setzen.
Siehe https://docs.evcc.io/docs/reference/configuration/loadpoints#enable
=> Spiel/Test mal damit ...

knight1009, 2022-04-25:
@thierolm  Ich hab jetzt mal nach deinem Vorschlag ein wenig gespielt... Obst funktioniert merke ich dann morgen, wenn die Sonne wieder scheint.
Aber so ganz klar ist mit der Punkt mit residualpower noch nicht. Hab den jetzt wie vorgeschlagen auf einen Durchnittsverbrauch von 400 W eingestellt. Aber eigentlich messe ich doch den Verbrauch mit dem Shelly 3EM? Oder stelle ich mit dem Parameter immer den "internen" Hausverbrauch ein, unabhängig von ob Strom bezogen oder erzeugt wird?