evcc-io/evcc #3269: NRGKick Bluetooth

nauti23, 2022-04-28:
### Describe the bug

After config wizzard bluetooth connection to NRGKick wont work.
Reading from NRG seems to work, if i or evcc tries to send stop/start commands error 
"out of sync" and following see below...

### Steps to reproduce

see above, normal start, status ok, 
seems to write charging-mode to NRGkick wont work...

### Configuration details

interval: 25s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.100.178.70
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.100.178.70
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.100.178.70
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: nrgkick-bluetooth
  mac: 00:1E:C0:3D:FC:15
  pin: ****
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

### Log details

Errors:
vor 1 Sekunde
charge meter: find device: Operation already in progress
vor 13 Sekunden
charger: find device: Properties.GetAll org.bluez.Device1: Method \"GetAll\" with signature \"s\" on interface \"org.freedesktop.DBus.Properties\" doesn't exist
vor 19 Sekunden
charge meter: characteristic not found
vor 31 Sekunden
charger out of sync: expected enabled, got disabled

in console:
jo@evcc:~ $ sudo evcc -l debug charger
[main  ] INFO 2022/04/28 22:13:01 evcc 0.90
[main  ] INFO 2022/04/28 22:13:01 using config file /etc/evcc.yaml
Power:  Pair failed: Page Timeout
Energy: find device: Properties.GetAll org.bluez.Device1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

Current L1..L3: find device: Operation already in progress
Charge status:  find device: Operation already in progress
Enabled:        find device: Operation already in progress

jo@evcc:~ $ sudo evcc -l debug charger
[main  ] INFO 2022/04/28 22:14:36 evcc 0.90
[main  ] INFO 2022/04/28 22:14:36 using config file /etc/evcc.yaml
WARN[0002] Recovering from panic: send on closed channel 
Power:          0W
Energy:         2299.6kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false

### What type of operating system are you running?

Linux

### Version

0.9 on rasperry pi nano

nauti23, 2022-04-28:
additional info: 
NRGKick bluetooth 32A Version from 2017

premultiply, 2022-04-28:
nrgkick is sloooooooow. Make sure `interval` is set to 18s or higher.

nauti23, 2022-04-29:
Intervall is now on 45s … 
Same: out of sync no Effect of EVCC commands

premultiply, 2022-04-29:
Aaah, did you made sure that the app or any other evcc instance or any other device can not try to connect at the same time?

nauti23, 2022-04-29:
First @premultiply thanks for your Support. Really hope to get this working.
the Only other App is on smartphone an is Not Running… :/

nauti23, 2022-04-29:
[image]

nauti23, 2022-04-29:
[image]

premultiply, 2022-04-29:
Is the pin correct?
Avoid leading zeros here.
Maybe it could help to set a new pin with the app. 🤷🏻‍♂️

nauti23, 2022-04-29:
Ok… pin was correct but had one with leading 0… changed that and will Test again!

premultiply, 2022-04-29:
From my mind: Do not write leading zeros in config. So „0012“ should be `pin: 12` in this case.
If the pin does not match all write operations will fail but read operations are valid.