evcc-io/evcc #3296: Clarify Kostal Modbus requirements

andig, 2022-05-01:
> big-endian (ABCD) Sunspec

@premultiply der WR ist als Sunspec konfiguriert, wir lesen aber Folgendes aus:

```
template: kostal-ksem-inverter
...
render: |
  type: custom
  power:
    source: modbus # use ModBus plugin
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 252 # (see ba_kostal_interface_modbus-tcp_sunspec.pdf)
      type: holding
      decode: float32s # may be float32 on specific firmware/devices
```

Macht das Sinn??? Müssten wir nicht auch SunSpec lesen *oder* der WR anders konfiguriert sein?

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3293#discussioncomment-2669922_

premultiply, 2022-05-02:
Entweder konnten die Kostal-Geräte vorher kein SunSpec oder man kam oder kommt dann nicht an die Grid-Daten.

andig, 2022-05-02:
> Entweder konnten die Kostal-Geräte vorher kein SunSpec oder man kam oder kommt dann nicht an die Grid-Daten.

Ließe sich das irgendwo testen? In jedem Fall sollten wir im Template dokumentieren, wie der WR eingestellt sein muss (`CDAB` für `little endian` damit das `flaot32s` funktioniert?)

closed 2022-05-29 by github-actions[bot]