evcc-io/evcc #3315: HardyBarth ecb1 Charger logic error

artspe, 2022-05-03:
### Describe the bug

HardyBarth ecb1 Ladestation. Fehler beim Stoppen und PV-Laden über UI. Ladevorgang kann nicht gestoppt und gestartet werden. 
Nach Blick in die API ist es erforderlich explizit das Starten und Stoppen der Ladestation über einen Post auf http://apidoc.ecb1.de/api/v1/chargecontrols/1/start bzw. http://apidoc.ecb1.de/api/v1/chargecontrols/1/stop durchzuführen.

### Steps to reproduce

Ladevorgang starten - Auto and Wallbox anschließen
Ladevorgang über UI auf Modus Stop oder PV schalten

### Configuration details

- type: template
  template: kostal-plenticore
  id: 71
  host: plenticore.fritz.box
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: plenticore.fritz.box
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3

chargers:
- type: template
  template: hardybarth-ecb1
  host: ecb1.fritz.box
  name: wallbox4

### Log details

Mai 03 13:22:16 evcc evcc[160562]: [lp-1  ] WARN 2022/05/03 13:22:16 charger logic error: disabled but charging
Mai 03 13:22:27 evcc evcc[160562]: [lp-1  ] WARN 2022/05/03 13:22:27 charger logic error: disabled but charging

### What type of operating system are you running?

Linux

### Version

0.91

andig, 2022-05-05:
@artspe hättest Du dazu bitte mal ein Log mit 

```
levels:
  http: trace
```

Die spannende Frage- wenn wir das ändern- ist, woran ich erkennen kann, ob die Ladung gestoppt ist? Bisher erfolgt das durch Check von `chargecontrols/manual_mode_amps`?

artspe, 2022-05-05:
[evcc.log](https://github.com/evcc-io/evcc/files/8633123/evcc.log)

Anbei das Log mit einmal Ladevorgang starten und stoppen über die UI. Ich habe mal mit Postman den Befehl zum setzen des manualmodeamp (/chargecontrols/1/mode/manual/ampere) versucht auf 0 Ampere zu setzen. Das wird anscheinend vom Ladecontroller ingnoriert und die Wallbox lädt weiter mit min. 6 Ampere. So auch in der UI der Wallbox sichtbar. Nur wenn ich über api/v1/chargecontrols/1/stop die Ladefreigabe stoppe beendet die Box die Ladung. Ebenso Start nur über api/v1/chargecontrols/1/stop. Erkennung ob Ladefreigabe besteht müsste über den Key "stateid" gehen:

Ladefreigabe: stateid = 4
{
    "chargecontrol": {
        "manualmodeamp": 0,
        "modeid": 3,
        "stateid": 4,

Ladevorgang: stateid = 5
{
    "chargecontrol": {
        "manualmodeamp": 0,
        "modeid": 3,
        "stateid": 5,

Ladeverbot: stateid = 17
{
    "chargecontrol": {
        "manualmodeamp": 0,
        "modeid": 3,
        "stateid": 17,

andig, 2022-05-07:
@artspe das braucht nochmal ein bisschen Analyse+Testen. Könntest Du für info@evcc.io eine Portweiterleitung zum Test einrichten?

artspe, 2022-05-08:
Portfowarding ist eingerichtet und Mail ist raus.