evcc-io/evcc #3336: Fehlermeldung bei Start von Version 0.91

userjp123, 2022-05-06:
### Describe the bug

Fehlermeldung bei Start von Version 0.91. 
Version 0.90 läuft

### Steps to reproduce

Start von V0.91 mit der unten angegebenen Konfiguration.

### Configuration details

meters:
- name: chargemeter1
  type: custom
  power:
    source: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 2
    value: Power
  energy:
    source: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 2
    value: Sum
  currents:
  - source: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 2
    value: CurrentL1
  - source: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 2
    value: CurrentL2
  - source: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 2
    value: CurrentL3

### Log details

[main  ] FATAL 2022/05/06 08:46:46 cannot create meter 'chargemeter1': cannot create meter 'custom': power: not a SunSpec device

### What type of operating system are you running?

Linux

### Version

0.91

andig, 2022-05-07:
Ich kann den Fehler nachvollziehen, Rest muss ich mir anschauen.

andig, 2022-05-07:
Fixed in https://github.com/evcc-io/evcc/commit/f327b5bb35520f2300c7070711d85975615a2f6b

closed 2022-05-07 by andig

userjp123, 2022-05-07:
Funktioniert :-)
Vielen Dank !