evcc-io/evcc #3353: Threshold for Standby Power in Visualisation.

naltatis, 2022-05-09:
Mir gefällt das Design immer noch hervorragend - selbst nach einer Woche! 😁 
In meinem Fall (bin da vielleicht auch eine Ausnahme) zieht die WB im Standby auch etwas Strom und das erscheint dann in der UI als kleiner Punkt. Ich verstehe die Idee des Punktes, aber evtl. macht es Sinn, unterhalb bestimmter Leistungen den jeweiligen Verbraucher in der Grafik vielleicht ganz zu unterschlagen?
[image]

_Originally posted by @Alestrix in https://github.com/evcc-io/evcc/issues/2889#issuecomment-1120478942_

thierolm, 2022-05-09:
@premultiply Bei den Smarthome Steckdosen ignorieren wir schon über den standbypower Parameter niedrige Ströme.
Siehe bspw: https://github.com/evcc-io/evcc/blob/a7f9f8bfda45f9da97bd480ff2edcdecb7e38365/charger/fritzdect.go#L145

closed 2022-06-05 by github-actions[bot]

Alestrix, 2022-06-05:
Ist das jetzt im Code geändert? Ich hab das bei mir zwischenzeitlich durch ne jquery in der evcc Config gelöst.