evcc-io/evcc #3356: Rework modbus configuration

andig, 2022-05-11:
- [ ] Use connection types: rs485/ethernet
- [ ] Use modus communication types: rtu/tcp

premultiply, 2022-05-11:
ethernet ➡️ ip ?
Weil geht ja auch via Wifi, Powerlan, DSL usw.

andig, 2022-05-11:
Relevant ist doch erstmal die physikalische Verbindung am Zähler. Und die ist Ethernet. Da mögen weitere Typen kommen, heute haben wir RS485 und Ethernet.

premultiply, 2022-05-11:
Ethernet wäre nur Kabel.
Hier ist aber nicht Ethernet (Transportmedium, OSI Layer 1 und 2) sondern TCP/IP (Layer 3/4) gemeint.
Das physikalische Transportmedium spielt keine Rolle.
Ethernet ist quasi nur ein Fall, WiFi wäre z.B. ein anderer.
Am Ende geht aber nur um die IP-Verbindung.

andig, 2022-05-11:
Stimmt natürlich. Wenn ich auf aktuelle Zähler die EM24 schaue gibts die aber in Varianten mit RS485 und Ethernet (=physische SSt.). Wäre gut wenn uns was Konsistentes UND Einfaches einfallen würde.

premultiply, 2022-05-11:
Ja, absolut.
Ich glaube die ursprüngliche Intension war hier auf die physikalischen Schnittstellen der Geräte zu schauen um es für den Anwender verständlicher zu halten...

andig, 2022-05-19:
Hab keine gute Idee wie wir das konsistent besser machen könnten. Close.

closed 2022-05-19 by andig