evcc-io/evcc #3359: Bender Template not available

Trivianni, 2022-05-12:
### Describe the bug

Hello,
I tried to connect my Webasto Live Wallbox to evcc. I used the predefined templates but I get a problem after debug.
I've added a screenshot of message and configuration.

Maybe someone can give me a hint.

Thanks and regards,
Christian

### Steps to reproduce

use configuration and start evcc --log debug

### Configuration details

chargers:
- type: template
  template: bender 
  host: x.x.x.x  
  port: 1502  
  name: wallbox4

loadpoints:
- title: Wallbox
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

### Log details

[main  ] INFO 2022/05/12 06:27:04 evcc 0.91
[main  ] INFO 2022/05/12 06:27:04 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/05/12 06:27:04 listening at :7070
[modbus] WARN 2022/05/12 06:27:05 missing modbus value or register - assuming Power
[main  ] FATAL 2022/05/12 06:27:06 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'bender': modbus: exception '1' (illegal function), function '3'

### What type of operating system are you running?

Linux

premultiply, 2022-05-12:
Hi, this is a known problem.
Maybe you can assist me to finally fix it by providing me access to to the modbus port of the charger.
Here is a draft PR https://github.com/evcc-io/evcc/pull/3171 to fix it but it has to be verified and tested. Nobody was able to provide me access yet. :(

Maybe you can contact me on slack channel?

andig, 2022-05-12:
...or send access information to info@evcc.io

closed 2022-05-13 by andig