evcc-io/evcc #3367: [Tasmota] Disabled but charging

sveba, 2022-05-13:
### Describe the bug

Tasmota-Relais wird nicht ausgeschaltet.

### Steps to reproduce

Loadpoint mit Tasmota als Charger definieren. Bei erreichen des Thresholds wird die Steckdose nicht ausgeschaltet. Auch beim Klick auf "Stop" passiert nichts.
  

### Configuration details

```
loadpoints:
  - title: Garage 
    charger: easee 
    soc:
      target: 80 
    vehicles:
      - mini 
    mode: pv 
  - title: 3D
    charger: 3d-Drucker
    mode: pv
    phases: 1
    mincurrent: 8
    enable:
      threshold: -7000
      # delay: 5m
chargers:
  - name: easee
    type: easee
    user: xxxx
    password: xxxx
    charger: xxx
  - name: 3d-Drucker
    type: template
    template: tasmota
    host: 192.168.10.245 # IP-Adresse oder Hostname
    standbypower: 0
```

### Log details

```
[tasmota] TRACE 2022/05/13 10:58:54 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 10:58:55 {"StatusSNS":{"Time":"2022-05-13T10:58:55","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.307,"Yesterday":0.251,"Today":0.116,"Power":11,"ApparentPower":25,"ReactivePower":22,"Factor":0.43,"Voltage":227,"Current":0.109}}}
[lp-2  ] DEBUG 2022/05/13 10:58:55 charge power: 11W
[tasmota] TRACE 2022/05/13 10:59:24 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 10:59:25 {"StatusSNS":{"Time":"2022-05-13T10:59:25","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.307,"Yesterday":0.251,"Today":0.116,"Power":11,"ApparentPower":25,"ReactivePower":23,"Factor":0.42,"Voltage":227,"Current":0.110}}}
[lp-2  ] DEBUG 2022/05/13 10:59:25 charge power: 11W
[tasmota] TRACE 2022/05/13 10:59:25 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 10:59:25 {"StatusSNS":{"Time":"2022-05-13T10:59:26","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.307,"Yesterday":0.251,"Today":0.116,"Power":11,"ApparentPower":25,"ReactivePower":23,"Factor":0.42,"Voltage":227,"Current":0.110}}}
[tasmota] TRACE 2022/05/13 10:59:25 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 10:59:26 {"StatusSNS":{"Time":"2022-05-13T10:59:26","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.307,"Yesterday":0.251,"Today":0.116,"Power":11,"ApparentPower":25,"ReactivePower":22,"Factor":0.43,"Voltage":227,"Current":0.109}}}
[lp-2  ] DEBUG 2022/05/13 10:59:26 charger status: C
[tasmota] TRACE 2022/05/13 10:59:26 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/13 10:59:26 {"Status":{"Module":0,"DeviceName":"3D_Drucker","FriendlyName":["3D_Drucker","3D_Drucker_Webcam"],"Topic":"3d_drucker_0A59CD","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T18:42:10","StartupUTC":"2022-05-12T14:17:16","Sleep":50,"CfgHolder":4617,"BootCount":3385,"BCResetTime":"2020-10-14T13:31:39","SaveCount":6950,"SaveAddress":"F4000"},"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:13:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"459/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iotatbatchovski","Sharimob"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C80001000680003C5A0A000000000000","00000281","00006000","00004000"]},"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"3d-drucker-0A59CD-6605","IPAddress":"192.168.10.245","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.8.1.6","DNSServer2":"192.168.10.2","Mac":"F4:CF:A2:0A:59:CD","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0},"StatusMQT":{"MqttHost":"iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0A59CD","MqttUser":"DVES_USER","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-05-13T08:59:26","Local":"2022-05-13T10:59:26","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"06:11","Sunset":"21:21"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2022-05-13T10:59:26","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.307,"Yesterday":0.251,"Today":0.116,"Power":11,"ApparentPower":25,"ReactivePower":22,"Factor":0.43,"Voltage":227,"Current":0.109}},"StatusSTS":{"Time":"2022-05-13T10:59:26","Uptime":"0T18:42:10","UptimeSec":67330,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":4,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"iotatbatchovski","BSSId":"D2:21:F9:17:BC:41","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:22"}}}
[lp-2  ] WARN 2022/05/13 10:59:26 charger logic error: disabled but charging
```

### What type of operating system are you running?

Docker container

### Version

0.91

VolkerK62, 2022-05-13:
https://docs.evcc.io/docs/devices/chargers#schaltbare-steckdosen
Standbypower = 0 dürfte das Problem sein

sveba, 2022-05-13:
Hm, 
hab jetzt `standbypower: -5`
Resultat ist: es wird nicht mehr der aktuelle Load angezeigt und die Steckdose bleibt weiter `ON` auch wenn ich auf `Stop` in der UI klicke.
Hier auch Log nach Klick auf `Stop`:
```
[lp-2  ] DEBUG 2022/05/13 12:00:49 charge power: 0W
[lp-2  ] DEBUG 2022/05/13 12:00:54 set charge mode: off
[lp-2  ] DEBUG 2022/05/13 12:00:54 pv timer elapse
[lp-2  ] DEBUG 2022/05/13 12:00:54 pv timer inactive
[tasmota] TRACE 2022/05/13 12:00:54 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/13 12:00:55 {"Status":{"Module":0,"DeviceName":"3D_Drucker","FriendlyName":["3D_Drucker","3D_Drucker_Webcam"],"Topic":"3d_drucker_0A59CD","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T19:43:39","StartupUTC":"2022-05-12T14:17:16","Sleep":50,"CfgHolder":4617,"BootCount":3385,"BCResetTime":"2020-10-14T13:31:39","SaveCount":6950,"SaveAddress":"F4000"},"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:13:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"459/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iotatbatchovski","Sharimob"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C80001000680003C5A0A000000000000","00000281","00006000","00004000"]},"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"3d-drucker-0A59CD-6605","IPAddress":"192.168.10.245","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.8.1.6","DNSServer2":"192.168.10.2","Mac":"F4:CF:A2:0A:59:CD","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0},"StatusMQT":{"MqttHost":"iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0A59CD","MqttUser":"DVES_USER","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2022-05-13T10:00:55","Local":"2022-05-13T12:00:55","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"06:11","Sunset":"21:21"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2022-05-13T12:00:55","ENERGY":{"TotalStartTime":"2020-10-08T14:38:53","Total":79.318,"Yesterday":0.251,"Today":0.127,"Power":10,"ApparentPower":25,"ReactivePower":23,"Factor":0.42,"Voltage":223,"Current":0.111}},"StatusSTS":{"Time":"2022-05-13T12:00:55","Uptime":"0T19:43:39","UptimeSec":71019,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"iotatbatchovski","BSSId":"D2:21:F9:17:BC:41","Channel":9,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:22"}}}
[lp-2  ] DEBUG 2022/05/13 12:00:55 charge power: 0W
[tasmota] TRACE 2022/05/13 12:00:55 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/13 12:00:55 {"Status":{"Module":0,"DeviceName":"3D_Drucker","FriendlyName":["3D_Drucker","3D_Drucker_Webcam"],"Topic":"3d_drucker_0A59CD","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T19:43:39","StartupUTC":"2022-05-12T14:17:16","Sleep":50,"CfgHolder":4617,"BootCount":3385,"BCResetTime":"2020-10-14T13:31:39","SaveCount":6950,"SaveAddress":"F4000"},"StatusFWR":{"Version":"11.0.0(tasmota)","BuildDateTime":"2022-02-12T14:13:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"459/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iotatbatchovski","Sharimob"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C80001000680003C5A0A000000000000","00000281","00006000","00004000"]},"StatusMEM":{"ProgramSize":620,"Free":380,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"3d-drucker-0A59CD-6605","IPAddress":"192.168.10.245","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.8.1.6","DNSServer2":"192.168.10.2","Mac":"F4:CF:A2:0A:59:CD","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0},"StatusMQT":{"MqttHost":"iot","M [... cut ...] Today":0.127,"Power":11,"ApparentPower":25,"ReactivePower":23,"Factor":0.42,"Voltage":223,"Current":0.111}},"StatusSTS":{"Time":"2022-05-13T12:00:55","Uptime":"0T19:43:39","UptimeSec":71019,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":4,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"iotatbatchovski","BSSId":"D2:21:F9:17:BC:41","Channel":9,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:22"}}}
[lp-2  ] DEBUG 2022/05/13 12:00:55 charger status: B```

sveba, 2022-05-13:
Hab mich übrigens daran orientiert:
https://github.com/evcc-io/evcc/discussions/2935#discussioncomment-2581328

VolkerK62, 2022-05-13:
"Charger status: B" zeigt eigentlich an, dass nicht geladen wird. Warum die Steckdose trotzdem "on" bleibt ... keine Ahnung.

andig, 2022-05-13:
/cc @thierolm

thierolm, 2022-05-13:
@sveba Da müsstest du mal etwas mehr trace output liefern.
Aktualisiere auch die Tasmota Version, bei mir läuft die 11.1 .

Ich kann den Fehler mit dem aktuellen evcc Nightly nicht nachstellen. Bei mir funktioniert der Switch im Static Mode einwandfrei (hab mit standbypower = -1111 getestet).

Wenn möglich teste auch mal außerhalb von docker.

```bash
thierolm@accffmmt:~/evcc$ ./evcc -l debug -c ./evcc_tasmotatest.yaml
[main  ] INFO 2022/05/13 14:49:37 evcc 0.91 (cf9bf6fa)
[main  ] INFO 2022/05/13 14:49:37 using config file ./evcc_tasmotatest.yaml
[main  ] INFO 2022/05/13 14:49:37 listening at :7070
[site  ] INFO 2022/05/13 14:49:37 site config:
[site  ] INFO 2022/05/13 14:49:37   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/05/13 14:49:37     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/05/13 14:49:37     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/05/13 14:49:37     pv 2:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/05/13 14:49:37 loadpoint 1:
[lp-1  ] INFO 2022/05/13 14:49:37   mode:        off
[lp-1  ] INFO 2022/05/13 14:49:37   charger:     power ✓ energy ✓ currents ✗ phases ✗
[lp-1  ] INFO 2022/05/13 14:49:37   meters:      charge ✓
[lp-1  ] INFO 2022/05/13 14:49:37     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2022/05/13 14:49:37   vehicles:    ✓
[lp-1  ] INFO 2022/05/13 14:49:37     vehicle 1: range ✗ finish ✗ status ✗ climate ✗
[lp-1  ] INFO 2022/05/13 14:49:37 vehicle updated: unknown -> Zero S
[site  ] DEBUG 2022/05/13 14:49:37 ----
[lp-1  ] DEBUG 2022/05/13 14:49:37 charge power: 0W
[site  ] DEBUG 2022/05/13 14:49:37 pv power: 4039W
[site  ] DEBUG 2022/05/13 14:49:37 grid power: -3126W
[site  ] DEBUG 2022/05/13 14:49:37 site power: -3126W
[lp-1  ] DEBUG 2022/05/13 14:49:37 charger status: B
[lp-1  ] DEBUG 2022/05/13 14:49:37 vehicle soc: 95%
[site  ] DEBUG 2022/05/13 14:49:47 ----
[lp-1  ] DEBUG 2022/05/13 14:49:47 charge power: 0W
[site  ] DEBUG 2022/05/13 14:49:47 pv power: 4070W
[site  ] DEBUG 2022/05/13 14:49:47 grid power: -3130W
[site  ] DEBUG 2022/05/13 14:49:47 site power: -3130W
[lp-1  ] DEBUG 2022/05/13 14:49:47 charger status: B
[site  ] DEBUG 2022/05/13 14:49:57 ----
[lp-1  ] DEBUG 2022/05/13 14:49:57 charge power: 0W
[site  ] DEBUG 2022/05/13 14:49:57 pv power: 4159W
[site  ] DEBUG 2022/05/13 14:49:57 grid power: -3267W
[site  ] DEBUG 2022/05/13 14:49:57 site power: -3267W
[lp-1  ] DEBUG 2022/05/13 14:49:57 charger status: B
[lp-1  ] DEBUG 2022/05/13 14:50:04 set charge mode: now
[lp-1  ] DEBUG 2022/05/13 14:50:04 pv timer elapse
[lp-1  ] DEBUG 2022/05/13 14:50:04 pv timer inactive
[site  ] DEBUG 2022/05/13 14:50:04 ----
[lp-1  ] DEBUG 2022/05/13 14:50:04 charge power: 0W
[site  ] DEBUG 2022/05/13 14:50:04 pv power: 4217W
[site  ] DEBUG 2022/05/13 14:50:04 grid power: -3343W
[site  ] DEBUG 2022/05/13 14:50:04 site power: -3343W
[lp-1  ] DEBUG 2022/05/13 14:50:05 charger status: B
[lp-1  ] DEBUG 2022/05/13 14:50:05 max charge current: 7A
[lp-1  ] DEBUG 2022/05/13 14:50:05 charger enable
[lp-1  ] DEBUG 2022/05/13 14:50:05 wake-up timer: start
[site  ] DEBUG 2022/05/13 14:50:07 ----
[lp-1  ] DEBUG 2022/05/13 14:50:07 charge power: 1111W
[site  ] DEBUG 2022/05/13 14:50:07 pv power: 4250W
[site  ] DEBUG 2022/05/13 14:50:07 grid power: -3373W
[site  ] DEBUG 2022/05/13 14:50:07 site power: -3373W
[lp-1  ] DEBUG 2022/05/13 14:50:07 charger status: C
[lp-1  ] DEBUG 2022/05/13 14:50:07 charge start energy: 142kWh
[lp-1  ] INFO 2022/05/13 14:50:07 start charging ->
[lp-1  ] DEBUG 2022/05/13 14:50:07 vehicle soc: 95%
[site  ] DEBUG 2022/05/13 14:50:17 ----
[lp-1  ] DEBUG 2022/05/13 14:50:17 charge power: 1111W
[site  ] DEBUG 2022/05/13 14:50:18 pv power: 4404W
[site  ] DEBUG 2022/05/13 14:50:18 grid power: -3518W
[site  ] DEBUG 2022/05/13 14:50:18 site power: -3518W
[lp-1  ] DEBUG 2022/05/13 14:50:18 charger status: C
[lp-1  ] DEBUG 2022/05/13 14:50:18 vehicle soc: 95%
[lp-1  ] DEBUG 2022/05/13 14:50:23 set charge mode: off
[lp-1  ] DEBUG 2022/05/13 14:50:23 pv timer elapse
[lp-1  ] DEBUG 2022/05/13 14:50:23 pv timer inactive
[site  ] DEBUG 2022/05/13 14:50:23 ----
[lp-1  ] DEBUG 2022/05/13 14:50:23 charge power: 1111W
[site  ] DEBUG 2022/05/13 14:50:23 pv power: 4456W
[site  ] DEBUG 2022/05/13 14:50:23 grid power: -3578W
[site  ] DEBUG 2022/05/13 14:50:23 site power: -3578W
[lp-1  ] DEBUG 2022/05/13 14:50:23 charger status: C
[lp-1  ] DEBUG 2022/05/13 14:50:23 vehicle soc: 95%
[lp-1  ] DEBUG 2022/05/13 14:50:23 charger disable
[lp-1  ] DEBUG 2022/05/13 14:50:23 wake-up timer: stop
[site  ] DEBUG 2022/05/13 14:50:27 ----
[lp-1  ] DEBUG 2022/05/13 14:50:27 charge power: 0W
[site  ] DEBUG 2022/05/13 14:50:27 pv power: 4488W
[site  ] DEBUG 2022/05/13 14:50:27 grid power: -3598W
[site  ] DEBUG 2022/05/13 14:50:27 site power: -3598W
[lp-1  ] DEBUG 2022/05/13 14:50:27 charger status: B
[lp-1  ] DEBUG 2022/05/13 14:50:28 final charge energy: 0kWh
[lp-1  ] INFO 2022/05/13 14:50:28 stop charging <-
[lp-1  ] DEBUG 2022/05/13 14:50:28 vehicle soc: 95%
[site  ] DEBUG 2022/05/13 14:50:37 ----
[lp-1  ] DEBUG 2022/05/13 14:50:37 charge power: 0W
[site  ] DEBUG 2022/05/13 14:50:37 pv power: 4511W
[site  ] DEBUG 2022/05/13 14:50:37 grid power: -3650W
[site  ] DEBUG 2022/05/13 14:50:37 site power: -3650W
[lp-1  ] DEBUG 2022/05/13 14:50:37 charger status: B
```

sveba, 2022-05-13:
Upgrade auf Tasmota 11.1.0 gemacht - keine Änderung
@thierolm 

1. Siehst du den current Load, wenn du `standypower=-1111` setzst?
2. Hast du dir meinen Konfig angeschaut? Vllt. mache ich da was falsch? Ich habe kein Vehicle dem Ladepunkt zugeordnet.
3. Kannst du mir bitte deinen Konfig zukommen lassen? Dann kann ich vergleichen  
4. Wenn ich log Level für `tasmota` auf trace setze, kann ich die HTTP-Requests sehen, und EVCC macht bei mir nichts ausser Status zu holen:

```
docker logs -f evcc | grep GET                                                                                                                                                  
[tasmota] TRACE 2022/05/13 15:21:06 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/13 15:21:06 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:21:36 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:21:36 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:21:37 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2022/05/13 15:22:06 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:22:36 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:22:36 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:22:36 GET http://192.168.10.245/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2022/05/13 15:22:36 GET http://192.168.10.245/cm?cmnd=Status+0&password=&user=
```

sveba, 2022-05-13:
Irgendwas stimmt da nicht. Hab jetzt den umgekehrten Fall getestet: einzuschalten. Also Ausgangsposition ist Relais ist off und threshold auf -1000. Momentan kommt von der PV-Anlage ca. 8000. Das hat den funktioniert, aber mit Fehlermeldung:
```
[lp-2  ] DEBUG 2022/05/13 15:41:02 pv charge current: 30.6A = 0A + 30.6A (-7031W @ 1p)
[lp-2  ] DEBUG 2022/05/13 15:41:02 site power -7031W <= -1000W enable threshold
[lp-2  ] DEBUG 2022/05/13 15:41:02 pv enable in 0s
[lp-2  ] DEBUG 2022/05/13 15:41:02 pv enable timer elapsed
[tasmota] TRACE 2022/05/13 15:41:02 GET http://192.168.10.245/cm?cmnd=Power+on&password=&user=
[tasmota] TRACE 2022/05/13 15:41:02 {"Time":"2022-05-13T15:41:02","Uptime":"0T00:28:05","UptimeSec":1685,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"iotatbatchovski","BSSId":"D2:21:F9:17:BC:41","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"},"POWER1":"ON"}
[lp-2  ] ERROR 2022/05/13 15:41:02 charger enable: switchOn failed
```

thierolm, 2022-05-13:
Ich denke ich kenne das Problem. Was nutzt du denn für eine Tasmota Hardware?
Ich habe einen Sonof Pow R2 auf dessen Basis das Tasmota Modul entwickelt wurde.
In dieser Sonof Pow R2 Modul Config, liefert Tasmota im Response Json nur einen `POWER` Wert:
```json
{"POWER":"ON"}
```

Wenn ich die JSONs in deinem Trace-Log anschaue, liefert deine Dose POWER1 und POWER2 ... . Diese werden vom aktuellen Modul nicht gelesen ...

thierolm, 2022-05-13:
Laut deinem Status JSON, ist das Tasmota Module (https://tasmota.github.io/docs/Modules/) mit der Nummer 0 installiert
```json
{
  "Status": {
    "Module": 0,
    "DeviceName": "3D_Drucker",
    "FriendlyName": [
...
```
Schau mal, ob es nicht ein passenders Template gibt, was zu deiner Hardware passt, konfiguriere die und schicke dann mal die Json's, die deine Tasmota Dose/Schalter dann auf die folgenden Befehle im Browser zurückgibt:

- `http://192.168.178.95/cm?cmnd=Status+0` und
- `http://192.168.178.95/cm?cmnd=Power+off`

Ich prüfe dann, ob ich die evcc Logik entsprechend erweitern kann.

thierolm, 2022-05-13:
Hab mal den PR #3368 vorbereitet ...

andig, 2022-05-13:
Mir ist nicht so richtig klar, warum die Dose zwei Leistungen liefern sollte?

sveba, 2022-05-13:
@thierolm 
meine Steckdose hat 2 Relais (220v+5v USB) Deswegen sind Power1 und Power2 vorhanden. Es gibt übrigens welche mit 4 Channels
Vllt. kann man das über optionale Konfig-Parameter steuern!?

sveba, 2022-05-13:
@thierolm 
GET http://192.168.10.245/cm?cmnd=Power+off
```
{
   "Time":"2022-05-13T20:49:59",
   "Uptime":"0T05:37:02",
   "UptimeSec":20222,
   "Heap":24,
   "SleepMode":"Dynamic",
   "Sleep":50,
   "LoadAvg":19,
   "MqttCount":1,
   "POWER1":"OFF",
   "POWER2":"ON",
   "Wifi":{
      "AP":1,
      "SSId":"iotatbatchovski",
      "BSSId":"D2:21:F9:17:BC:41",
      "Channel":9,
      "Mode":"11n",
      "RSSI":70,
      "Signal":-65,
      "LinkCount":1,
      "Downtime":"0T00:00:04"
   }
}
```
GET http://192.168.10.245/cm?cmnd=Status+0
```
{
   "Status":{
      "Module":0,
      "DeviceName":"3D_Drucker",
      "FriendlyName":[
         "3D_Drucker",
         "3D_Drucker_Webcam"
      ],
      "Topic":"3d_drucker_0A59CD",
      "ButtonTopic":"0",
      "Power":2,
      "PowerOnState":3,
      "LedState":1,
      "LedMask":"FFFF",
      "SaveData":1,
      "SaveState":1,
      "SwitchTopic":"0",
      "SwitchMode":[
         0,
         0,
         0,
         0,
         0,
         0,
         0,
         0
      ],
      "ButtonRetain":0,
      "SwitchRetain":0,
      "SensorRetain":0,
      "PowerRetain":0,
      "InfoRetain":0,
      "StateRetain":0
   },
   "StatusPRM":{
      "Baudrate":115200,
      "SerialConfig":"8N1",
      "GroupTopic":"tasmotas",
      "OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz",
      "RestartReason":"Software/System restart",
      "Uptime":"0T05:40:23",
[... 340 lines cut ...]
   "StatusSTS":{
      "Time":"2022-05-13T20:54:37",
      "Uptime":"0T05:41:40",
      "UptimeSec":20500,
      "Heap":24,
      "SleepMode":"Dynamic",
      "Sleep":50,
      "LoadAvg":19,
      "MqttCount":1,
      "POWER1":"ON",
      "POWER2":"ON",
      "Wifi":{
         "AP":1,
         "SSId":"iotatbatchovski",
         "BSSId":"D2:21:F9:17:BC:41",
         "Channel":9,
         "Mode":"11n",
         "RSSI":70,
         "Signal":-65,
         "LinkCount":1,
         "Downtime":"0T00:00:04"
      }
   }
}
```

thierolm, 2022-05-13:
> meine Steckdose hat 2 Relais (220v+5v USB) Deswegen sind Power1 und Power2 vorhanden. Es gibt übrigens welche mit 4 Channels
> Vllt. kann man das über optionale Konfig-Parameter steuern!?

Ich sehe baue in den PR mal einen Relais Channel Üarameter ein (Default = 1) und teste morgen mit meinem POW ...

thierolm, 2022-05-13:
Soweit ich das in der Tasmota Doku sehe, werden bis zu 8 Relay Channels unterstützt:

[image]

Werde das mal so einbauen ...

thierolm, 2022-05-13:
Channel Parameter ist im PR #3368 eingebaut. Mit meinem POW funktionierts, d.h. Abwärtskompatibilität ist gegeben.

@sveba Wenn @andig den PR merged, könntest du mit dem nightly Release mal testen ...

thierolm, 2022-05-14:
@sveba Wäre klasse, wenn du die "Channel" Erweiterung für beide deiner Schaltausgänge testen könntest, sobald die Erweiterung verfügbar ist ... :-)

sveba, 2022-05-14:
Mache ich. Kann auch selber von deinem branch bauen. Leider etwas knapp die Zeit am Wochenende

sveba, 2022-05-16:
@thierolm  Jetzt kann ich berichten. Gebaut aus deinem Branch und getestet mit Channels
Channels funzen!
Allerdings nur enable Threshold wird mMn beachtet. Hab mit 2 Tasmotas getestet (1CH und 2CH). Bei beiden wurde disable-Threshold nicht ausgelöst. Hier mein Konfig für den Disable-Testfall:
```
loadpoints:
  - title: 3D
    charger: 3d-Drucker
    mode: pv
    phases: 1
    mincurrent: 10
    enable:
      threshold: -10000
      delay: 10s
    disable:
      threshold: -7000
      delay: 10s
chargers:
  - name: 3d-Drucker
    type: template
    template: tasmota
    host: 192.168.10.245 
    channel: 1
    standbypower: -2000
```
Und Log dazu:
```
[lp-1  ] DEBUG 2022/05/16 08:33:46 max charge current: 10A
[site  ] DEBUG 2022/05/16 08:33:46 ----
[lp-1  ] DEBUG 2022/05/16 08:33:46 charge power: 2000W
[site  ] DEBUG 2022/05/16 08:33:46 pv power: 6084W
[site  ] DEBUG 2022/05/16 08:33:46 grid power: -5086W
[site  ] DEBUG 2022/05/16 08:33:46 site power: -5586W
[lp-1  ] DEBUG 2022/05/16 08:33:46 charger status: C
[lp-1  ] DEBUG 2022/05/16 08:33:46 charge start energy: 80kWh
[lp-1  ] INFO 2022/05/16 08:33:46 start charging ->
[lp-1  ] DEBUG 2022/05/16 08:33:47 pv charge current: 34.3A = 10A + 24.3A (-5586W @ 1p)
[lp-1  ] DEBUG 2022/05/16 08:33:47 max charge current: 16A
```

thierolm, 2022-05-16:
@sveba Danke für die Testung!
Das Tasmota Charger Modul ist unabhängig von der Regelung des loadpoints. Der Loadpoint nutzt nur die Funktionen, die das Charger Modul bereit stellt.
D.h. du konntest manuell über die GUI über `Sofort` und `Aus` an- und ausschalten, richtig?
Dann funktionieren die Änderungen.

Dein Log oben geht auch nur über 2 Sekunden. Um herauszufinden was da schief läuft ist ein längeres Monitoring nötig mit. Ich würde auch mincurrent bei einer Steckdose auf 6A setzen und mit deinem gewünschten realen Szenario testen.

@andig : Den PR können wir m.E. dann mergen.

sveba, 2022-05-16:
Alles klar.
`Sofort` und `Aus` funzen auch bei mehreren Channels.  
Wegen `disable` Threshold mache ich ein neues Ticket.
Vielen Dank!

andig, 2022-05-18:
Gibts hier noch was zu tun?

thierolm, 2022-05-18:
M.E. nicht.

closed 2022-05-18 by andig