evcc-io/evcc #3377: Auto lädt nicht, obwohl Überschuss vorhanden

binderth, 2022-05-15:
### Discussed in https://github.com/evcc-io/evcc/discussions/3357

<div type='discussions-op-text'>

<sup>Originally posted by **binderth** May 11, 2022</sup>
Ich bin mir unsicher, ob ich das als Bug posten soll,daher erstmal hier:

evcc läuft auf einem Raspberry Pi auf bullseye in Version 0.91 mit dieser Config:

<details>

```
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  host: 0.0.0.0
  #host: evcc.local

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

interval: 10s # control cycle interval

log: info

meters:
- type: custom
  name: ErzeugungsLeistung
  power:
    source: http
    uri: http://localhost:8080/rest/items/ErzeugungsLeistung
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
- type: custom
  name: Netzeinspeisung
  power:
    source: http
    uri: http://localhost:8080/rest/items/Netzeinspeisung
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    scale: -1
- type: custom
  name: LadeleistungHaus
  power:
    source: http
    uri: http://localhost:8080/rest/items/LadeleistungHaus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
    scale: -1
  soc:
    source: http
    uri: http://localhost:8080/rest/items/SoCHaus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]

chargers:
- type: template
  template: tinkerforge-warp
  fw2: true
  host: 192.168.78.9
  port: 1883
  topic: warp2/pl12
  timeout: 30s
  name: WarpWallbox

vehicles:
- name: EV6
  type: custom
  title: Kia EV6 # display name for UI
  capacity: 77 # byttery capacity (kWh)
  soc: # battery soc (%)
    source: http
    uri: http://192.168.78.10:8080/rest/items/MQTTKia_batteryStatus 
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  range: # optional electric range (km)
    source: http
    uri: http://192.168.78.10:8080/rest/items/MQTTKia_availableRange 
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]

loadpoints:
- title: Warp
  charger: WarpWallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicles:
    - EV6

site:
  title: Zuhause
  meters:
    grid: Netzeinspeisung
    pvs:
    - ErzeugungsLeistung
    batteries:
    - LadeleistungHaus
```
</details>

ich habe im TRACE-Modus dieses ergebnis - da zu groß für hier, als Attachment
[evcc_log.txt](https://github.com/evcc-io/evcc/files/8668376/evcc_log.txt)</div>

binderth, 2022-05-15:
Habe das heute nochmal ausgiebig getestet und laut Logfiles wird der WARP2 zwar gesagt, mit wieviel Stromstärke sie laden soll, aber es wird nicht das "LADESTART" Signal gegeben.

siehe log.txt:
[log.txt](https://github.com/evcc-io/evcc/files/8694824/log.txt)

hier sehe ich z.B.:
```
[warp  ] TRACE 2022/05/15 13:24:06 send warp2/pl12/evse/external_current_update: '{ "current": 6000 }'
```
 aber es wird kein `warp2/pl12/evse/start_charging` gesendet.

andig, 2022-05-16:
@RTTTC könntest Du hier nochmal rein schauen? Wir setzen bei `Enable(true)` tatsächlich nur den Strom, genauso wie `Enable(false)` den Strom auf Null setzt. M.E. hatten wir das aber auch so getestet.

/cc @Alestrix Du hast ja auch die FW2 drauf?

andig, 2022-05-16:
@binderth: welche FW Version hast Du auf der Warp?

RTTTC, 2022-05-16:
> @RTTTC könntest Du hier nochmal rein schauen? 

Maybe it was not meant for me??

andig, 2022-05-16:
Ich denke die Ursache ist gefunden- deshalb "close". Danke @Alestrix und @rtrbt.

closed 2022-05-16 by andig

binderth, 2022-05-16:
> > ich habe Auto-Start eigentlich deswegen abgeschaltet, damit evcc für mich startet! ;)
> 
> @binderth Schau mal bei der Erklärung hier vorbei, dann wird vieles klarer: https://www.tinkerunity.org/topic/7453-warp-firmware-200-beta-2-und-warp2-firmware-200-beta-3-mit-ladetracking/?do=findComment&comment=40512

Ok, ich glaube, ich habe verstanden, wenn evcc "0" als `evse/external_current_update` schickt, lädt die WARP auch bei 'AutoStart' nicht.  Damit kann ich das ja problemlos wieder aktivieren und dann reicht in der Tat die Steuerung über die Ampere-Stärke aus.

Danke!