evcc-io/evcc #3389: Empty EVCC Webpage in Windows 10 / Development - Bad MIME type

powelllens, 2022-05-16:
### Describe the bug

Since Commit: 63b7ffd7e74ef47534432d7b80cc1d82358db18e the webpage of EVCC in windows 10 was blank on my dev machine.
Body empty, no content shown.

Browser Console showed: _`Failed to load module script: Expected a JavaScript module script but the server responded with a MIME type of "text/plain". Strict MIME type checking is enforced for module scripts per HTML spec.`_

Seems to be an Issue with MIME on windows.
Found a workaround / fix:
Changed Registry: `HKEY_CLASSES_ROOT\.js\Content Type` from `text/plain` to `text/javascript`

Is there a way to fix that without the need to update the registy?

### Steps to reproduce

On Windows with Registry: `HKEY_CLASSES_ROOT\.js\Content Type` -> `text/plain`
`Clone Master`
`go run main.go`

127.0.0.1:7070 will be empty

### Configuration details

fresh clone from master, no specific setup

### Log details

-

### What type of operating system are you running?

Windows

powelllens, 2022-05-17:
Found the same Issue related to Django Server:

[Django Troubleshooting](https://django-debug-toolbar.readthedocs.io/en/latest/installation.html#troubleshooting)

Troubleshooting
On some platforms, the Django runserver command may use incorrect content types for static assets. To guess content types, Django relies on the [mimetypes](https://docs.python.org/3/library/mimetypes.html#module-mimetypes) module from the Python standard library, which itself relies on the underlying platform’s map files. If you find improper content types for certain files, it is most likely that the platform’s map files are incorrect or need to be updated. This can be achieved, for example, by installing or updating the mailcap package on a Red Hat distribution, mime-support on a Debian distribution, or by editing the keys under HKEY_CLASSES_ROOT in the Windows registry.

Maybe it would be sufficient to just add such a section to the Readme for Windows contributors...

andig, 2022-05-17:
Can't reproduce:

```
HTTP/1.1 200 OK
Content-Encoding: gzip
Transfer-Encoding: Identity
Accept-Ranges: bytes
Content-Type: text/javascript; charset=utf-8
Vary: Accept-Encoding
Date: Tue, 17 May 2022 09:04:55 GMT
```

Der Mime Type für js Files stimmt.

andig, 2022-05-19:
ping @powelllens gibts hier noch was zu tun?

powelllens, 2022-05-19:
@andig ich hatte das ganze Thema zu beginn mit @naltatis besprochen und einige Infos ausgetauscht. Seit dem Registry fix kann ich den Fehler aber leider nicht mehr nachstellen, selbst nach manueller Änderung der Einträge. Issue kann geschlossen werden, wie man so etwas für die Nachwelt bewahrt weis ich nicht. Der Fehler scheint auf Windows kein unbekannter zu sein, warum sich das von einem auf den anderen Commit geändert hat kann ich ebenfalls nicht sagen.

andig, 2022-05-19:
Passt, der "wirkliche" Fehler bleibt leider erstmal im Dunkeln- evcc *schickt* den richtigen MIME Type.

powelllens, 2022-05-20:
Issue still present with latest Master: 6cbc7b34797d833ca7eaba327bfc8970184101b3
[image]

andig, 2022-05-21:
Ich kanns leider nicht nachvollziehen. Wie liefern eindeutig 

    application/javascript

aus. Warum bei Dir text/plain ankommen sollte ist mir ein Rätsel.

Passiert das nur unter VSCode im Debug Mode oder auch wenn Du auf der Kommandozeile startest? Gibts einen Commit ab dem es nicht mehr ging (`git bisect`)?

andig, 2022-06-04:
Ich mache hier mal zu. Mangels Insights wo/warum sich der Content-Type ändert gibts applikationsseitig leider nix zu tun :(

closed 2022-06-04 by andig