evcc-io/evcc #3396: (manual) install instructions

mstormi, 2022-05-18:
### Describe the bug

Manual pkg install on Debian etc fails because of changes to repo key signing.

Please update instructions, see https://github.com/openhab/openhabian/issues/1688#issuecomment-1129282952

### Steps to reproduce

manual install

### Configuration details

-

### Log details

-

### What type of operating system are you running?

Linux

Macrisu, 2022-05-18:
https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt is the problem

the fix is:

`nano /etc/apt/sources.list.d/evcc-stable.list
`
**remove**

```
deb [signed-by=/usr/share/keyrings/evcc-stable-archive-keyring.gpg] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main

deb-src [signed-by=/usr/share/keyrings/evcc-stable-archive-keyring.gpg] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main
```

**add**

```
deb [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main

deb-src [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main
```

andig, 2022-05-18:
PR to docs repo welcome.

andig, 2022-05-22:
Findet sich jemand der einen PR macht?

Macrisu, 2022-05-22:
> Findet sich jemand der einen PR macht?

Sorry, I do not know how to use Github :(

Macrisu, 2022-05-22:
So, someone must edit 2 files:

**1. For Stable**

https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt

The content must be:

# Source: evcc
# Site: https://github.com/evcc-io/evcc
# Repository: evcc / stable
# Description: Sonne tanken ☀️🚘 - Releases

```
deb [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main

deb-src [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main
```

**2. For Unstable**

https://dl.cloudsmith.io/public/evcc/unstable/debian.deb.txt

The content must be:

```
# Source: evcc
# Site: https://github.com/evcc-io/evcc
# Repository: evcc / unstable
# Description: Sonne tanken ☀️🚘 - Nightly builds

deb [signed-by=/etc/apt/trusted.gpg.d/evcc-unstable.asc] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian any-version main

deb-src [signed-by=/etc/apt/trusted.gpg.d/evcc-unstable.asc] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian any-version main
```

andig, 2022-05-22:
I think there are 2 problems:

1. the evcc docs contain the wrong instructions- that's easy to fix
2. cloudsmith delivers a file that's wrong. I have no idea where that comes from or how it's generated :(. Probably @DerAndereAndi would know, maybe he could give a hint.

Macrisu, 2022-05-22:
Exactly, the easy fix should be in the evcc docs, but for this, Linux command code is needed.

this line must be replaced.

**Stable:**
curl -1sLf 'https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/evcc-stable.list

**Unstable:**
curl -1sLf 'https://dl.cloudsmith.io/public/evcc/unstable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/evcc-unstable.list

This **should** work:

**Stable**

```
echo 'deb [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main' | sudo tee -a sudo tee /etc/apt/sources.list.d/evcc-stable.list

echo 'deb-src [signed-by=/etc/apt/trusted.gpg.d/evcc-stable.asc] https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version main' | sudo tee -a sudo tee /etc/apt/sources.list.d/evcc-stable.list
```

**Unstable:**

```
echo 'deb [signed-by=/etc/apt/trusted.gpg.d/evcc-unstable.asc] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian any-version main'  | sudo tee -a sudo tee /etc/apt/sources.list.d/evcc-unstable.list

echo 'deb-src [signed-by=/etc/apt/trusted.gpg.d/evcc-unstable.asc] https://dl.cloudsmith.io/public/evcc/unstable/deb/debian any-version main' | sudo tee -a sudo tee /etc/apt/sources.list.d/evcc-unstable.list
```

andig, 2022-05-22:
@lskillen could you lend a hand here (similar to https://github.com/openhab/openhabian/issues/1688#issuecomment-1129282952)? It seems that https://dl.cloudsmith.io/public/evcc/stable/debian.deb.txt does no longer work. We can of course replace that with something else but I'd rather understand if there should be a fix at cloudsmith?

Unfortunately I'm not a unix guy, so a bit lost here as to what is cause and effect.

Macrisu, 2022-05-22:
I've verified the commands and it seems to work.
I'll try a PR (my first ever) on the docs.

Wish me luck...
[try.txt](https://github.com/evcc-io/evcc/files/8749502/try.txt)

andig, 2022-05-22:
@Macrisu there already is a pr now.

closed 2022-05-23 by andig