evcc-io/evcc #3397: Alle paar Sekunden: charger out of sync: expected disabled, got enabled

emcepe, 2022-05-18:
### Describe the bug

Ich habe die 2 € "investiert" und mir einen Token besorgt und entsprechend konfiguriert.

Ich bekomme alle paar Sekunden diese Meldung:

```
May 17 17:21:00 raspberry-pi-4b evcc[24194]: [lp-1  ] WARN 2022/05/17 17:21:00 charger out of sync: expected disabled, got enabled
```

mittlerweile habe ich per rsyslogd diese Meldung ins Nirvana loggen lassen, da ansonsten:

```
root@raspberry-pi-4b:~# bzgrep -c 'charger out of sync' /var/log/evcc.log-20220516.bz2`
8603
```

Diese Meldung kommt unabhängig vom Ladestatus, also sowohl beim Laden als auch beim Nichtstun, wobei die Meldung beim Laden etwas anders ist:

```
May 18 09:11:17 raspberry-pi-4b evcc[1312]: [lp-1  ] WARN 2022/05/18 09:11:17 charger logic error: disabled but charging
```

Ebenso habe ich bisher keinerlei Veränderung oder Verbesserung oder irgendwas in der Art feststellen können, ob ich nun via openWB das SOC Modul evcc nutze oder openWB mit dem SOC Modul Kia / Hyundai :-(

Evtl. hat es ja was mit dem o.g. Log Eintrag zu tun. Sieht mir irgendwie so aus, als ob evcc mit der openWB nicht richtig "sprechen" kann. Wobei ein Auto anschließen erkannt wird:
```
May 17 22:43:25 raspberry-pi-4b evcc[25274]: [lp-1  ] INFO 2022/05/17 22:43:25 car connected
```
ebenso der Stop/Start eines Ladevorgangs:
```
May 18 09:16:48 raspberry-pi-4b evcc[1312]: [lp-1  ] INFO 2022/05/18 09:16:48 stop charging <-
May 18 09:18:00 raspberry-pi-4b evcc[1312]: [lp-1  ] INFO 2022/05/18 09:18:00 start charging ->
```

Mein FHEM kann mit der openWB jedenfalls sehr gut "sprechen" ;-)

Was mich ebenso wundert ist, daß bei der openWB niemals Einstellungen geändert werden, zumindest nicht sichtbar (weder via GUI noch per MQTT), egal was man bei evcc auswählt - das würde ich zumindest erwarten, aber vielleicht habe ich auch einfach noch keine Ahnung von evcc.

Dankeschön.

### Steps to reproduce

evcc starten

### Configuration details

```
network:
  schema: http
  port: 7070

log: debug
levels:
  cache: error

interval: 10s

meters:
- type: template
  template: varta-energiespeicher-battery-only
  usage: grid
  host: 192.168.1.1
  port: 502
  name: grid
- type: template
  template: varta-energiespeicher-battery-only
  usage: battery
  host: 192.168.1.1
  port: 502
  name: battery
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.1.2
  password: nfye
  name: pv

chargers:
- type: template
  template: openwb
  host: 192.168.1.3
  loadpointid: 1
  name: wallbox

vehicles:
- type: template
  template: hyundai
  title: Hyundai KONA
  user: foo@bar
  password: blabla
  vin: foobar
  capacity: 64
  language: de
  name: ev

loadpoints:
- title: Garage
  charger: wallbox
  vehicles:
  - ev
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: false
  soc:
    target: 80

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pvs:
    - pv
    batteries:
    - battery
  residualPower: 100
  bufferSoC: 95
  prioritySoC: 90

sponsortoken: nfye
```

### Log details

```
[main  ] INFO 2022/05/17 18:44:08 evcc 0.91
[main  ] INFO 2022/05/17 18:44:08 using config file /etc/evcc.yaml
[main  ] INFO 2022/05/17 18:44:08 listening at :7070
[openwb] INFO 2022/05/17 18:44:09 connecting evcc-663536033 at tcp://192.168.1.3:1883
[openwb] DEBUG 2022/05/17 18:44:09 tcp://192.168.1.3:1883 connected
[openwb] DEBUG 2022/05/17 18:44:09 tcp://192.168.1.3:1883 subscribe openWB/system/Timestamp
[openwb] DEBUG 2022/05/17 18:44:09 openWB/lp/1/boolChargePointConfigured wait for initial value
[site  ] INFO 2022/05/17 18:44:11 site config:
[site  ] INFO 2022/05/17 18:44:11   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/05/17 18:44:11     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/05/17 18:44:11     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/05/17 18:44:11     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/05/17 18:44:11 loadpoint 1:
[lp-1  ] INFO 2022/05/17 18:44:11   mode:        pv
[lp-1  ] INFO 2022/05/17 18:44:11   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/05/17 18:44:11   meters:      charge ✓
[lp-1  ] INFO 2022/05/17 18:44:11     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/05/17 18:44:11   vehicles:    ✓
[lp-1  ] INFO 2022/05/17 18:44:11     vehicle 1: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/05/17 18:44:11 vehicle updated: unknown -> Hyundai KONA
[site  ] DEBUG 2022/05/17 18:44:11 ----
[lp-1  ] DEBUG 2022/05/17 18:44:11 charge power: 0W
[site  ] DEBUG 2022/05/17 18:44:11 pv power: 583W
[site  ] DEBUG 2022/05/17 18:44:11 battery power: 2086W
[site  ] DEBUG 2022/05/17 18:44:11 grid power: 7W
[site  ] DEBUG 2022/05/17 18:44:11 battery soc 0: 80%
[site  ] DEBUG 2022/05/17 18:44:11 site power: 2193W
[lp-1  ] DEBUG 2022/05/17 18:44:11 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/05/17 18:44:11 charger status: A
```

### What type of operating system are you running?

Linux

### Version

```
evcc version 0.91
openWB version 1.9.265 (letzte aktuelle Stable)
```

premultiply, 2022-05-18:
Vermutlich betreibst du die openWB nicht wie vorgesehen im "Nur Ladepunkt-Modus".
https://docs.evcc.io/docs/devices/chargers/#openwb

closed 2022-05-18 by premultiply

premultiply, 2022-05-18:
https://docs.evcc.io/docs/devices/chargers/#openwb
[image]

emcepe, 2022-05-18:
grmpf, ok.

Ich hatte das zwar gelesen, aber im Kopf abgehakt als "natürlich ist meine openWB ein Ladepunkt"

Danke - ich teste weiter.