evcc-io/evcc #3399: SOC poll interval

MarkusGH, 2022-05-18:
### Describe the bug

Es scheint als ob soc/poll/interval beim Porsche API nicht berücksichtigt würde.

soc/poll/interval am Ladepunkt steht auf 30m.

Die Abfragen beim Porsche API erfolgen laut Trace jedoch alle 15 Minuten.

### Steps to reproduce

Fahrzeug am Ladepunkt einstecken.

### Configuration details

Porsche Cayenne Hybrid, PMCC

evcc.yaml

network:
  schema: http
  host: localhost
  port: 7070
 
interval: 3s
 
log: info
levels:
  porsche: trace
  lp-1: debug
  eebus: trace
  openwb: debug
 
mqtt:
  broker: xxx
  topic: evcc
 
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.1942
 
  feedin:
    type: fixed
    price: 0.0816
 
meters:
- name: grid1
  type: movingaverage
  decay: 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power
      scale: -1
      timeout: 30s
    energy:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import
      scale: 1
      timeout: 30s
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1
        timeout: 30s
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2
        timeout: 30s 
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L3
        timeout: 30s
- name: pv1
  type: custom
  power:
    source: mqtt
    topic: mbmd/solaredge1-1/Power
    timeout: 30s
  energy:
    source: mqtt
    topic: mbmd/solaredge1-1/Export
    timeout: 30s
  currents:
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L1
      timeout: 30s
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L2
      timeout: 30s
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L3
      timeout: 30s

chargers:
- name: wallbox1
  type: template
  template: pmcc
  ski: xxx
  enforcePVLimits: true

vehicles:
- name: ev1
  type: template
  template: porsche
  title: xxx
  user: xxx
  password: xxx
  vin: xxx
  capacity: 17.9 
 
loadpoints:
- title: PMCC
  charger: wallbox1
  vehicles:
  - ev1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 30m
    min: 0
    target: 100
    estimate: true
  phases: 1
  mincurrent: 2
  maxcurrent: 20
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m
  guardduration: 5m
 
site:
  title: xxx
  meters:
    grid: grid1
    pvs:
    - pv1
 
eebus:
  interfaces:
  - ens160
  certificate:

### Log details

Siehe Kommentare

### What type of operating system are you running?

Linux

### Version

evcc 0.91

### Log details

Hier das Log:

[putty.log](https://github.com/evcc-io/evcc/files/8698767/putty.log)

### What type of operating system are you running?

Linux

### Version

evcc 0.91

premultiply, 2022-05-18:
Ich weiss das ist wirklich verwirrend aber der Parameter poll.Interval gibt nur das Intervall an wenn das Fahrzeug NICHT lädt.
Vgl. https://docs.evcc.io/docs/reference/configuration/loadpoints/#poll
Die 15 Minuten kommen aus dem vehicle-Cache.

MarkusGH, 2022-05-18:
Hatte mir fast schon gedacht dass ich was übersehen habe :-(
Würde es Sinn machen das Intervall für charging on/off separat einstellbar zu machen?

premultiply, 2022-05-18:
Aus der Anwendersicht wären einzelne Intervallangaben für disconnected, connected, charging bestimmt wesentlich leichter verständlich. Da bin ich völlig bei dir.

Problem: Vom Programmcode funktioniert dies derzeit aber völlig anders:

Eigentlich werden die APIs bei jedem evcc-Zyklus (Default alle 10s) abgefragt. Dies wird aber durch den Cache abgefangen. Dieser lässt Anfragen nur nach aussen durch wenn die Daten im Cache fehlen oder veraltet sind.

Nicht verbundene Fahrzeuge werden normalerweise auf diese Art und Weise nie abgefragt sondern hier gibt es eine Extraroutine die diese je nach Konfiguration (siehe oben) nochmal gesondert ausserhalb des normalen Zyklus abfragt.

MarkusGH, 2022-05-18:
Verstanden. Kann man denn die Lebensdauer der Daten im Cache steuern?

premultiply, 2022-05-18:
Ja, dazu gibt es beim Vehicle einen `cache:` Parameter.
Ich stelle aber gerade fest, dass der wohl noch in der Doku fehlt...

premultiply, 2022-05-18:
Ich mache ansonsten hier erstmal zu, da das kein Bug als solcher ist.

closed 2022-05-18 by premultiply