evcc-io/evcc #3401: [Tasmota] charger enable: switchOn failed

MSi0r, 2022-05-18:
### Describe the bug

My Tasmota device (11.1.0.3) response status is lowercase, as a result EVCC is detecting an error

I changed the file charger/tasmota.go and modified the functions Enable() and Enabled()
to check for uppercase status

strings.ToUpper(res.StatusSTS.Power)

instead of:

res.StatusSTS.Power

That solves the problem.
I'm new to git, how can I provide the updated file?

### Steps to reproduce

Manually start charging ("Sofort"). Tasmota device is enabled immediately, but the error "charger enable: switchOn failed" occurs.

### Configuration details

chargers:
- type: tasmota
  name: wallbox
  uri: http://192.168.x.x  # IP-Adresse oder Hostname
  user: xxx
  password: xxx
  channel: 1
  standbypower: 15  # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet

### Log details

[tasmota] TRACE 2022/05/18 11:42:13 GET http://192.168.x.x/cm?cmnd=Power1+on&password=xxx&user=xxx
[tasmota] TRACE 2022/05/18 11:42:13 {"POWER1":"on"}
[lp-1  ] ERROR 2022/05/18 11:42:13 charger enable: switchOn failed

### What type of operating system are you running?

Linux

### Version

evcc version 0.91 (aad8166b)

thierolm, 2022-05-18:
Interesting ... will provide a fix this noon ...

MSi0r, 2022-05-18:
I generated pull request #3402

PR #3402 (2022-05-19): Fix for issue #3401
Some Tasmota devices return their status lowercase, changed tasmota.go to reflect both.

Fix #3401
closed 2022-05-19 by andig