evcc-io/evcc #3409: UI not visible in Home Assistant (Add-on)

djfanatix, 2022-05-19:
### Describe the bug

Since the update to 0.92, I cannot see the UI anymore when used in Home Assistant.
I managed to get rid of the breaking changes for some chargers (smart plugs)

### Steps to reproduce

click in sidebar HA on EVCC

### Configuration details

#uri: 192.168.1.30:7070 # uri for ui
interval: 5s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c3IiOiJkZW1vIiwic3BlIjp0cnVlLCJleHAiOjE2NDYzMDQ1MzQsImlhdCI6MTYzNzY2NDUzNCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6ImRlbW8ifQ.-poaZ9-pBH9QGyFEV-yj-O-HSCZllgPVxykFArepzNQ

# log settings
log: info
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters

meters:
  - name: grid
    type: custom
    power:
      source: modbus
      model: sunspec
      uri: 192.168.1.30:502 # Port 502 (SetApp) or 1502 (LCD)
      id: 1
      subdevice: 1 # Metering device
      value: 203:W
      scale: -1
  - name: pv
    type: custom
    power:
      source: calc
      add:
       - source: modbus
         model: sunspec
         uri: 192.168.1.30:502 # Port 502 (SetApp) or 1502 (LCD)
         id: 1
        # value: 103:DCW
       - source: modbus
         uri: 192.168.1.30:502 # Port 502 (SetApp) or 1502 (LCD)
         id: 1
         register:
           address: 62836 # Battery 1 Instantaneous Power
           type: holding
           decode: float32s
  - name: battery
    type: custom
    power:
     source: modbus
     uri: 192.168.1.30:502 # Port 502 (SetApp) or 1502 (LCD)
     id: 1
     register:
      address: 62836 # Battery 1 Instantaneous Power
      type: holding
      decode: float32s
     scale: -1
    soc:
     source: modbus
     uri: 192.168.1.30:502 # Port 502 (SetApp) or 1502 (LCD)
[... 204 lines cut ...]
[lp-1  ] INFO 2022/05/19 13:09:52   meters:      charge ✓
[lp-1  ] INFO 2022/05/19 13:09:52     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2022/05/19 13:09:52   vehicles:    ✓
[lp-1  ] INFO 2022/05/19 13:09:52     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-2  ] INFO 2022/05/19 13:09:52 loadpoint 2:
[lp-2  ] INFO 2022/05/19 13:09:52   mode:        pv
[lp-2  ] INFO 2022/05/19 13:09:52   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/05/19 13:09:52   meters:      charge ✓
[lp-2  ] INFO 2022/05/19 13:09:52     charge:    power ✓ energy ✓ currents ✗
[lp-2  ] INFO 2022/05/19 13:09:52   vehicles:    ✓
[lp-2  ] INFO 2022/05/19 13:09:52     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/05/19 13:09:53 vehicle updated: unknown -> Vanmoof S1
[lp-2  ] INFO 2022/05/19 13:09:53 vehicle updated: unknown -> Powerplus
[lp-2  ] DEBUG 2022/05/19 13:09:53 max charge current: 0A
[lp-2  ] DEBUG 2022/05/19 13:09:53 charger disable: contactor delay 4m59s
[lp-1  ] DEBUG 2022/05/19 13:09:53 charge power: 0W
[lp-2  ] DEBUG 2022/05/19 13:09:53 charge power: 0W
[lp-1  ] DEBUG 2022/05/19 13:09:53 charger status: B
[lp-1  ] DEBUG 2022/05/19 13:10:00 vehicle soc: 50%
[lp-1  ] DEBUG 2022/05/19 13:10:00 charge power: 0W
[lp-2  ] DEBUG 2022/05/19 13:10:00 charge power: 0W

### What type of operating system are you running?

Linux

Matze4711, 2022-05-19:
Same for me

andig, 2022-05-19:
@naltatis related to https://github.com/evcc-io/evcc/issues/3389?

Hamudulu, 2022-05-19:
Same here. 
UI nicht sichtbar.

andig, 2022-05-19:
"Same" hilft nicht. Wir brauchen irgendwas greifbares. Was sagt die Entwicklerkonsole im Browser? Abgebrochene HTTP Requests?

Hamudulu, 2022-05-19:
Protokoll sagt:

Using config file:  /config/evcc.yaml
starting evcc: 'evcc --config  /config/evcc.yaml'
[main  ] INFO 2022/05/19 13:51:01 evcc 0.92
[main  ] INFO 2022/05/19 13:51:01 using config file /config/evcc.yaml
[main  ] INFO 2022/05/19 13:51:01 listening at :7070
[lp-1  ] WARN 2022/05/19 13:51:10 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/05/19 13:51:10 site config:
[site  ] INFO 2022/05/19 13:51:10   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/05/19 13:51:10     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/05/19 13:51:10     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/05/19 13:51:10     pv 2:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/05/19 13:51:10 loadpoint 1:
[lp-1  ] INFO 2022/05/19 13:51:10   mode:        pv
[lp-1  ] INFO 2022/05/19 13:51:10   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/05/19 13:51:10   meters:      charge ✓
[lp-1  ] INFO 2022/05/19 13:51:10     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/05/19 13:51:10   vehicles:    ✓
[lp-1  ] INFO 2022/05/19 13:51:10     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/05/19 13:51:10 vehicle updated: unknown -> Kona

naltatis, 2022-05-19:
Yes, Screenshot of your Browsers Network-Tab would help. @powelllens had temporary issues on Window due to mime type hiccups.

Something like this:

[image]

andig, 2022-05-19:
> Protokoll sagt:

evcc läuft, ist aber irrelevant. Fraglich ist warum der Browser nicht will.

Hamudulu, 2022-05-19:
[image]

Hamudulu, 2022-05-19:
Das ist die Meldung die der Browser ausspuckt, wenn ich den evcc link anwähle

naltatis, 2022-05-19:
Hm, das ist irgendwie noch nicht viel Information. Mehr als die Meldung oben bekommst du nicht angezeigt? Siehst du ob das JavaScript von evcc (`./assets/index.1062968a.js`) geladen wird.

andig, 2022-05-19:
> Yes, Screenshot of your Browsers Network-Tab would help. 

Zeig uns mal bitte das Netzwerk Tab in der Entwicklerkonsole. Nicht irgendwas ;)

andig, 2022-05-19:
Netzwerk Tab!

Hamudulu, 2022-05-19:
Moment, bin nicht so schnell bei dem Zeug, muss mich orientieren was du meinst....

andig, 2022-05-19:
Danke. Blöd: hier ist alles grün. Mhhm. Gibts da noch mehr? Keinen Request der auf Fehler läuft?

Betrifft das alle Browser?

Hamudulu, 2022-05-19:
Das hier spuckt Chrome aus:

ha-menu-button.ts:49 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'localize')
    at r.value (ha-menu-button.ts:49:1)
    at r.update (styles.ts:368:1)
    at r.performUpdate (styles.ts:368:1)
    at r.scheduleUpdate (styles.ts:368:1)
    at r._$EC (styles.ts:368:1)

Hamudulu, 2022-05-19:
Ich geh erstmal zurück auf 0.91
Danke für die Super Arbeit und Unterstützung.

sebnaf, 2022-05-19:
Vielleicht möchte @djfanatix seinen öffentlichen mqtt broker mit einem login versehen?

djfanatix, 2022-05-19:
@sebnaf thanks... forgot about it

goebelmeier, 2022-05-19:
> Vielleicht möchte @djfanatix seinen öffentlichen mqtt broker mit einem login versehen?

Und wenn's öffentlich erreichbar ist wohl auch mal lieber mit 'nem Zertifikat. Und der Sponsortoken ist jetzt dann damit wohl auch verbrannt.

arboeh, 2022-05-19:
Bei mir auch. Aus dem Log:

```
Logger: aiohttp.server
Source: components/frontend/__init__.py:278
First occurred: 17:16:14 (2 occurrences)
Last logged: 17:16:53

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/hassio/addon_panel.py", line 61, in post
    await _register_panel(self.hass, addon, data)
  File "/usr/src/homeassistant/homeassistant/components/hassio/addon_panel.py", line 81, in _register_panel
    await panel_custom.async_register_panel(
  File "/usr/src/homeassistant/homeassistant/components/panel_custom/__init__.py", line 120, in async_register_panel
    frontend.async_register_built_in_panel(
  File "/usr/src/homeassistant/homeassistant/components/frontend/__init__.py", line 278, in async_register_built_in_panel
    raise ValueError(f"Overwriting panel {panel.frontend_url_path}")
ValueError: Overwriting panel 49686a9f_evcc
```

arboeh, 2022-05-19:
Eventuell hilft das:

```
Logger: aiohttp.server
Source: /usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py:405
First occurred: 17:59:30 (1 occurrences)
Last logged: 17:59:30

Unhandled exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1183, in _sendfile_fallback
    read = await self.run_in_executor(None, file.readinto, view)
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 514, in start
    resp, reset = await task
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 460, in _handle_request
    reset = await self.finish_response(request, resp, start_time)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 613, in finish_response
    await prepare_meth(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_fileresponse.py", line 286, in prepare
    return await self._sendfile(request, fobj, offset, count)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_fileresponse.py", line 99, in _sendfile
    await loop.sendfile(transport, fobj, offset, count)
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1162, in sendfile
    return await self._sendfile_fallback(transport, file,
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1192, in _sendfile_fallback
    await proto.restore()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 263, in restore
    self._transport.resume_reading()
  File "/usr/local/lib/python3.9/asyncio/sslproto.py", line 343, in resume_reading
    self._ssl_protocol._transport.resume_reading()
AttributeError: 'NoneType' object has no attribute 'resume_reading'
```

andig, 2022-05-19:
Das letzte sieht aus wie ein Bug in HA:

    self._ssl_protocol._transport.resume_reading()
    AttributeError: 'NoneType' object has no attribute 'resume_reading'

Das hilft uns aber alles nix. Wir müssen rausfinden, _worauf_ HA da zugreifen will das nicht funktioniert. Eine URL oder sonstwas. Die Stack Traces sind mangels inhaltlicher Information leider völlig wertlos :(

arboeh, 2022-05-19:
> Das letzte sieht aus wie ein Bug in HA:
>
> ```
> self._ssl_protocol._transport.resume_reading()
> AttributeError: 'NoneType' object has no attribute 'resume_reading'
> ```
>
> Das hilft uns aber alles nix. Wir [...]

Okay. Würde gerne helfen...
Waren die Änderungen im neuen Release so umfangreich, dass man anhand der Änderungen keine Fehlerquelle finden kann?

PR #3413 (2022-05-19): VueRouter: support for non-root paths (e.g. homeassistant)
fixes #3409
closed 2022-05-19 by naltatis