evcc-io/evcc #3410: Remove guardDuration

andig, 2022-05-19:
It does since we call enable/disable explicitly before/after switching. I've always wanted to remove that though, both the explicit calls and guard duration per se. It is from the time when the algorithm wasn't stable and I didn't want to damage my car...

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3139#discussioncomment-2781172_

andig, 2022-05-19:
/cc @premultiply brauchen wir das noch??? Im Zweifel schützt es gegen amok laufende APIs von außen; die Verantwortung könnte man aber auch dem Anwender des APIs aufbürden?

TheSpirit, 2022-05-19:
ich hätte auch gerne eigene "enable/disable" Zeiten für die 1p/3p Umschaltung

andig, 2022-05-19:
> Ich nutze die guardduration - und gerne um unnötiges an/aus der Schütze zu vermeiden und das BMS zu schonen.
> Finde ich durchaus praktisch.

Klar. Aber das kannst Du über enable/disable Delays genauso erreichen. Das ist hier nur der *doppelte* Boden!

RTTTC, 2022-05-19:
My understanding is that `guardDuration` is definitely not needed for PV enable/disable, as it is completely redundant. 
But - it does point out that contactor engagements should be thought about, as I sometimes see `delay 0` in peoples configurations... One may simply not realise what it means and would do physically.

For the phase switching safety ? - no idea..

Hofyyy, 2022-05-26:
Ich habe gerade einmal meine Config geprüft. Ich schütze durch enable und disable auch das zu häufige Schalten. Aus meiner Sicht kann Guardduaration raus.

closed 2022-06-21 by github-actions[bot]