evcc-io/evcc #3416: Nissan API, Unauthorized - Session has timed out

ftm2mbk, 2022-05-20:
### Describe the bug

Die Verbindung zur Nissan API meldet einen Authorisierungsfehler. Aufgetreten ist das (wieder) nach dem Update auf 0.92. Denke aber nicht, dass es mit dem Update zusammenhängt, da ich das mit 0.91 auch schon hatte. Denke eher, wenn der Container neu gebaut wird, gibt es Probleme mit dem Session Token.
Beim letzten Mal hat die API wieder funktioniert, nachdem ich eine neue öffentliche IP Adresse nach Routerneustart bekommen hab. Jedenfalls vermute ich hier einen Zusammenhang.

### Steps to reproduce

Nach dem Rebuild des Docker Containers kommt der Unauthorized Fehler (falls es zuvor funktioniert hat).

### Configuration details

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: trace
levels:
  cache: error

interval: 10s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.2.11  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.2.11  
  name: pv2

chargers:
- type: template
  template: keba 
  host: 192.168.2.20  
  name: wallbox4

vehicles:
- type: template
  template: nissan 
  title: Leaf  
  user: EMAIL
  password: PW
  vin: VIN
  capacity: 40  
  name: ev3
  phases: 1
  targetSoC: 80

loadpoints:
- title: Garage
  charger: wallbox4
  vehicles:
  - ev3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  soc:
    poll:
      mode: connected
      interval: 60m
    estimate: true

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2

### Log details

today at 07:31:18[nissan] TRACE 2022/05/20 07:31:18 POST https://prod.eu2.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate
today at 07:31:18[nissan] TRACE 2022/05/20 07:31:18 {"authId":"eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJvdGsiOiJmN29ub2xoYmg5NWxyb3BtZXJ1Z3NpcDVqYyIsInJlYWxtIjoiL2EtbmNiLXByb2QiLCJzZXNzaW9uSWQiOiJ1Nm5IMmY3T21ycU5OOWpEcFNpcXFleFc3WjguKkFBSlRTUUFDTURJQUFsTkxBQnhTUzFwWFVqbHlPWGR1TVhWaGVqWXJZMUpXVVVGWFlrWnlhMDA5QUFSMGVYQmxBQWxKVGw5TlJVMVBVbGtBQWxNeEFBSTRNZy4uKiJ9.OzXiYlvK5ArPwBZBKWRdGXd52hYgp0qA3HtEMqUV1mw","template":"","stage":"LDAP1","header":"Sign in","callbacks":[{"type":"NameCallback","output":[{"name":"prompt","value":"User Name:"}],"input":[{"name":"IDToken1","value":"***"}]},{"type":"PasswordCallback","output":[{"name":"prompt","value":"Password:"}],"input":[{"name":"IDToken2","value":"***"}]}]}
today at 07:31:18--
today at 07:31:18{"code":401,"reason":"Unauthorized","message":"Session has timed out","detail":{"errorCode":"110"}}

### What type of operating system are you running?

Docker container

### Version

0.92

ftm2mbk, 2022-05-20:
Habe das Auto jetzt einfach nochmal drangehängt und laden lassen. Nach ner halben Stunde scheint die Verbindung zur API zu funktionieren. Der SoC wurde jetzt korrekt angezeigt. Der Poll Timer hatte aber noch 33 Minuten bis zur nächsten API Abfrage. Ein Trace hab ich dazu aber leider nicht.

andig, 2022-05-20:
Ich verstehe nicht, was hier die Frage ist?

ftm2mbk, 2022-05-20:
Das Problem liegt darin, dass die Nissan API bei einem Rebuild des Docker Containers ein "Unauthorized" zurückgibt und es somit keine Ladungszustandsanzeige gibt. Ich hatte dies erst kürzlich bei der 0.91 und da gab es über Tage keinen erfolgreichen Login. 
Als ich gestern auf 0.92 geupdated hab, ist wieder das gleiche passiert. "Unauthorized","message":"Session has timed out". Zuvor hatte die Verbindung zur API funktioniert und hat den SoC sauber angezeigt. Als ich das Ticket aufmachte, dachte ich, dass es jetzt wieder längere Zeit nicht geht. Jedoch kam die Verbindung zur API wohl nach 30min zustande, obwohl der Poll Timer (60min) eigentlich noch nicht abgelaufen war.

Dachte, es wäre nützlich dieses Problem mitzuteilen.

Noch was anderes. Ist es "normal" das der soc poll bei einer Ladung alle 10 sec läuft, obwohl ich 60 min eingestellt habe (siehe Config)? Hier das log dazu:

today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 charge power: 1960W
today at 08:57:33[site  ] DEBUG 2022/05/20 08:57:33 pv power: 2444W
today at 08:57:33[site  ] DEBUG 2022/05/20 08:57:33 grid power: -53W
today at 08:57:33[site  ] DEBUG 2022/05/20 08:57:33 site power: -53W
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 charge currents: [8.9 0 0]A
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 detected phases: 1p
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 charger status: C
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 next soc poll remaining time: 59m50s
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 soc estimated: 61.13% (vehicle: 61.00%)
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 vehicle soc: 61%
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 vehicle range: 171km
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 pv charge current: 9.39A = 9.16A + 0.23A (-53W @ 1p)
today at 08:57:33[lp-1  ] DEBUG 2022/05/20 08:57:33 max charge current: 9.39A
today at 08:57:43[site  ] DEBUG 2022/05/20 08:57:43 ----
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 charge power: 2028W
today at 08:57:43[site  ] DEBUG 2022/05/20 08:57:43 pv power: 2478W
today at 08:57:43[site  ] DEBUG 2022/05/20 08:57:43 grid power: -39W
today at 08:57:43[site  ] DEBUG 2022/05/20 08:57:43 site power: -39W
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 charge currents: [9.22 0 0]A
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 detected phases: 1p
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 charger status: C
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 next soc poll remaining time: 59m49s
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 soc estimated: 61.14% (vehicle: 61.00%)
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 vehicle soc: 61%
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 vehicle range: 171km
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 pv charge current: 9.56A = 9.39A + 0.169A (-39W @ 1p)
today at 08:57:43[lp-1  ] DEBUG 2022/05/20 08:57:43 max charge current: 9.56A

andig, 2022-05-20:
> Das Problem liegt darin, dass die Nissan API bei einem Rebuild des Docker Containers ein "Unauthorized" zurückgibt

Bitte Repro des Problems ohne Docker.

> Noch was anderes. Ist es "normal" das der soc poll bei einer Ladung alle 10 sec läuft, obwohl ich 60 min eingestellt habe 

OT: Nein uns ist aus dem Log auch nicht erkennbar, dass das der Fall wäre. Gerne in Discussions weiter besprechen, kein Issue.

closed 2022-05-20 by andig