evcc-io/evcc #3418: Zwei Easee Home -> nur eine wird im HomeManager 2.0 erkannt

Chris8er, 2022-05-20:
### Describe the bug

Hallo zusammen,

ich habe evcc seit gestern im Einsatz. Meine PV Anlage besteht aus:

Sunny Tripower (TP8.0-3AV-40 421)
Sunny HomeManager 2.0
2x Easee Home

Im Sunny Portal wird nur eine der beiden Wallboxen erkannt (Stellplatz). Garage wird nicht gefunden.
Darüber hinaus wird auch der Hausverbrauch nicht ermittelt und steht somit immer auf 0W.

Der UDP-Port in use (siehe logs) verwundert mich etwas. Ich habe evcc vorher gestoppt bevor ich dann mit debug logging gestartet hab.

### Steps to reproduce

* evcc startet zusammen mit dem RasPi
* im SunnyPortal neue Geräte suchen
-> Es wurden keine Geräte gefunden.

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: 
xxx

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: xxx
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: xxx
  password: xxx
  name: pv1

chargers:
- type: template
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  name: wallbox2
- type: template
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  name: wallbox3

loadpoints:
- title: Stellplatz
  charger: wallbox3
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Garage
  charger: wallbox2
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid1

hems:
  type: sma
  AllowControl: false

### Log details

[main  ] INFO 2022/05/20 09:22:26 evcc 0.91
[main  ] INFO 2022/05/20 09:22:26 using config file /etc/evcc.yaml
[main  ] INFO 2022/05/20 09:22:26 listening at :7070
[lp-1  ] WARN 2022/05/20 09:22:36 ignoring phases config (3p) for switchable charger
[lp-1  ] DEBUG 2022/05/20 09:22:36 phase timer inactive
[lp-2  ] WARN 2022/05/20 09:22:36 ignoring phases config (3p) for switchable charger
[lp-2  ] DEBUG 2022/05/20 09:22:36 phase timer inactive
[semp  ] WARN 2022/05/20 09:22:36 SEMP_BASE_URL unspecified, using http://192.168.178.33:7070 instead
[semp  ] ERROR 2022/05/20 09:22:36 listen udp4 224.0.0.0:1900: bind: address already in use
[semp  ] ERROR 2022/05/20 09:22:36 listen udp4 224.0.0.0:1900: bind: address already in use
[semp  ] ERROR 2022/05/20 09:22:36 listen udp4 224.0.0.0:1900: bind: address already in use
[site  ] INFO 2022/05/20 09:22:37 site config:
[site  ] INFO 2022/05/20 09:22:37   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/05/20 09:22:37     grid:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/05/20 09:22:37 loadpoint 1:
[lp-1  ] INFO 2022/05/20 09:22:37   mode:        pv
[lp-1  ] INFO 2022/05/20 09:22:37   charger:     power ✓ energy ✗ currents ✓ phases ✓
[lp-1  ] INFO 2022/05/20 09:22:37   meters:      charge ✓
[lp-1  ] INFO 2022/05/20 09:22:37     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/05/20 09:22:37   vehicles:    ✗
[lp-2  ] INFO 2022/05/20 09:22:37 loadpoint 2:
[lp-2  ] INFO 2022/05/20 09:22:37   mode:        pv
[lp-2  ] INFO 2022/05/20 09:22:37   charger:     power ✓ energy ✗ currents ✓ phases ✓
[lp-2  ] INFO 2022/05/20 09:22:37   meters:      charge ✓
[lp-2  ] INFO 2022/05/20 09:22:37     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2022/05/20 09:22:37   vehicles:    ✗
[lp-2  ] DEBUG 2022/05/20 09:22:37 max charge current: 6A
[site  ] DEBUG 2022/05/20 09:22:37 ----
[lp-1  ] DEBUG 2022/05/20 09:22:37 charge power: 0W
[lp-2  ] DEBUG 2022/05/20 09:22:37 charge power: 1451W
[site  ] DEBUG 2022/05/20 09:22:37 grid power: 335W
[site  ] DEBUG 2022/05/20 09:22:37 grid currents: [4.82 1.68 1.57]A
[site  ] DEBUG 2022/05/20 09:22:37 site power: 335W
[lp-1  ] DEBUG 2022/05/20 09:22:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/05/20 09:22:37 charger status: A
[site  ] DEBUG 2022/05/20 09:22:47 ----
[lp-1  ] DEBUG 2022/05/20 09:22:47 charge power: 0W
[lp-2  ] DEBUG 2022/05/20 09:22:47 charge power: 1451W
[site  ] DEBUG 2022/05/20 09:22:47 grid power: 330W
[site  ] DEBUG 2022/05/20 09:22:47 grid currents: [4.82 1.71 1.58]A
[site  ] DEBUG 2022/05/20 09:22:47 site power: 330W
[lp-2  ] DEBUG 2022/05/20 09:22:47 charge currents: [6.07 0.01 0.01]A
[lp-2  ] DEBUG 2022/05/20 09:22:47 charger status: C
[lp-2  ] INFO 2022/05/20 09:22:47 start charging ->
[lp-2  ] DEBUG 2022/05/20 09:22:47 available power 1121W < 4140W min 3p threshold
[lp-2  ] DEBUG 2022/05/20 09:22:47 start phase scale1p timer
[lp-2  ] DEBUG 2022/05/20 09:22:47 phase scale1p in 3m0s
[lp-2  ] DEBUG 2022/05/20 09:22:47 pv charge current: 5.52A = 6A + -0.478A (330W @ 3p)
[site  ] DEBUG 2022/05/20 09:22:57 ----
[lp-1  ] DEBUG 2022/05/20 09:22:57 charge power: 0W
[lp-2  ] DEBUG 2022/05/20 09:22:57 charge power: 1451W
[site  ] DEBUG 2022/05/20 09:22:57 grid power: 345W
[site  ] DEBUG 2022/05/20 09:22:57 grid currents: [4.83 1.7 1.56]A
[site  ] DEBUG 2022/05/20 09:22:57 site power: 345W
[lp-1  ] DEBUG 2022/05/20 09:22:57 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/05/20 09:22:57 charger status: A
[site  ] DEBUG 2022/05/20 09:23:07 ----
[lp-1  ] DEBUG 2022/05/20 09:23:07 charge power: 0W
[lp-2  ] DEBUG 2022/05/20 09:23:07 charge power: 1451W
[site  ] DEBUG 2022/05/20 09:23:07 grid power: 340W
[site  ] DEBUG 2022/05/20 09:23:07 grid currents: [4.82 1.69 1.56]A
[site  ] DEBUG 2022/05/20 09:23:07 site power: 340W
[lp-2  ] DEBUG 2022/05/20 09:23:07 charge currents: [6.07 0.01 0.01]A
[lp-2  ] DEBUG 2022/05/20 09:23:07 detected phases: 1p
[lp-2  ] DEBUG 2022/05/20 09:23:07 charger status: C
[lp-2  ] DEBUG 2022/05/20 09:23:07 phase timer reset
[lp-2  ] DEBUG 2022/05/20 09:23:07 phase timer inactive
[lp-2  ] DEBUG 2022/05/20 09:23:07 pv charge current: 4.52A = 6A + -1.48A (340W @ 1p)
[lp-2  ] DEBUG 2022/05/20 09:23:07 site power 340W >= 0W disable threshold
[lp-2  ] DEBUG 2022/05/20 09:23:07 pv disable timer start: 3m0s
[lp-2  ] DEBUG 2022/05/20 09:23:07 pv disable in 3m0s
[site  ] DEBUG 2022/05/20 09:23:17 ----

### What type of operating system are you running?

Linux

### Version

0.91

premultiply, 2022-05-20:
Läuft vielleicht noch eine andere evcc-Instanz? Im Zweifelsfall mal Reboot ausführen.

Die SEMP-Schnittstelle kannst du einfach über http://192.168.178.33:7070/semp abrufen.
Diese Ausgabe ist interessant. Tauchen dort beide Boxen auf?

Chris8er, 2022-05-20:
Rebooted habe ich schon ein paar mal. Hier nach einem frischen reboot:

ps aux | grep evcc
root       719  2.7  2.1 833800 19960 ?        Ssl  09:27   0:26 /usr/bin/evcc
root      3579  0.0  0.0   4368   504 pts/1    S+   09:43   0:00 grep evcc

semp sieht gut aus?!

```
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Device2EM xmlns="http://www.sma.de/communication/schema/SEMP/v1">
	<DeviceInfo>
		<Identification>
			<DeviceId>F-28081973-c3959b158043-00</DeviceId>
			<DeviceName>Stellplatz</DeviceName>
			<DeviceType>EVCharger</DeviceType>
			<DeviceSerial>b827ebac3c2f-0</DeviceSerial>
			<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
		</Identification>
		<Characteristics>
			<MinPowerConsumption>1380</MinPowerConsumption>
			<MaxPowerConsumption>11040</MaxPowerConsumption>
		</Characteristics>
		<Capabilities>
			<CurrentPower>
				<Method>Measurement</Method>
			</CurrentPower>
			<Timestamps>
				<AbsoluteTimestamps>false</AbsoluteTimestamps>
			</Timestamps>
			<Interruptions>
				<InterruptionsAllowed>true</InterruptionsAllowed>
			</Interruptions>
			<Requests>
				<OptionalEnergy>true</OptionalEnergy>
			</Requests>
		</Capabilities>
	</DeviceInfo>
	<DeviceInfo>
		<Identification>
			<DeviceId>F-28081973-c3959b158044-00</DeviceId>
			<DeviceName>Garage</DeviceName>
			<DeviceType>EVCharger</DeviceType>
			<DeviceSerial>b827ebac3c2f-1</DeviceSerial>
			<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
		</Identification>
		<Characteristics>
			<MinPowerConsumption>1380</MinPowerConsumption>
			<MaxPowerConsumption>11040</MaxPowerConsumption>
		</Characteristics>
		<Capabilities>
			<CurrentPower>
				<Method>Measurement</Method>
			</CurrentPower>
			<Timestamps>
				<AbsoluteTimestamps>false</AbsoluteTimestamps>
			</Timestamps>
			<Interruptions>
				<InterruptionsAllowed>true</InterruptionsAllowed>
			</Interruptions>
			<Requests>
				<OptionalEnergy>true</OptionalEnergy>
			</Requests>
		</Capabilities>
	</DeviceInfo>
	<DeviceStatus>
		<DeviceId>F-28081973-c3959b158043-00</DeviceId>
		<EMSignalsAccepted>false</EMSignalsAccepted>
		<Status>Off</Status>
		<PowerConsumption>
			<PowerInfo>
				<AveragePower>0</AveragePower>
				<Timestamp>0</Timestamp>
				<AveragingInterval>60</AveragingInterval>
			</PowerInfo>
		</PowerConsumption>
	</DeviceStatus>
	<DeviceStatus>
		<DeviceId>F-28081973-c3959b158044-00</DeviceId>
		<EMSignalsAccepted>false</EMSignalsAccepted>
		<Status>On</Status>
		<PowerConsumption>
			<PowerInfo>
				<AveragePower>3102</AveragePower>
				<Timestamp>0</Timestamp>
				<AveragingInterval>60</AveragingInterval>
			</PowerInfo>
		</PowerConsumption>
	</DeviceStatus>
	<PlanningRequest>
		<Timeframe>
			<DeviceId>F-28081973-c3959b158044-00</DeviceId>
			<EarliestStart>0</EarliestStart>
			<LatestEnd>86400</LatestEnd>
			<MinEnergy>0</MinEnergy>
			<MaxEnergy>1000</MaxEnergy>
			<MaxPowerConsumption>11040</MaxPowerConsumption>
			<MinPowerConsumption>1380</MinPowerConsumption>
		</Timeframe>
	</PlanningRequest>
</Device2EM>
```

premultiply, 2022-05-20:
Dann sollte alles passen.
Für die unzuverlässigkeit des SHM/SP kann evcc nichts.

Ggf. hilft mal ein Reboot des SHM (Knopf sicher mehr als 7 Sekunden drücken).

Chris8er, 2022-05-20:
Reboot des SHM hab ich durchgeführt. Ich habe mal die Reihenfolge der Loadpoints geändert und Gerät im SHM gelöscht. Jetzt erkennt er die andere Wallbox. Es scheint als würde er immer nur das erste Gerät erkennen. Ggf. irgendwas mit der XML nicht in Ordnung?

andig, 2022-05-21:
@premultiply kannst Du das "trocken" mit dem SHM testen? Ich wüsste nicht woran's hier hängen soll?

Chris8er, 2022-05-21:
Ich habe heute noch den "Smart Appliance Enabler" installiert, der ebenfalls SEMP supported. Auch hier habe ich testweise zwei Geräte eingerichtet. Diese werden sofort beide vom SHM erkannt. Hier mal der Output des SEMP Endpoints. Evtl. hilfts ja.

```
<Device2EM xmlns="http://www.sma.de/communication/schema/SEMP/v1">
	<DeviceInfo>
		<Identification>
			<DeviceId>F-14071983-000000000001-00</DeviceId>
			<DeviceName>Waschmaschine</DeviceName>
			<DeviceType>WashingMachine</DeviceType>
			<DeviceSerial>12345</DeviceSerial>
			<DeviceVendor>Bosch</DeviceVendor>
		</Identification>
		<Characteristics>
			<MaxPowerConsumption>2000</MaxPowerConsumption>
		</Characteristics>
		<Capabilities>
			<CurrentPower>
				<Method>Estimation</Method>
			</CurrentPower>
			<Timestamps>
				<AbsoluteTimestamps>false</AbsoluteTimestamps>
			</Timestamps>
			<Interruptions>
				<InterruptionsAllowed>false</InterruptionsAllowed>
			</Interruptions>
			<Requests>
				<OptionalEnergy>false</OptionalEnergy>
			</Requests>
		</Capabilities>
	</DeviceInfo>
	<DeviceInfo>
		<Identification>
			<DeviceId>F-14071983-000000000002-00</DeviceId>
			<DeviceName>Trockner</DeviceName>
			<DeviceType>Dryer</DeviceType>
			<DeviceSerial>54321</DeviceSerial>
			<DeviceVendor>Bosch</DeviceVendor>
		</Identification>
		<Characteristics>
			<MaxPowerConsumption>2500</MaxPowerConsumption>
		</Characteristics>
		<Capabilities>
			<CurrentPower>
				<Method>Estimation</Method>
			</CurrentPower>
			<Timestamps>
				<AbsoluteTimestamps>false</AbsoluteTimestamps>
			</Timestamps>
			<Interruptions>
				<InterruptionsAllowed>false</InterruptionsAllowed>
			</Interruptions>
			<Requests>
				<OptionalEnergy>false</OptionalEnergy>
			</Requests>
		</Capabilities>
	</DeviceInfo>
	<DeviceStatus>
		<DeviceId>F-14071983-000000000001-00</DeviceId>
		<EMSignalsAccepted>false</EMSignalsAccepted>
		<Status>Offline</Status>
		<PowerConsumption>
			<PowerInfo>
				<AveragePower>0</AveragePower>
				<Timestamp>0</Timestamp>
				<AveragingInterval>60</AveragingInterval>
			</PowerInfo>
		</PowerConsumption>
	</DeviceStatus>
	<DeviceStatus>
		<DeviceId>F-14071983-000000000002-00</DeviceId>
		<EMSignalsAccepted>false</EMSignalsAccepted>
		<Status>Offline</Status>
		<PowerConsumption>
			<PowerInfo>
				<AveragePower>0</AveragePower>
				<Timestamp>0</Timestamp>
				<AveragingInterval>60</AveragingInterval>
			</PowerInfo>
		</PowerConsumption>
	</DeviceStatus>
</Device2EM>
```

premultiply, 2022-05-22:
Um das aktuell selbst zu testen fehlt mir leider etwas die Zeit und Ruhe, kann das aber gerne nachholen sobald es wieder geht.

Probier doch mal bitte testweise mit dem SAE die Geräte exakt mit der von evcc unter /semp ausgegeben DeviceIDs und DeviceSerial anzulegen. Tauchen sie dann im SP beide auf?
Falls nein, ändert es etwas wenn du die enthaltenen Buchstaben alle groß schreibst?

Wir verwenden als DeviceType ja EVCharger. Vielleicht mag der (dein?) SHM nicht mehr als einen EVCharger?

premultiply, 2022-05-23:
Ich habe es nun gerade bei mir selbst getestet und einen zweiten Ladepunkt direkt in evcc hinzugefügt.
Dieser wurde wie erwartet und vorgesehen sofort vom Sunny Portal erkannt und ich könnte ihn hinzufügen:
[image]

andig, 2022-05-23:
Dann stelle ich hier mal auf Discussion um- erstmal kein technisches Problem bei evcc.