evcc-io/evcc #3421: Mehrfache Push-Updates für chargeCurrent in Serie

premultiply, 2022-05-20:
### Describe the bug

Die Push-Updates für chargeCurrent (z.B. per MQTT) erfolgen pro Regelzyklus mehrfach und in direkter zeitlicher Folge jeweils mit zunächst veralteten Werten (aus dem letzten Zyklus) und dann den neuen, aktualisierten Werten.

Das macht z. B. Zeitreihengraphen unnötig eckig und erhöht das Datenvolumen:
[image]

### Steps to reproduce

MQTT-Subscription für chargeCurrent einrichten und zeitliche Abfolge der Updates beobachten.

### Configuration details

-

### Log details

-

### What type of operating system are you running?

Linux

### Version

v0.92

andig, 2022-05-20:
Ich denk grad nochmal drüber nach. Ist nicht genau diese Kurve richtig?! Es geht um den maximalen (!) Ladestrom, also maxCurrent (ja, der Name suggeriert Anderes). Genau der ändert sich aber nicht linear, sondern ist für das Intervall immer konstant.

These: wir sollten hier gar nichts tun!

premultiply, 2022-05-20:
Nee, das kommt im Graph von alleine wenn man nur Updates schickt wenn etwas geändert wurde.
Der Fehler ist hier das direkt hintereinander der alte und der neue Wert geschickt wird.

andig, 2022-05-20:
> Nee, das kommt im Graph von alleine wenn man nur Updates schickt wenn etwas geändert wurde.

*DAS* ist aber eine Eigenschaft der Darstellung, nicht der Daten! 

> Der Fehler ist hier das direkt hintereinander der alte und der neue Wert geschickt wird.

*HIER* steckts in den Daten.

Ist mir aber im Grunde gleich, siehe Vorschlag zu Test.

premultiply, 2022-05-20:
Genau. Und ich hab ich gewundert warum man die Darstellung nicht mit den entsprechenden Filtern "smoothen" konnte.
So bin ich erst darauf gekommen dass da mehrfache Updates direkt hintereinander einprasseln.

closed 2022-08-07 by github-actions[bot]